platformio/platformio-core #4157: No update of c_cpp_properties.json and launch.json files on mapped network drive with PIO version 5.2.5a5

EricPERONNIN, 2022-01-19:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 5.2.5a5

### Description of problem
Creating projects on a mapped network drive works fine with 5.2.5a5 PIO Version.
But .vscode configurations files (launch.json , c_cpp_properties.json ...) are not generating when loading an old project stored on a mapped network drive.
Then, if one of them is deleted, they will never come back.

When adding a library to a project, c_cpp_properties.json is not updated. This breaks automatic update of intellisense.

#### Steps to Reproduce

1. Map a network drive on Windows (for users who don't have a network, share a directory and map it with localhost ip)
2. Create a new project with PIO
3. Delete .vscode json files
4. Close VS Code and reload the project

### Actual Results
.vscode json files are not generating.

ivankravets, 2022-01-19:
Do you see any issues when trying to build the project?

ivankravets, 2022-01-20:
Try to run in PlatformIO CLI `pio project init --ide vscode`. Restart VSCode.

EricPERONNIN, 2022-01-20:
Good test to find where it breaks :
PS Z:\Documents\PlatformIO\Projects\SAe1_BaseAutomatisation> pio project init --ide vscode
The next files/directories have been created in \\ERIC-PC-DELL\reseauIUT\Documents\PlatformIO\Projects\SAe1_BaseAutomatisation
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File
Error: Traceback (most recent call last):
  File "c:\vscode\.platformio\penv\lib\site-packages\platformio\__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "c:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "c:\vscode\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "c:\vscode\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\vscode\.platformio\penv\lib\site-packages\platformio\commands\project.py", line 194, in project_init
    ProjectGenerator(config, environment, ide, board).generate()
  File "c:\vscode\.platformio\penv\lib\site-packages\platformio\project\generator.py", line 147, in generate
    tpl_vars = self._load_tplvars()
  File "c:\vscode\.platformio\penv\lib\site-packages\platformio\project\generator.py", line 101, in _load_tplvars
    "src_files": self.get_src_files(),
  File "c:\vscode\.platformio\penv\lib\site-packages\platformio\project\generator.py", line 129, in get_src_files
    os.path.relpath(os.path.join(os.path.realpath(root), f))
  File "C:\VSCode\Python37\lib\ntpath.py", line 562, in relpath
    path_drive, start_drive))
ValueError: path is on mount 'Z:', start on mount '\\\\ERIC-PC-DELL\\reseauIUT'

EricPERONNIN, 2022-01-20:
I've put some prints in the relpath function of ntpath.py to check some variables
Weirdly, when I create a new project with the CLI function, these prints are not executed => relpath is not called when project is created.

Test : 
PS C:\Users\eric> pio project init -d Z:\Documents\PlatformIO\Projects\ -b mattairtech_samd21e18a --ide vscode
The next files/directories have been created in \\ERIC-PC-DELL\reseauIUT\Documents\PlatformIO\Projects
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File

Project has been successfully initialized including configuration files for `vscode` IDE.

ivankravets, 2022-01-21:
Could you provide an output from the `net use` command?

EricPERONNIN, 2022-01-21:
Net use 
 Déconnectée  E:        \\localhost\MusiqueE      Microsoft Windows Network
Déconnectée  T:        \\toutatis.iut-nantes.univ-nantes.prive\commun
                                                Microsoft Windows Network
OK           Z:        \\ERIC-PC-DELL\reseauIUT  Microsoft Windows Network

EricPERONNIN, 2022-01-22:
net use return :
OK           E:        \\localhost\MusiqueE      Microsoft Windows Network
Déconnectée  T:        \\toutatis.iut-nantes.univ-nantes.prive\commun
                                                Microsoft Windows Network
OK           Z:        \\ERIC-PC-DELL\reseauIUT  Microsoft Windows Network

ivankravets, 2022-02-03:
I can't reproduce this issue with the latest Python and PlatformIO Core.

> C:\VSCode\Python37\lib\ntpath.py

Could you uninstall all Pythons from the system including `C:\VSCode\Python37\`, remove `c:\vscode.platformio\penv` and `c:\vscode.platformio\python3`. 

Restart VSCode. Open project from network drive. Does it work now?