platformio/platformio-core #4163: `pio exec`: Run custom toolchain commands

mamins1376, 2022-01-31:
**Feature Request**: Support `pio exec` command, which is useful to run binaries associated with the current project, without getting involved with including toolchain directories in `$PATH` for each project.

As an example, for _stm8_ platform, there are `sdcc`, `stm8flash` and `stm8-binutils` which is a powerful combination. Problem is, without adding their relevant prefixes in the path, one is going to have a painful experience. Either they can include them in the shell (possibly automating this by sourcing a script), or call them by their path. Both have their flaws: inclusion is prone to versioning mismatch and loss of cross platform compatibility, while the second one is tedious and repetitive.

My proposal is to add support for `pio exec cmd --arg` (sub)command, which is just a shortcut for `PATH=~/.platformio/...:$PATH cmd --arg` at it's core.

ivankravets, 2022-01-31:
Thanks for filing this feature request. We have recently discussed it with @vshymanskyy and it would be great to have this functionality.

closed 2022-02-04 by ivankravets

ivankravets, 2022-02-04:
See docs https://docs.platformio.org/en/latest/core/userguide/cmd_exec.html

@mamins1376  , @vshymanskyy , please re-test with `pio upgrade --dev`.

commit 9c10e002 (2022-02-04): Run command from a PlatformIO package with a new `pio exec` command // Resolve #4163