platformio/platformio-core #4174: PIO Embedded Python Installation - SSL: CERTIFICATE_VERIFY_FAILED

fabltd, 2022-02-09:
### Configuration

**Operating system**:

`macOS Monterey - 12.1`

**PlatformIO Version** (`platformio --version`):

`PlatformIO Core, version 5.2.5b1`

### Description of problem

Using custom build flags as follows:

`build_flags = !python3 build_flags.py`

```
This python file uses the requests lib requests makes a get request to fetch some parameters for the build flags.

The code fails to run due to the ability to check the SSL certifcate:

ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)
```

This it appears is a know error on mac and the recommendation is to run - install_certificates.command however this does not appear to be present in the PIO Embedded installation.

#### Steps to Reproduce

```
create a pyhton file and import it as a buildflags paramter:

build_flags = !python3 build_flags.py

example build_flags.py
```

```
import requests

x = requests.get('https://w3schools.com')
print(x.status_code)

```

### Actual Results

```
python3.8/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)

During handling of the above exception, another exception occurred:

```
### Expected Results

`Code should complie with out the above errors`

ivankravets, 2022-02-09:
> This it appears is a know error on mac and the recommendation is to run - install_certificates.command however this does not appear to be present in the PIO Embedded installation.

1. If you use PlatformIO IDE on macOS, it will download our own portable Python with valid certificates.
2. If you prefer to use your own system-wide Python, there is a note in the docs https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

fabltd, 2022-02-09:
Is there some modifcation required to get requests to work with your portable Python? 

It seems the certificates are some how not found by the requests libs.

ivankravets, 2022-02-11:
Please reopen if you still need help

closed 2022-02-11 by ivankravets

Joebeazelman, 2024-01-05:
I am currently having the same issue. I posted it on community.platformio.org.

https://community.platformio.org/t/ssl-certificate-verify-failed-certificate-verify-failed-unable-to-get-local-issuer-certificate/37642