platformio/platformio-core #4182: library.json: build->includeDir, will it support multiple include paths?

IOsetting, 2022-02-15:
- [ ] **Feature Request**.
      If `includeDir` supports multiple include paths, it will make libraries integration much easier.

Take FreeRTOS as an example, FreeRTOS requires three directories to be included in the compiler’s include path, it is not doable in current library.json without moving the headers files. It requires that `includeDir` supports multiple include paths, something like the following
```
{
    "name": "FreeRTOS",
    "version": "10.4.6",
    "build": {
        "includeDir": [
            "./",
            "include",
            "portable/GCC/ARM_CM3"
        ],
        "srcFilter": [
            "+<*.c>",
            "+<portable/GCC/ARM_CM3/port.c>",
            "+<portable/MemMang/heap_4.c>"
        ]
    }
}
```

Best regards!

ivankravets, 2022-02-15:
Please use https://docs.platformio.org/en/latest/librarymanager/config.html#flags and specify relative paths to the library.
```json
{
    "name": "FreeRTOS",
    "version": "10.4.6",
    "build": {
        "flags": [
            "-Iportable/GCC/ARM_CM3"
        ],
        "srcFilter": [
            "+<*.c>",
            "+<portable/GCC/ARM_CM3/port.c>",
            "+<portable/MemMang/heap_4.c>"
        ]
    }
}
```

IOsetting, 2022-02-15:
Cool, thank you!