platformio/platformio-core #4184: GitHub dependency graph for lib_deps

tysonmatanich, 2022-02-16:
What kind of issue is this?

- [x] **Feature Request**.

------------------------------------------------------------------

### Configuration

**Operating system**: Win10

**PlatformIO Version** (`platformio --version`): 5.2.5

### Description of problem
There appears to be no current support for GitHub dependency graph and Dependabot alerts. Supporting GitHub dependency graph would be great for security updates in addition to the dependency graph it adds.

#### Steps to Reproduce

1. Create new PlatformIO project via "New Project" button (configure name, board, framework, location)
2. Open the "Libraries" tab find and select a library to install (for example: esp32-camera)
3. Click "Add to Project" button
4. Select the project that the dependency should be added to (seems like this should only be the current project)
5. Click the "Add" button
6. Push project to GitHub

### Actual Results

Dependency is installed in `/.pio/libdeps` and added to `lib_deps` in `/platformio.ini` file which [GitHub doesn't support](https://docs.github.com/en/code-security/supply-chain-security/understanding-your-software-supply-chain/about-the-dependency-graph#supported-package-ecosystems) for package management:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = espressif/esp32-camera@^2.0.0
```
### Expected Results

Dependency is configured in a way compatible with GitHub's dependency graph.

### Additional info

Community suggestion by @nicolas-rabault https://community.platformio.org/t/make-platformio-work-with-github-dependencies-graph/12965

leon0399, 2023-08-08:
Dependency graph reports was added into `vcpkg` recently, using Github Dependency Graph Reports API, so I guess it can be useful:

https://devblogs.microsoft.com/cppblog/vcpkg-2023-06-20-and-2023-07-21-releases-github-dependency-graph-support-android-tested-triplets-xbox-triplet-improvements-and-more/
https://devblogs.microsoft.com/cppblog/vcpkg-integration-with-the-github-dependency-graph/
https://github.com/microsoft/vcpkg-tool/pull/989/files

DjordjeMandic, 2024-09-05:
Looking for same solution