platformio/platformio-core #4190: A symlink .h file leads to compile errors

hamishcunningham, 2022-02-25:
hi,

I'm not sure if this is the same problem as #4018, but I can't seem to get a symlink header file to be seen.

For example a symlink in src/ gets me an error like
```
*** [.pio/build/featheresp32/src/main.cpp.o] Implicit dependency 'src/private.h' not found, needed by target '.pio/build/featheresp32/src/main.cpp.o'.
```

A symlink from lib/ gets me 
```
FileNotFoundError: [Errno 2] No such file or directory: '/home/ubuntu/project/lib/private/private.h':
  File "/home/ubuntu/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
```

I'm on Ubuntu 20.04 and running PlatformIO core 5.2.4.

I've tried various settings of lib_ldf_mode and lib_compat_mode but they don't seem to make a difference.

THanks!

ivankravets, 2022-03-06:
Could you make a symlink with FULL path, not relative. When you build your project, PlatformIO changes the current working directory to the project folder.

ivankravets, 2022-04-02:
Please reopen if you still need help

closed 2022-04-02 by ivankravets

hamishcunningham, 2022-04-04:
Thanks Ivan, sorry I didn't close it earlier!