platformio/platformio-core #4193: Support cross-arch toolchain packages

quentinmit, 2022-02-28:
- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

### Configuration

**Operating system**: Debian 11 on aarch64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.2.5`

### Description of problem

Some processors support multiple instruction sets (e.g. x64 supports x86 binaries, arm64 supports arm32 binaries). Currently PlatformIO doesn't know that it can use 32-bit packages on a 64-bit system, so it refuses to use a 32-bit package unless Python is run inside `linux32`. This is because platformio is currently just calling `platform.machine()` to get a single architecture name: https://github.com/platformio/platformio-core/blob/834c7b0defe39ff998cd9dc856657aeb837c86ca/platformio/util.py#L92

I'm not sure of the best generic way to solve this; there are many different combinations of architectures that can be used together (e.g. x86 on x64, arm32 on arm64, ppc on ppc64, but also x86 on arm with `qemu-user-binfmt`. Maybe it could default to checking what [multiarch directories](https://wiki.debian.org/Multiarch/Tuples) exist in `/usr/lib` with a configuration override to enable extra architectures?

#### Steps to Reproduce

On aarch64:

```
# If not already enabled on your system
sudo dpkg --add-architecture armel
sudo apt update
sudo apt install libc6:armel libstdc++6:armel
# Now build hello world
mkdir /tmp/test
cd /tmp/test
pio project init --board esp32-c3-devkitm-1
pio run
linux32 pio run
```

### Actual Results

`pio run` fails with:
```
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2' requirements for your system 'linux_aarch64'
```

`linux32 pio run` succeeds

### Expected Results

`pio run` with or without `linux32` succeeds.

### If problems with PlatformIO Build System:

Exactly as generated by `pio project init`:

**The content of `platformio.ini`:**
```ini
[env:esp32-c3-devkitm-1]
platform = espressif32
board = esp32-c3-devkitm-1
framework = espidf
```

**Source file to reproduce issue:**
```cpp
void app_main() {}
```

### Additional info
See also https://github.com/espressif/crosstool-NG/issues/21 requesting that `espressif/toolchain-xtensa-esp32` be built for aarch64; if that's resolved, you won't be able to use the reproduction steps above, but any other 32-on-64 combination will also be affected.

jpcornil-git, 2022-03-20:
I just root caused a related issue (ESPhome using platformio on RPi and running within a 32b userland but with a 64b kernel), cfr. 
https://github.com/esphome/issues/issues/1386#issuecomment-1073117862

Platformio is indeed only using python platform.machine() (aarch64 here) in its package selection process and picks 64b (iso 32b) in this case. As a consequence, when binaries are invoked, you get a (counterintuitive) 'not found'
```shell
sh: 1: xtensa-lx106-elf-g++: not found
````

I patched [platformio/util.py](https://github.com/platformio/platformio-core/blob/develop/platformio/util.py#L92) to modify get_systype() as it follows (overwrite arch with armv7l if userland/python binary is 32b):

```python
def get_systype():
    type_ = platform.system().lower()
    arch = platform.machine().lower()
    if type_ == "windows" and "x86" in arch:
        arch = "amd64" if "64" in arch else "x86"
    if arch == "aarch64":
        if platform.architecture()[0] == "32bit":
            arch = "armv7l"
    return "%s_%s" % (type_, arch) if arch else type_
```

With this patch, platformio works on RPi with both 64b kernel/64b userland and 64b kernel/32b userland.

quentinmit, 2022-03-21:
I hadn't even considered the case of a 32-bit userland on a 64-bit kernel. Indeed, that just makes `platform.machine` completely wrong.  Does running esphome with `linux32` not fix this for you (since it changes what `platform.machine` returns)?

jpcornil-git, 2022-03-21:
platformio is invoked from ESPHome (not CLI) in this context.

The above patch fix the (HomeAssistant/ESPHome on RPi/arm platforms) issue for now for people running 32-bit userspace on a 64-bit kernel when then can't easily migrate to a full 64b platform (avoid the issue altogether).

Wrt platformio, get_systype should probably return (OS/system, kernel/machine, userland/architecture) to enable proper decision upper up.

quentinmit, 2022-03-21:
> platformio is invoked from ESPHome (not CLI) in this context.

Yes, my workaround was to invoke all of ESPHome inside `linux32` - the uname information is inherited by child processes. That worked for me, can you try it? (Obviously proper multiarch support would still be better.)

jpcornil-git, 2022-03-22:
It works as expected but requires to modify one file (same as proposed patch) and force 32b userspace (proposed patch is not affecting true 64b kernel/userspace execution [on ARM here/should be generic ideally])

quentinmit, 2022-03-22:
> It works as expected but requires to modify one file (same as proposed patch) and force 32b userspace (proposed patch is not affecting true 64b kernel/userspace execution [on ARM here/should be generic ideally])

I don't understand what you mean by "modify one file" - you can just invoke esphome from the command line with linux32... `linux32 esphome upload foo.yaml`. No file editing required. If you're running esphome from Docker, you'll need to launch the Docker container with `linux32` in the commandline - e.g. `docker run --entrypoint "linux32 /entrypoint.sh dashboard /config" esphome/esphome`

I don't like your patch even as a workaround because I don't think it's necessarily the case that `armv7l` is the correct 32-bit architecture name to use. For example, the original RPi used `armv6l` or `armhf` and some RPi distros still target that. I'd much rather rely on some other part of the system to report the correct architecture name(s).

jpcornil-git, 2022-03-22:
WIth HomeAssistant OS, you are not supposed to use the command line but the dashboard (running within a container and started by HAOS), i.e. that linux32 prefix has to be instantiated somewhere (container init script, HAOS service script, ...) and I would rather fix this issue within platformio rather than expose/maintain the workaround upper up + not enforce 32b forever.

But I agree with you that the patch above is not a final implementation by far, just an example of how to fix it for a specific configuration + confirmation for HomeAssistant/esphome users reporting the issue (and maintainers) that issue is within platformio.

Most effective workaround for HAOS users today is to match kernel and userspace architectures, i.e. 32 or 64b for both

manyos-alex, 2022-03-30:
"Most effective workaround for HAOS users today is to match kernel and userspace architectures, i.e. 32 or 64b for both"

I'm running HAOS 64bit on raspberry 4 aarm64. 
But I'm getting the following error when i try to install esp32-s2 with the following setting:
esp32:
  board: featheresp32-s2
  framework:
    type: esp-idf

Error: Could not find the package with 'platformio/toolchain-esp32s2ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'

esphome: 2022.3.2
## System Health

version | core-2022.3.8
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.92-v8
arch | aarch64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4947
Installed Version | 1.24.0
Stage | running
Available Repositories | 1072
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 16. April 2022, 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.03.5
docker_version | 20.10.9
disk_total | 57.8 GB
disk_used | 4.1 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Mosquitto broker (6.0.1), Duck DNS (1.14.0), Terminal & SSH (9.3.0), ledfx (91682fa), ESPHome (2022.3.2)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 4
views | 10
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

Version | core-2022.3.8
-- | --
Installationstyp | Home Assistant OS
Entwicklung | false
Supervisor | true
Docker | true
Benutzer | root
Virtuelle Umgebung | false
Python-Version | 3.9.9
Betriebssystemfamilie | Linux
Betriebssystem-Version | 5.10.92-v8
CPU-Architektur | aarch64

Version | core-2022.3.8
-- | --
Installationstyp | Home Assistant OS
Entwicklung | false
Supervisor | true
Docker | true
Benutzer | root
Virtuelle Umgebung | false
Python-Version | 3.9.9
Betriebssystemfamilie | Linux
Betriebssystem-Version | 5.10.92-v8
CPU-Architektur | aarch64

jpcornil-git, 2022-03-30:
Hi @manyos-alex,

This is a different issue, from what I can see in [esp32s2 compatibility](https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32s2/compatibility), ther3 is no 64b support yet for a linux/arm platform.

You can always install EspHome on your PC if you want to keep 64b/mainstream build for HAOS on RPi

jpcornil-git, 2022-03-31:
There is definitively an issue in the way PlatformIO handles mixed kernel/userspace architectures

To select a package/toolchain PlatformIO is using platform.machine() which returns OS architecture and not userspace one (cfr. https://github.com/platformio/platformio-core/issues/4193#issuecomment-1073254509 above) , e.g. with a RPi running a 64b OS but a 32b userspace, it picks an incorrect "ELF64" toochain given 'aarch64' is returned (while platform.architecture() would give the correct 32b hint) -> binaries can't be executed as there is no underlying 64b library/service support on such platform.

ivankravets, 2022-04-01:
It seems that @quentinmit has provided solution in this comment https://github.com/platformio/platformio-core/issues/4193#issuecomment-1075629276

I don't see any other reporters on this issue. Let's keep it open for some time and discuss it later.