platformio/platformio-core #4199: Platformio --help or -h returns: Error: No such command "home"

guysoft, 2022-03-10:
What kind of issue is this?

- [ x ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Ubuntu 21.10

**PlatformIO Version** (`platformio --version`): ``PlatformIO Core, version 5.3.0a3`` . Also develop branch

Python version:
```
$ python3 --version
Python 3.9.7
```

### Description of problem
Platformio --help or -h returns: Error: No such command "home"

#### Steps to Reproduce

1. Install platformIO
```
sudo pip3 install platformio --upgrade
```
3. run ``platformio --help``

### Actual Results

```
$ platformio --help
Usage: platformio [OPTIONS] COMMAND [ARGS]...
Try 'platformio -h' for help.

Error: No such command "home"
```

or 
```
$ platformio -h
Usage: platformio [OPTIONS] COMMAND [ARGS]...
Try 'platformio -h' for help.

Error: No such command "home"
```

### Expected Results
Help me, please

Also, I tried digging in to the code and figuring out what to fix, but I can't find where stuff goes from: https://github.com/platformio/platformio-core/blob/develop/platformio/__main__.py#L36

ivankravets, 2022-03-10:
This is a known issue. How did you install PlatformIO? Do you use PlatformIO IDE for VSCode?

guysoft, 2022-03-10:
I initially installed via vscode.
Then I tried via atom.
Since I have the issue where when you click "open project" It just opens a file manager (dolphin) and not inside the ide. I resorted to installing it via pip.
I was just wondering if I should open an issue about that.

I found talks about that here:
https://community.platformio.org/t/stupid-question/4392
And here: #2523
(I tried to run ``pio init --ide atom`` or ``vscode`` and it does not solve the issue)

ivankravets, 2022-03-10:
1. Please don't use Atom, we plan to deprecate support for this editor
2. Please ensure that you use the latest version of VSCode and PlatformIO IDE extension for the VSCode.

This issue has been fixed in the latest version of the PlatformIO IDE extension for the VScode.

guysoft, 2022-03-10:
1. ok, will remove :)
2. I did upgrade to the latest vscode .
[image]

Should I upgrade anything else?

ivankravets, 2022-03-10:
What is your version of the PlatformIO IDE extension in the VSCode?

ivankravets, 2022-03-10:
I mean this extension https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide

guysoft, 2022-03-10:
v2.4.2

ivankravets, 2022-03-10:
Did you disable the built-in PlatformIO Core in the PlatformIO IDE extension settings? Please go to VSCode settings, then check that all PlatformIO settings are set to default values. Restart VSCode if you made changes.

guysoft, 2022-03-10:
Just went to the platformio vscode settings and hit "reset" on all 18 settings. Then restarted the editor. They didn't seem to change. Still has the issue :-/

ivankravets, 2022-03-10:
I know the solution to this issue. I just do not understand why the latest version of the extension didn't resolve it automatically. Maybe, you have multiple PlatformIO Core in the system.

Could you open the Ubuntu terminal WITHOUT VSCode? And type `pio --version`? Does this command work?

guysoft, 2022-03-10:
Ok, I think I have a hint. I had the pip version installed, which I removed. But I ALSO had a deb package installed.

I ran:
```
sudo apt-get remove --purge platformio
```
And it says:
```
Removing platformio (4.3.4-1) 
```
That is definatly an older verison.

Now I get:
```
$ pio
bash: /usr/bin/pio: No such file or directory
```

However it **still** opens the project in dolphin

ivankravets, 2022-03-10:
I see. So, please ensure that all global PlatformIO Cores are removed ( https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system ).

> Also tried: Moving ~/.platformio to ~/platformio.bak which triggered a re-install of the environment. And still it fails and opens dolphin and not the project

What do you mean under "opens dolphin"?

guysoft, 2022-03-10:
> What do you mean under "opens dolphin"?

When platformio pops the message``Project has been successfully configured and is located by this path`` It also pops up a window of dolphin, the default file manager in the system pointing to the location of one folder above the project folder.

ivankravets, 2022-03-10:
Please open PlatformIO Core (CLI), see the terminal button on the bottom status bar in VScode. Type 
```
pio system info
```

guysoft, 2022-03-10:
Does this Does this give us a hint? It seems like the terminal only sees the system-installed (or uninstalled) pio
[image]

ivankravets, 2022-03-10:
See https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli

P.S: I see that PlatformIO Home works for you. Can we close this issue?

guysoft, 2022-03-10:
Yes, --help works the built in terminal :)

Also now:
$ pio system info
--------------------------  -------------------------------------------------
PlatformIO Core             5.2.5
Python                      3.9.7-final.0
System Type                 linux_x86_64
Platform                    Linux-5.13.5-051305-generic-x86_64-with-glibc2.34
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/guy/.platformio
PlatformIO Core Executable  /home/guy/.platformio/penv/bin/platformio
Python Executable           /home/guy/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       3
Tools & Toolchains          11
--------------------------  -------------------------------------------------
```

guysoft, 2022-03-10:
The non-opening issue remains, should I open another issue?

guysoft, 2022-03-10:
Ok, when I open ``http://127.0.0.1:8008/`` and click on a project to open I get the exact same behavior that it opens the file manager - perhaps its an issue with chromium or filetypes?

ivankravets, 2022-03-10:
It looks like VSCode can not connect to PlatformIO Home server via Websockets. Please close VSCode and kill all Python processes. Does `pio home -p 8010` work for you? PlatformIO IDE for VSCode uses 8010 port by default. Maybe this port is already used by another process. You can change default port for PIO Home in PIO IDE/VSCode settings.