platformio/platformio-core #4200: ESP32 (CH9102X) not auto-detected

moussie369, 2022-03-12:
### Configuration

**Operating system**: Ubuntu 21.10, kernel is 5.13.0-1020-raspi

**PlatformIO Version** (`platformio --version`): 5.2.5

### Description of problem
ESP32 (CH9102X) is not present in the devices tab nor auto-detected with serial monitor/upload

#### Steps to Reproduce

Select serial monitor/upload option or refresh the connected devices list

### Actual Results

Only port ttyAMA0 is detected (regardless if the board present or not)

### Expected Results

Port ttyAMA0 and port ttyCH343USB0 detection

### Additional info

Drivers installed: https://github.com/WCHSoftGroup/ch343ser_linux
The board is registered as ttyCH343USB0 with ls /dev/tty* and I'm able to connect to it if I ad the upload/monitor_port lines in platformio.ini, looks like it's simply filtered/not detected

ivankravets, 2022-03-13:
Could you connect the CH9102X device and type `pio device list`?

moussie369, 2022-03-13:
It only returns:

/dev/ttyAMA0
------------
Hardware ID: fe201000.serial
Description: ttyAMA0