platformio/platformio-core #4201: Add JoC - Javaino board

clincks, 2022-03-17:
Hi,

- [ ] **Development Platform or Board**.
Is there a chance that https://www.joc.systems/en/hardware/javaino.html board could be added to the project?

Java is my main developing language and having a jvm on board is just amazing.

Thanks for your support

Kind regards

maxgerhardt, 2022-03-22:
The entire VSCode extenion and PlatformIO core is built around C/C++/ASM being the target language and mainly GCC being the compiler. I think adding Java would be significant work in how everything in PlatformIO works. (But on a basis, the to PlatformIO underlying [SCons](https://scons.org/doc/0.96.1/HTML/scons-user/c2590.html) build system supports Java). 

Where would one buy a board? They don't even sell it at https://www.saelig.com/miva/merchant.mvc?Screen=PROD&Product_code=javaino anymore, same on [amazon](https://www.amazon.de/Javaino-Arduino-kompatibles-Entwicklungsboard-Microcontroller/dp/B07VYMS9T9)...

ivankravets, 2022-03-22:
@clincks do you use Javaino board in production?

clincks, 2022-03-22:
> The entire VSCode extenion and PlatformIO core is built around C/C++/ASM being the target language and mainly GCC being the compiler. I think adding Java would be significant work in how everything in the PlatformIO core and extension works. (But on a basis, the to PlatformIO underlying [SCons](https://scons.org/doc/0.96.1/HTML/scons-user/c2590.html) build system supports Java).
> 
> Where would one buy a board? They don't even sell it at https://www.saelig.com/miva/merchant.mvc?Screen=PROD&Product_code=javaino anymore, same on [amazon](https://www.amazon.de/Javaino-Arduino-kompatibles-Entwicklungsboard-Microcontroller/dp/B07VYMS9T9)...

Hello,

Thanks for looking at my request.
I found this website where you can buy it: [JavaIno](https://www.saelig.com/miva/merchant.mvc?Screen=PROD&Product_code=javaino)

I already contacted an assembler (http://www.icd.lk/) to build an industrial board with same chip but with extra features like dubbel RJ45 (one for remote monitoring/configuration), one for the normal use, with a Oled Display, built-in relay, .... 
A mix of the best ideas of Controllino, ESP32, Tensy, ... with the possibility to have extension I/O modules (0/10V, 20mA, Relays, ...).

The processor is : SoC NXP LPC1768 but other processors could be used.

I'm much faster to develop in Java that in C or C++... and it will be much easier for me to integrate with existing Java Server application.

I'm seriously thinking to make a open hardware project with this... but I'm not an expert in electronic... and will need a lot of support. Also I don't know where to start.