platformio/platformio-core #4202: Regression: Access CMake with advance scripting python package installer

rashedtalukder, 2022-03-18:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows

**PlatformIO Version** (`platformio --version`): PlatformIO 5.2.5

### Description of problem
I have a script that downloads python packages with one package that needs to be compiled in C (cryptoauthlib). It used to use the CMake that is bundled in PIO but something changed in some version and that's not currently the case.

#### Steps to Reproduce

1. Clone https://github.com/m5stack/Core2-for-AWS-IoT-EduKit.git
2. Go to Blinky-Hello-World directory
3. `pio run -e core2foraws-device_reg -t register_thing`

### Actual Results
```
Building wheel for cryptoauthlib (setup.py): finished with status 'error'
  error: subprocess-exited-with-error
  
  Running setup.py clean for cryptoauthlib
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [33 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-3.10
      creating build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\atcab.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\atcacert.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\atcaenum.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\atjwt.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\device.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\exceptions.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\iface.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\library.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\sha206_api.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\status.py -> build\lib.win-amd64-3.10\cryptoauthlib
      copying cryptoauthlib\talib.py -> build\lib.win-amd64-3.10\cryptoauthlib
[... 134 lines cut ...]
board = m5stack-core2
build_unflags = -mfix-esp32-psram-cache-issue

; If PlatformIO does not auto-detect the port the device is virtually mounted to, 
; uncomment the line below to set the upload_port (remove the ";") and paste the
; copied port you identified from PIO Quick Access menu --> PIO Home --> Devices 
; after the equal sign. You do not need to add quotes 
;(e.g. upload_port = /dev/cu.SLAB_USBtoUART).
;
; upload_port = 

; Additional scripts
extra_scripts =
    register_thing.py ; Script to register the device to AWS IoT using available AWS CLI credentials

; Please visit documentation for the other PlatformIO
; configuration options and examples
; http://docs.platformio.org/page/projectconf.html
```

**Source file to reproduce issue:**
https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/blob/master/Blinky-Hello-World/register_thing.py
https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/blob/master/Blinky-Hello-World/utilities/AWS_IoT_registration_helper/requirements.txt

### Additional info

rashedtalukder, 2022-03-18:
Seems like this might be fixed now with the latest patch. Not sure how or why.

ivankravets, 2022-03-18:
The issue is not related to the PlatformIO. You use global Python 3.10 and there are no Python wheels (pre-compiled library) for `cryptoauthlib`. See https://pypi.org/project/cryptoauthlib/#files

Possible solutions:
1. Ask `cryptoauthlib` maintainers to publish to PyPi wheels for Python 3.10 => https://github.com/MicrochipTech/cryptoauthlib
2. Remove global Python 3.10, restart VSCode. The latest PlatformIO IDE extension will pull our pre-built Python 3.9.

Did it help?

ivankravets, 2022-03-18:
https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/blob/master/Blinky-Hello-World/register_thing.py#L37:L45

Please refactor this code according to https://docs.platformio.org/en/latest/scripting/examples/extra_python_packages.html

1. Use `env.Execute("$PYTHONEXE -m pip...`
2. Use `try/catch` to avoid unnecessary installation procedures if packages are already installed
3. It seems you don't need to install `cmake` package, this is not your issue, see my comment above.

rashedtalukder, 2022-03-18:
I see what you’re saying and this is helpful. I will make your suggested edits. 

I think it is working now in a fresh environment since the Python pre installation is now fixed. It’s using the bundled Python 3.x that is less than 3.10. 

Thank you Ivan.

ivankravets, 2022-03-19:
Please reopen if you still need help.

closed 2022-03-19 by ivankravets