platformio/platformio-core #4204: Multiple tests defined in platformio.ini not running on remote agent

sguarin, 2022-03-21:
### Configuration

**Operating system**: Fedora 34

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.2.5

### Description of problem

Multiple tests defined in platformio.ini not running on remote agent.

#### Steps to Reproduce

1. Having test_filter with mutiple tests in platformio.ini

for ex:
```
[env:esp32_prod_test]
platform = espressif32
board = esp32dev
framework = arduino
build_type = release
build_flags = ${config.build_flags}
lib_deps = ${config.lib_deps}
test_filter = test_SystemConfig, test_MySensors
```
2. Have an agent running on other hosts with board connected.
3. platformio remote test -e esp32_prod_test

Log shows that both tests are built and uploaded, but output corresponds only to the last processed test.
No matter the order in test_filter, seems they are processed in alphabetic order. 

### Actual Results

Output corresponds only to the last test processed.

### Expected Results

I was expecting to get different outputs for each test.

### Additional info

Log output
```
└─$ platformio remote test -e esp32_prod_test 
Building project locally
Verbose mode can be enabled via `-v, --verbose` option
Collected 13 items

Processing test_MySensors in esp32_prod_test environment
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
======================================================================================================== [PASSED] Took 4.41 seconds ========================================================================================================

Processing test_SystemConfig in esp32_prod_test environment
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Building...
======================================================================================================== [PASSED] Took 4.27 seconds ========================================================================================================
Testing project remotely
Verbose mode can be enabled via `-v, --verbose` option
Collected 13 items

Processing test_MySensors in esp32_prod_test environment
--------------------------------------------------------------------------------
Uploading...

Warning! Please install `99-platformio-udev.rules`.                                                                                                                                                                                          
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules                                                                                                                                                                
                                                                                                                                                                                                                                             
Testing...                                                                                                                                                                                                                                   
If you don't see any output for the first 10 secs, please reset board (press reset button)

ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
test/test_SystemConfig/test_SystemConfig.cpp:124:test_begin     [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:126:test_reset_config      [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:128:test_get_default_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:130:test_set_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:132:test_set_invalid_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:124:test_begin     [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:126:test_reset_config      [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:128:test_get_default_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:130:test_set_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:132:test_set_invalid_value [PASSED]
-----------------------
10 Tests 0 Failures 0 Ignored
========================== [PASSED] Took 9.02 seconds ==========================

Processing test_SystemConfig in esp32_prod_test environment
--------------------------------------------------------------------------------
Uploading...

Warning! Please install `99-platformio-udev.rules`.                                                                                                                                                                                          
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules                                                                                                                                                                
                                                                                                                                                                                                                                             
                                                                                                                                                                                                                                             
Testing...
If you don't see any output for the first 10 secs, please reset board (press reset button)

ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
test/test_SystemConfig/test_SystemConfig.cpp:124:test_begin     [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:126:test_reset_config      [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:128:test_get_default_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:130:test_set_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:132:test_set_invalid_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:124:test_begin     [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:126:test_reset_config      [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:128:test_get_default_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:130:test_set_value [PASSED]
test/test_SystemConfig/test_SystemConfig.cpp:132:test_set_invalid_value [PASSED]
-----------------------
10 Tests 0 Failures 0 Ignored
========================== [PASSED] Took 8.95 seconds ==========================

Test               Environment      Status    Duration
-----------------  ---------------  --------  ------------
test_MySensors     esp32_prod_test  PASSED    00:00:09.021
test_SystemConfig  esp32_prod_test  PASSED    00:00:08.948
========================= 2 succeeded in 00:00:17.969 =========================

```

ivankravets, 2022-04-28:
Thanks for the PR! Does the latest `pio upgrade --dev` work for you?

sguarin, 2022-04-28:
After some new dependencies installed I got into some conflict with my system installed cryptography library.
I'm using Fedora and I cannot remove it (because I need it).
```
ImportError: The version of cryptography does not match the loaded shared object. This can happen if you have multiple copies of cryptography installed in your Python path. Please try creating a new virtual environment to resolve this issue. Loaded python version: 36.0.1, shared object version: b'37.0.1'
```
So starting a fresh environment from a fresh container (with toolbox) got it almos working... now I'm having some unity issue:

```
/home/sguarino/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/sguarino/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/lib/libunity.a(unity.c.obj):/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/unity/unity/src/unity.c:56: multiple definition of `UnityStrErrFloat'; .pio/build/esp32_prod_test/libe9f/libUnity.a(unity.c.o):/home/sguarino/Gdrive/SMMR/SMMR/.pio/libdeps/esp32_prod_test/Unity/src/unity.c:63: first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32_prod_test/firmware.elf] Error 1
Building stage has failed, see errors above
```

Despite of those, the pio seems to iterate only on the filtered test.

ivankravets, 2022-05-07:
Duplicate of https://github.com/platformio/platformio-core/issues/3270

Please use `-r` flag for the `pio remote test` command.

closed 2022-05-07 by ivankravets