platformio/platformio-core #4206: `lib_extra_dirs = ...` no longer works with 5.3.0a3

mcspr, 2022-03-23:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:
Fedora 35 (Container Image)

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.3.0a3 

### Description of problem

When `lib_extra_dirs` directory already has a library installed, `pio run` will still do an installation into the `.pio/libdeps/${env}/`

(plus, in case of non-registry dependency, it would also mean it will fetch the git repo twice)

This was not the case when using the 5.2.x
Probably related to the #4198? 

#### Steps to Reproduce

1. Create a dummy project via `pio project init -b esp32dev`
2. Go into the project and manually install some lib
```
$ mkdir my_libraries
$ pio lib -d my_libraries/ install 'bblanchon/ArduinoJson@6.19.3'
Library Storage: /home/builder/esp32dev/my_libraries
Library Manager: Installing bblanchon/ArduinoJson @ 6.19.3
Unpacking  [####################################]  100%
Library Manager: ArduinoJson @ 6.19.3 has been installed!
```
3. Add both dependency and the `lib_extra_dirs` entry to the environment
```ini
lib_deps =
    bblanchon/ArduinoJson@6.19.3
lib_extra_dirs =
    my_libraries/
```
4. `pio run` 
```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing bblanchon/ArduinoJson @ 6.19.3
^C
Aborted!
^C⏎
```
### Expected Results

Don't install things twice and re-use the existing library storage.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
    bblanchon/ArduinoJson@6.19.3
lib_extra_dirs =
    my_libraries/
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

ivankravets, 2022-03-24:
I don't think this is a bug. Starting from Core 5.3, PlatformIO will resolve all environment dependencies before it starts any operations. It solves a lot of other issues. Maybe, it makes sense to deprecate `lib_extra_dirs` or to add notes to docs that this is bad practice to use `lib_extra_dirs`. The `pio lib` and `pio platform` commands will be hidden from CLI in the next patch release (5.3.1). We will maintain legacy CLI but will warn if someone tries to use it.

Have you tried the latest PlatformIO Core (`pio upgrade --dev`) and the new `pio pkg` command? I'll like it.A few demos from https://github.com/esphome/esphome

**List installed packages**

[image]

**List packages that depend on `ESPAsyncTCP-esphome`**

[image]

**Check for outdated packages**
[image]

mcspr, 2022-03-24:
My main reason for `lib_extra_dirs = ...` with the previous version is multiple environments that share the dependency tree
https://github.com/xoseperez/espurna/tree/dev/code/platformio.ini

For example, right now there are `esp8266-1m-base` building with latest released framework version and `esp8266-1m-git-base` building from a development branch. Pre extra script will call `pio lib -d libraries/ install $lib_deps contents$`, so by the time it gets to the dependency resolution the libraries are already there in the shared directory. Running once, it will skip the time expensive fetch & install phases and start building immediately.

I am not too familiar with esphome build process, but I do see that at least the Dockerfile is setting `PLATFORMIO_GLOBALLIB_DIR=/piolibs` to effectively achieve the same result.

What also comes to mind is the `go mod vendor` (https://go.dev/ref/mod#vendoring)
(which helps when distributing the source, storing the external library alongside the project files)

BTW I have tried the pkg command, but it only seems to work for already installed packages and libraries. I don't see ArduinoJson dependency unless there is .pio/libdeps/ArduinoJson (either from pio run or pio pkg install -e esp32dev).

ivankravets, 2022-03-25:
There is no sense to use global or extra library storage if you use PlatformIO Registry. PlatformIO caches downloaded packages and API requests. So, depending on the same packages in the different configuration environments will not lead to multiple package downloading.

If you don't want PlatformIO to manage your dependencies, then please don't use `lib_deps`. You can manage dependencies manually and later just instruct PlatformIO to use the libraries which you prepared in some storage.

How to automate this? I'll describe a simple solution but I don't recommend using it. This is not a good practice.

1. Do not use `lib_deps`
2. Let use this config https://github.com/xoseperez/espurna/blob/dev/code/platformio.ini
3. Add PRE script

```python
import os
Import(env)

config = env.GetProjectConfig()
lib_deps = config.get("common", "lib_deps")
# cleanup lib_deps, split into the list, etc...
lm = LibraryPackageManager("/path/to/your/common/storage")
for dep in lib_deps:
  if not lm.get_package(dep):
      lm.install(dep)
```

P.S: I didn't test it. Just give you some hints where to dig.

ivankravets, 2022-03-25:
@mcspr  , @xoseperez, what is the reason that you use Git instead of prepared packages? It takes time and requires Git to be installed on the user machine. Do you use any libraries that are not listed in the registry?

mcspr, 2022-03-25:
> There is no sense to use global or extra library storage if you use PlatformIO Registry. PlatformIO caches downloaded packages and API requests. So, depending on the same packages in the different configuration environments will not lead to multiple package downloading.

I *do* want to have an easy way to install such libraries, that was the whole reason to use PIO instead of some custom solution in the first place; plus, a simple way for the users to build it in a single step.
(and b/c even in the arduino-cli form, Arduino's official builder is still a no-go for anything larger than a single .ino file)

Why can't there be a local registry then?
Wouldn't there be a problem when offline and when the caching expires?
Can't the git blob be cached as well, btw? Perhaps solving the above, by packing the resulting source tree and attaching some metadata?
(...b/c I am feeling the idea is to have a per-env libdeps directory so we don't mix filesystem locations...)

> what is the reason that you use Git instead of prepared packages? It takes time and requires Git to be installed on the user machine. Do you use any libraries that are not listed in the registry?

Mostly historical reasons. And you can also see some of my own forks (nofuss) that aren't published, some libraries didn't have a correct published version at the time of writing or there was no release yet, some we don't have rights to distribute but technically can still allow the user to fetch them (bsec :/), and etc.

I could've done a builder timing test via https://github.com/mcspr/espurna-nightly-builder/ for multiple environments, but I would need to fix the above first for any real comparison.

ivankravets, 2022-03-26:
> Wouldn't there be a problem when offline and when the caching expires?

There is no 100% guarantee. You are affected in the same way when Github goes down.
Our package registry infrastructure has been working since September 2020 without any issues. Even if some mirrors go down, there are others. Users were not affected. 

See https://stats.uptimerobot.com/ZWBXVu5NGx
There are some "Down" issues but that is not our fault. The "uptime" service had "timeout issues" internally. See "Connection Timeout" in https://stats.uptimerobot.com/ZWBXVu5NGx/784792197

> Mostly historical reasons.

Let's together improve the developer experience. Using Github repositories as dependencies is not good practice.  How about ping authors of the libs that are outdated? Or, even to make your own copies under your own namespace? I saw [ESPHome](https://registry.platformio.org/search?q=owner%3Aesphome) did it already.

mcspr, 2022-03-27:
Yes, we could use the registry, but the core of the problem is inability to use a lib dir between environments, which still can be recognized by the package manager.

ESPurna builder doesn't really gain anything from the separation, so I'd still ask for at least some way to share lib_deps list between envs. Suppose, the ability to substitute `.pio/libdeps/$env` for a single directory via config or the extra script that could change some internal values.

Perhaps OT, but why it would not be a good practice to use a git repo? While we do depend on the Github, we also just depend on the protocol of git over https (which could also be hosted elsewhere).
Plus, installation of the git tool is not really a problem, as it would be required to install ESPurna in the first place.

ivankravets, 2022-03-29:
> which still can be recognized by the package manager.

You can manage global packages using `pio pkg --global` command.

> Perhaps OT, but why it would not be a good practice to use a git repo? 

1. Git is not installed by default on all machines. Users are affected.
2. It is hard to apply SemVer to Git. 
3. Git is slow, need to fetch files and metadata, multiple connections.

It is good for a single project, legacy workflow established 30 years ago. If we talk about collaborative development, strict dependency declaration using Semantic Versioning is the standard today for high-level languages (Java, Python, NodeJS, etc.)

mcspr, 2022-03-30:
> You can manage global packages using pio pkg --global command.

*`pio pkg install --global -l $spec`

The idea is to retain the project-global libraries storage, I would not feel comfortable with an advice to do that for user installs. Why mix user packages with what's only needed for our build?
(btw it won't use libraries from the right now, libraries are still installed to the `.pio/libdeps/$env` when running the environment)

We do agree that introducing extra steps or forcing to run / install any extra tools should be avoided. Right now ESPurna builds simply doing `pio run`, all the magic happens under the hood. We won't need to document any extra steps and I can simply track the PIO changes myself, and add or change the extra logic we need.

Right now we only lose the ability to optimize this specific library installation step... :/ Which we know is working for us (...or, worked, until the recent changes)

mcspr, 2022-03-31:
After poking around the libraries, packages and environment builder... A couple issues so far with the approach I proposed (...or the way I see it implemented)
- config as-is does not know about environments or what is the active one is, it's something we add after the fact. I wonder if config object could've interpolated something like `${internal.env}`, but that would mean it needed to be set every time
- config handling of libdeps_dir is mostly used as-is, which slightly complicates the changes where I'd want to substitute the project_env for something else

Brute-force approach just for the `pio run`, so now it finally uses the `libraries/` I have installed in the ESPurna's project dir (e.g. `~/dev/espurna/code/libraries` where `code/` is the project root with the platformio.ini), and after adding `[env] libdeps_env_dir = libraries/`
```diff
diff --git a/platformio/package/commands/install.py b/platformio/package/commands/install.py
index c8e525b4..b458e020 100644
--- a/platformio/package/commands/install.py
+++ b/platformio/package/commands/install.py
@@ -199,9 +199,13 @@ def _install_project_env_custom_tools(project_env, options):
 def _install_project_env_libraries(project_env, options):
     already_up_to_date = not options.get("force")
     config = ProjectConfig.get_instance()
-    lm = LibraryPackageManager(
-        os.path.join(config.get("platformio", "libdeps_dir"), project_env)
-    )
+
+    libdeps_dir = config.get(f"env:{project_env}", "libdeps_env_dir")
+    if not libdeps_dir:
+        libdeps_dir = os.path.join(config.get("platformio", "libdeps_dir"), project_env)
+
+    lm = LibraryPackageManager(libdeps_dir)
+
     if options.get("silent"):
         lm.set_log_level(logging.WARN)
     for library in config.get(f"env:{project_env}", "lib_deps"):
@@ -222,9 +226,13 @@ def _install_project_env_libraries(project_env, options):
 def _install_project_env_custom_libraries(project_env, options):
     already_up_to_date = not options.get("force")
     config = ProjectConfig.get_instance()
-    lm = LibraryPackageManager(
-        os.path.join(config.get("platformio", "libdeps_dir"), project_env)
-    )
+
+    libdeps_dir = config.get(f"env:{project_env}", "libdeps_env_dir")
+    if not libdeps_dir:
+        libdeps_dir = os.path.join(config.get("platformio", "libdeps_dir"), project_env)
+
+    lm = LibraryPackageManager(libdeps_dir)
+
     if not options.get("silent"):
         lm.set_log_level(logging.DEBUG)
     specs_to_save = []
diff --git a/platformio/project/options.py b/platformio/project/options.py
index ee5fd603..56ac9950 100644
--- a/platformio/project/options.py
+++ b/platformio/project/options.py
@@ -576,6 +576,17 @@ ProjectOptions = OrderedDict(
                 multiple=True,
                 sysenvvar="PLATFORMIO_LIB_EXTRA_DIRS",
             ),
+            ConfigEnvOption(
+                group="directory",
+                name="libdeps_env_dir",
+                description=(
+                    "Internal storage specific to the environment, where Library Manager will install project "
+                    "dependencies declared via `lib_deps` option."
+                ),
+                sysenvvar="PLATFORMIO_LIBDEPS_ENV_DIR",
+                default="",
+                validate=validate_dir,
+            ),
             ConfigEnvOption(
                 group="library",
                 name="lib_ldf_mode",                                                                                                                                                                                                                                                                        
```

ivankravets, 2022-03-31:
Sorry, PlatformIO is already complex in terms of managing libraries. That was our mistake with adding support for "extra dirs" and global libraries. Adding "yet another" storage logic for library dependencies will finally destroy all our efforts on improving the developer experience. The goal is to reduce this complexity and encourage people to use SemVer declarative approach. We've not yet found a better methodology to manage dependencies. 

I close this issue. If you don't want to use default storage and declarative approach with `lib_deps` and SemVer, please check this comment ( https://github.com/platformio/platformio-core/issues/4206#issuecomment-1078881377 ). This example allows you to do everything that you want with storage and packages.

Thanks for understading.

closed 2022-03-31 by ivankravets

mcspr, 2022-03-31:
Ok. `$PROJECT_DIR/lib` worked with the proposed script above, I'll update the project
(unless `lib/` is going to be removed as well at some point :)

Only cleanup I found required is this additional step that works like original 'multiple' config value of `lib_deps`
```diff
- lib_deps = config.get("common", "lib_deps")
+ lib_deps = config.parse_multi_values(config.get("common", "lib_deps"))
```

ivankravets, 2022-03-31:
Great that you found the solution.

We've completed work on `pio pkg` command. There is also rich documentation. Do you have any wishes?

mcspr, 2022-03-31:
I'll need to play around with it some more, as I was going to ask about search and turns out it is already there :)
Thanks for the continued support!

Jason2866, 2022-04-03:
Will `lib_extra_dirs` still be there in future? Since we (Tasmota) have heavily modified official libs for our needs (which are in Platformio Registry) and we do not want to upload this special versions. We do often changes in this adopted libs. It would add horrible administrative work to upload this changes. This modified libs are only useable with Tasmota (so no use for other projects). So we keep ALL our libs local, NO extern libs from anywhere. Not affected from any downtime of any infrastructure. All needed is there.
We use `lib_extra_dirs` to enable specific needed libs for different build variants.
So we can manage to keep compile short and incompatible libs are not seen for different build variants.

ivankravets, 2022-04-03:
> Will lib_extra_dirs still be there in future

We never remove any functionality that can break existing projects. This is our metrology that was introduced in 2014. Any project created in 2014 is fully compatible with the latest PlatformIO Core. We don't plan to change it. 

This issue was related to the Package Manager (`pio pkg`) command and `lib_deps`. We will encourage developers to use a declarative approach for critical safety embedded development based on `lib_deps` and Semantic versioning. See upcoming `pio pkg` command ( https://docs.platformio.org/en/latest/core/history.html ) and [specifications and version requirements](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html).

This approach does not depend on `lib_extra_dirs` and PlatformIO will never manage automatically (check updates, etc) custom user storage until they use `pio pkg --globa --storage-dir /path/to/extra-storage`.

We would be thankful if you [switch Tasmota](https://github.com/platformio/platform-espressif32/blob/develop/.github/workflows/examples.yml#L44) to the latest development version of PlatormIO Core and report any issues. So, we will be confident that the upcoming stable release will not break popular projects. We even added them to our docs ( https://docs.platformio.org/en/latest/tutorials/index.html#projects ).

Jason2866, 2022-04-03:
Thx, for your fast answer. I will change to develop Platformio version for our [special](https://github.com/Jason2866/Tasmota-build/blob/start_build/.github/workflows/Build_special_firmware.yml) versions builds. This are the biggest and most complex ones, so any issues are best seen there.

ivankravets, 2022-04-03:
Thanks, @Jason2866! 🙏