platformio/platformio-core #4216: how to solve it in China?

zsm666, 2022-04-07:
[image]

long ago,the get-plantformio.py was wrong,but i get PIO by anyway.
now,the plantformio_init is wrong now.i can't solve it.so sad...please help

ivankravets, 2022-04-07:
Thanks for contacting us! Do you mean that https://api.registry.platformio.org is blocked in China?

zsm666, 2022-04-07:
> 

yes,i think maybe.

zsm666, 2022-04-07:
> Thanks for contacting us! Do you mean that https://api.registry.platformio.org is blocked in China?
yes,i think maybe.

zsm666, 2022-04-07:
do you know how to manual download?

zsm666, 2022-04-07:
or do you have any mirror in China?

zsm666, 2022-04-07:
if my word rude,please forgive me.

zsm666, 2022-04-07:
i'm f**king solved it.I use my mobile phone to open the VPN and hotspot, and then use the computer to connect my phone.

ivankravets, 2022-04-07:
Could you try to download this file - what is your speed?

https://dl.registry.platformio.org/download/platformio/tool/toolchain-xtensa32/2.50200.97/toolchain-xtensa32-windows-2.50200.97.tar.gz

zsm666, 2022-04-07:
> Could you try to download this file - what is your speed?
> 
> https://dl.registry.platformio.org/download/platformio/tool/toolchain-xtensa32/2.50200.97/toolchain-xtensa32-windows-2.50200.97.tar.gz

[image]
without VPN

and
[image]
with VPN

emmm..what about this?

ivankravets, 2022-04-07:
@Gianthard-cyh is PlatformIO blocked in China? Do you have access to our registry?

zsm666, 2022-04-07:
[image]
it's ok now.i using chinese to tell other people in China how to solve it.

用手机开VPN和热点,然后拿电脑连接手机热点即可.

ivankravets, 2022-04-07:
Thanks for the solution. Please do not hesitate to contact us if still have problems with our registry.

Happy coding with PlatformIO!

closed 2022-04-07 by ivankravets

zsm666, 2022-04-07:
> Thanks for the solution. Please do not hesitate to contact us if still have problems with our registry.
> 
> Happy coding with PlatformIO!

Thanks for your previous help!! :)

dennisdebel, 2025-02-10:
What VPN can be forwarded through a phone's Hot Spot?