platformio/platformio-core #4219: Support for SonarLint in Visual Studio Code

Insomniac80, 2022-04-09:
SonarLint has recently added support for C and C++ when using CMake.

[https://community.sonarsource.com/t/sonarlint-for-visual-studio-code-3-4-released-support-for-c-and-c/61110](url)

Being able to use SonarLint with PlatformIO would be awesome.

zeroisnan, 2022-04-27:
All you need to get this to work is to generate a compilation database from PlatformIO.

- install the plugin in VSCode
- PlatformIO > Advanced > Compilation Database
- Shift + Ctrl + P (or your platform equivalent): > SonarLint: Configure the compilation database for C and C++ analysis - it should list the file generate from the step above

Done!

Insomniac80, 2022-04-27:
> All you need to get this to work is to generate a compilation database from PlatformIO.
>
> * install the plugin in VSCode
> * PlatformIO > Advanced > Compilation Database
> * Shift + Ctrl + P (or your [...]

Was the first I have tried. Sadly, I do not get any hints from SonarLint as f.e. in a Java project. My suggestion was there is something missing "under the hood".

Have you tried your solution, and been able to see hints from SonarLint in your code?

zeroisnan, 2022-04-28:
Yes this is working on my end, VSCode 1.66.2 (stable) on Windows + WSL2 + PIO 5.2.5
I get inline suggestions, 'Problems' panel populated, links to rules explanation, etc
I asked SonarLint to install and use a local JRE if that makes any difference

You should see something like this added to `.vscode/settings.json`:

``` json
"sonarlint.pathToCompileCommands": "/path/to/your/workspace/.pio/build/<target>/compile_commands.json"
```

Insomniac80, 2022-04-30:
Finally I had time to take a deeper look (VSCode 1.66.2, Windows 10, PIO 5.2.5, Java 15.0.2).

I created a brand new project: board Lolin32, Framework Arduino. Kept the mainly empty main.cpp (include Arduino.h makes no difference). Created and added compile_commands.json using PIO -> Advanced -> ..

simple SonarLint log reads:

> [Info  - 14:47:35.678] Analyzing file 'file:///c:/Users/Inso/Documents/PlatformIO/Projects/Text_SonarLint/src/main.cpp'...
[Info  - 14:47:35.684] Index files
[Info  - 14:47:35.685] 1 file indexed
[Error - 14:47:35.694] Error executing sensor: 'CFamily'
[Error - 14:47:35.694] java.lang.IllegalStateException: unexpanded response file
	at com.sonar.cpp.analyzer.ClangDriver.onCapture(ClangDriver.java:329)
	at com.sonar.cpp.analyzer.CompilerDrivers.onCapture(CompilerDrivers.java:35)
	at com.sonar.cpp.plugin.SonarLintSensor.lambda$processCompileCommands$1(SonarLintSensor.java:151)
	at java.base/java.util.HashMap.forEach(HashMap.java:1425)
	at com.sonar.cpp.plugin.SonarLintSensor.processCompileCommands(SonarLintSensor.java:145)
	at com.sonar.cpp.plugin.SonarLintSensor.process(SonarLintSensor.java:131)
	at com.sonar.cpp.plugin.SonarLintSensor.execute(SonarLintSensor.java:94)
	at com.sonar.cpp.plugin.CFamilySensor.execute(CFamilySensor.java:169)
	at org.sonarsource.sonarlint.core.analysis.container.analysis.sensor.SensorsExecutor.executeSensor(SensorsExecutor.java:81)
	at org.sonarsource.sonarlint.core.analysis.container.analysis.sensor.SensorsExecutor.execute(SensorsExecutor.java:72)
	at org.sonarsource.sonarlint.core.analysis.container.analysis.AnalysisContainer.doAfterStart(AnalysisContainer.java:122)
	at org.sonarsource.sonarlint.core.plugin.commons.pico.ComponentContainer.startComponents(ComponentContainer.java:119)
	at org.sonarsource.sonarlint.core.plugin.commons.pico.ComponentContainer.execute(ComponentContainer.java:104)
	at org.sonarsource.sonarlint.core.analysis.container.module.ModuleContainer.analyze(ModuleContainer.java:71)
	at org.sonarsource.sonarlint.core.analysis.command.AnalyzeCommand.execute(AnalyzeCommand.java:58)
	at org.sonarsource.sonarlint.core.analysis.command.AnalyzeCommand.execute(AnalyzeCommand.java:32)
	at org.sonarsource.sonarlint.core.analysis.AnalysisEngine$AsyncCommand.execute(AnalysisEngine.java:152)
	at org.sonarsource.sonarlint.core.analysis.AnalysisEngine.executeQueuedCommands(AnalysisEngine.java:70)
	at java.base/java.lang.Thread.run(Thread.java:832)

[Info  - 14:47:35.697] Found 0 issues

verbose log and compilation_commands.json are attached. If you add a project using the same parameters (board, framework), runs it fine for you?

[SonarLint verbose log output.txt](https://github.com/platformio/platformio-core/files/8596939/SonarLint.verbose.log.output.txt)
[compile_commands.json.txt](https://github.com/platformio/platformio-core/files/8596942/compile_commands.json.txt)

zeroisnan, 2022-05-01:
I tried with your exact board/framework and it worked fine. I added some simple code to trigger lint issues and could see them reported ok by SonarLint. This looks like a Java setup issue on your end, have you tried re-installing SonarLint and requesting the plugin to use its own private JRE?

I don't think PlatformIO folks can do or should do much here. I think it could be nice if they provided some option to automatically regenerate the compilation database when changes to the codebase are detected (files added or removed, changes to include statements).

Insomniac80, 2022-06-19:
I have discussed this in [the SonarLint forums](https://community.sonarsource.com/t/error-executing-sensor-cfamily-with-java-lang-illegalstateexception-unexpanded-response-file/62822/3)
and this was the result from one of the devs:

> I would better check [...] why temporary response files end up in the compilation database. It is not a good practice that temporary response files appear in the compilation database as those files may not be available.

ivankravets, 2022-06-20:
The [Compilation database](https://docs.platformio.org/en/latest/integration/compile_commands.html) is generated at the root of the project starting from PlatformIO Core 6.0.

@Insomniac80, thanks for the report. The bug is fixed in the latest development version of PlatformIO Core. Please re-test with `pio upgrade --dev`. Does it work for you now?

jbtronics, 2022-06-29:
@ivankravets With the development version the compile_commands.json can be parsed correctly by SonarLint in my project.
But I  running into the same issue @Insomniac80 described in his thread with an "IO failure on output stream: invalid argument" error in sonarlint. But i guess thats most likely an issue in sonarlint not in PlatformIO.

Insomniac80, 2022-06-29:
Have not updated here until I have a clue, actually discussing the problem in the sonar forum thread linked above:

[https://community.sonarsource.com/t/error-executing-sensor-cfamily-with-java-lang-illegalstateexception-unexpanded-response-file/62822/6](url)

dzid26, 2022-08-19:
Everything seems to work.
I would like the `compile_commands.json` to be created automatically in the same way as `c_cpp_properties.json` is.

Insomniac80, 2022-09-10:
Update: with the actual version of SonarLint (3.9.0) and PlatformIO (Core 6.1.4) it works:

If SonarLint asks for configuration first create a compilation database in PlatformIO (Project Tasks -> Advanced -> Compilation Database), then choose "configure" within the SonarLint configuration prompt. It should find your file and be enabled from now on.

Therefore I will close this issue. @dzid26 maybe add your suggestion to the forum?

closed 2022-09-10 by Insomniac80