platformio/platformio-core #4220: Path for extra script inside a library

FernandoGarcia, 2022-04-09:
Hi!

I was asking on [forum](https://community.platformio.org/t/path-for-extra-script-inside-a-library) about the right path to execute a script inside a library.

At first time I was using:

```
extra_scripts = 
 pre:.pio\libdeps\esp32dev\esp8266-react\scripts\build_interface.py
```

But changing the `ENV` brakes the path. So Max suggested me to use `$PROJECT_LIBDEPS_DIR\$PIOENV\esp8266-react\scripts\build_interface.py` but while PIO is showing the right path on terminal it gives an error of wrong path.

[image]

Then Max [asked](https://community.platformio.org/t/path-for-extra-script-inside-a-library/26964/10?u=fernando) me to open this issue.

 Best regards.

ivankravets, 2022-04-10:
Could you provide a simple project to reproduce this issue?

FernandoGarcia, 2022-04-10:
Hi!

Try this:

```cpp
#include <Arduino.h>
void setup(){}
void loop(){}
```

```ini
[env:esp32dev]
platform = espressif32
framework = arduino
board = esp32dev
monitor_speed = 115200
lib_deps = 
	maakbaas/ESP8266 IoT Framework
extra_scripts = 
	pre:$PROJECT_LIBDEPS_DIR\$PIOENV/ESP8266 IoT Framework/scripts/preBuild.py
```

Best regards.

ivankravets, 2022-04-10:
Thanks for the report! Fixed in the latest dev-version. Please re-test with `pio upgrade --dev`.

closed 2022-04-10 by ivankravets

FernandoGarcia, 2022-04-10:
Yes, it's working now.
Thank you!

commit 0501d55c (2022-04-10): Fixed an issue with calling an extra script located outside a project // Resolve #4220