platformio/platformio-core #4221: Build flags for library.json not working

paulocsanz, 2022-04-10:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 11 + WSL2

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.2.5

### Description of problem

Library's library.json specifies some build flags here: https://github.com/internet-of-plants/iop/blob/95f1a92cf5c1a30722cb64479b812afe4d8f3358/library.json#L30

```
  "build": {
    "flags": [
      "-D ARDUINOJSON_ENABLE_ARDUINO_STRING=0",
       ...
    ],
    ...
  }
```
And then checks if the flags are set here: https://github.com/internet-of-plants/iop/blob/95f1a92cf5c1a30722cb64479b812afe4d8f3358/include/iop/api.hpp#L7

```
#ifndef ARDUINOJSON_ENABLE_ARDUINO_STRING
#error "Must define ARDUINOJSON disable"
#endif
```

But if this repo's posix target won't compile: https://github.com/internet-of-plants/example-firmware/blob/12355ccfe33536dc9a1fb6e276d58dbd2811a93a/platformio.ini#L54

If you add the section below to the posix target, like the other targets do, it works (you can test the same problem with the other targets, removing the defines below will break the compilation. But shouldn't, as the library should have that definition).

```
-D ARDUINOJSON_ENABLE_ARDUINO_STRING=0
-D ARDUINOJSON_ENABLE_ARDUINO_STREAM=0
-D ARDUINOJSON_ENABLE_ARDUINO_PRINT=0
-D ARDUINOJSON_ENABLE_PROGMEM=0
```

A similar issue happens with the `-std=c++17` flag that the posix target sets, without it `iop` fails to compile, even when `iop` defines it. And I'm not sure what is going on. It might be a misunderstanding of mine, but I wasn't able to find any documentation on what is the appropriate behavior.

I would expect the library to have the defined flags that it needs to compile by itself.

#### Steps to Reproduce

1. Clone this repo https://github.com/internet-of-plants/example-firmware/tree/12355ccfe33536dc9a1fb6e276d58dbd2811a93a
2. Run `pio run --environment posix`
3. See the compilation error when you shouldn't

It can be reproduced for other platforms, but they will have deeper errors, as not having this flag breaks our code even without the `#error`

### Actual Results

```
In file included from .pio/libdeps/posix/iop/include/iop/loop.hpp:3,
                 from .pio/libdeps/posix/factory_reset_button/src/factory_reset_button.hpp:5,
                 from .pio/libdeps/posix/factory_reset_button/src/factory_reset_button.cpp:1:
.pio/libdeps/posix/iop/include/iop/api.hpp:8:2: error: #error "Must define ARDUINOJSON disable"
    8 | #error "Must define ARDUINOJSON disable"
      |  ^~~~~
In file included from .pio/libdeps/posix/iop/include/iop/loop.hpp:3,
                 from src/main.cpp:1:
.pio/libdeps/posix/iop/include/iop/api.hpp:8:2: error: #error "Must define ARDUINOJSON disable"
    8 | #error "Must define ARDUINOJSON disable"
      |  ^~~~~
*** [.pio/build/posix/lib7a2/factory_reset_button/factory_reset_button.o] Error 1
*** [.pio/build/posix/src/main.o] Error 1
```

### Expected Results

```
Linking .pio/build/posix/program
/usr/bin/ld: .pio/build/posix/lib1b3/iop-hal/upgrade.o: in function `iop_hal::Upgrade::run(iop::Network const&, iop::StaticString, std::basic_string_view<char, std::char_traits<char> >)':
/home/no/coding/internet-of-plants/example-firmware/.pio/libdeps/posix/iop-hal/src/cpp17/upgrade.hpp:21: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
================================================================================ [SUCCESS] Took 2.93 seconds ================================================================================

Environment    Status    Duration
-------------  --------  ------------
posix          SUCCESS   00:00:02.926
================================================================================ succeeded in 00:00:02.926 ================================================================================
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:posix]
platform = native
build_type = debug
lib_deps = 
	https://github.com/internet-of-plants/dht
	https://github.com/internet-of-plants/dallas_temperature
	https://github.com/internet-of-plants/soil_resistivity
	https://github.com/internet-of-plants/factory_reset_button
	https://github.com/internet-of-plants/iop
build_flags = 
	-std=c++17
	-Wconversion
	-Wall
	-Wextra
	-Wconversion
	-Wpedantic
	-pedantic-errors
	-fstack-protector
	-Wstack-protector

	-D IOP_POSIX
	-D IOP_LOG_LEVEL=iop::LogLevel::INFO"
```

### Additional info

ivankravets, 2022-04-10:
'library.json` flags are isolated per a library?

paulocsanz, 2022-04-10:
Sorry, I'm not sure I understand the question.

ivankravets, 2022-04-10:
Do you expect the macro defined in `library.json` outside its library?

ivankravets, 2022-04-10:
It seems like a duplicate of https://github.com/platformio/platformio-core/issues/1941 ?

paulocsanz, 2022-04-10:
No, I don't expect them to be defined outside of this library.

Maybe it would be a nice feature, but my problem is that it's not available _inside_ of the library.

`iop` library defines those in library.json, and then `iop` itself tries to use it, and it fails.

ivankravets, 2022-04-10:
Could you provide a simple "hello world" project to reproduce this issue? It will save my time digging into your project. Thanks.

paulocsanz, 2022-04-10:
Ok, will try!

paulocsanz, 2022-04-10:
Done, this won't compile, with the same error (inside of `lib-poc`): https://github.com/internet-of-plants/proj-poc/tree/1fd28fb8f36bb42ea6743abf4d01602f1ddfe495

If you uncomment the definitions in platformio.ini it works. But the project shouldn't have to define, the library already does and nobody outside of the projects accesses those definitions.

paulocsanz, 2022-04-11:
Do you have any idea of what is going on? Is this a bug? Is there a work-around? Is it actually a feature and I'm misguided?

Because I've seen other projects doing similar things without issues, so I assumed there would be an easy workaround?

ivankravets, 2022-04-13:
This is what I asked above. You are going to export flags to the global project scope. The "flags" are passed by default to the source files of the library (not includes). See example in https://docs.platformio.org/en/latest/librarymanager/config.html#extrascript

```python
global_env = DefaultEnvironment()
global_env.Append(
    CPPDEFINES=[
        ("ARDUINOJSON_ENABLE_ARDUINO_STRING", 1),
    ]
)
```

Datu-Puti, 2023-11-20:
I'm having trouble with this.  My library source files (.cpp files located in .\src\ relative to the library.json file) don't get the symbol defined via library.json however it does work if globally defined using the platformio.ini file.  
Excerpt of my JSON:
`"build": {
    "flags": "-D CGATE_VER=1"
  },`
  
  Is there something wrong with this format?

ivankravets, 2023-11-23:
@Datu-Puti, please ensure that your JSON file is valid using https://jsonformatter.curiousconcept.com/#