platformio/platformio-core #4225: Building Arduino, espidf projects broken since core v5.3.0

marcovannoord, 2022-04-13:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

Windows 10
```
--------------------------  ------------------------------------------------------
PlatformIO Core             5.3.0b5
Python                      3.9.10-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Marco\.platformio
PlatformIO Core Executable  C:\Users\Marco\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\Marco\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          11
--------------------------  ------------------------------------------------------
```

### Description of problem
Since the PlatformIO has upgraded to v5.3.0b5, i am unable to compile any projects that use the espidf platform together with Arduino.

#### Steps to Reproduce

1. Start with any empty project with above platformio.ini
2. Press `build` 
3. Observe an error in the build.

### Actual Results
```
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
AttributeError: 'PlatformIOLibBuilder' object has no attribute 'dependent':
  File "C:\Users\Marco\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Marco\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Marco\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Marco\.platformio\platforms\espressif32@3.3.0\builder\main.py", line 228:
    target_elf = env.BuildProgram()
  File "C:\Users\Marco\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Marco\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\Marco\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Marco\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 124:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Marco\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Marco\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 346:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\Marco\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 660:
    return method(*args, **kw)
  File "C:\Users\Marco\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Marco\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Marco\.platformio\platforms\espressif32@3.3.0\builder\frameworks\espidf.py", line 1221:
    project_lib_includes = get_project_lib_includes(env)
  File "C:\Users\Marco\.platformio\platforms\espressif32@3.3.0\builder\frameworks\espidf.py", line 95:
    if not lb.dependent:
```

### Expected Results
To have the project successfully built. This has been working for about a year without an issue now. 

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
; Set the default build target here, Comment out to compile for all environments.
default_envs = esp32

[env:esp32] ; ESP32 board
platform = espressif32@3.3.0
board = esp-wrover-kit
framework = arduino, espidf
platform_packages =  framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0 
```

### Additional info

Steps i have taken to try to fix the issue:
- Remove and reinstall Platformio IDE from VS-code
- Delete the .platformio folder and do a fresh install

Is there any way to downgrade the platformio core to 5.2.5 so i can get up and running again?

ivankravets, 2022-04-13:
Sorry for the issue. Please re-run `pio upgrade --dev`. Does it work now?

closed 2022-04-13 by ivankravets

marcovannoord, 2022-04-13:
Awesome, it looks like you've fixed it!

commit 5515bef3 (2022-04-13): Add backward compatibility with ESP-IDF build script // Resolve #4225