platformio/platformio-core #4229: Allow for pyserial serial URL's when unit testing (test_port)

timkoers, 2022-04-15:
What kind of issue is this?

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 11

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.2.5

### Description of problem
I'd like to request that the test_port behaves the same as the upload_port and monitor_port. You can put in an custom URL, with a socket connection in all but the test_port. I'd like to have the test_port behave the same.

ivankravets, 2022-05-02:
> You can put in an custom URL, with a socket connection in all but the test_port.

What is your use case for the socket connection? We are working on the new unit testing engine and new documentation. It would be great to describe this case as an example of the custom runner that is a part of PlatformIO Core 6.0.

ivankravets, 2022-05-03:
Thanks for the feature request. Please re-test with `pio upgrade --dev`. See updated docs https://docs.platformio.org/en/latest/projectconf/section_env_test.html#test-port

Does it work for you?

timkoers, 2022-05-06:
Thanks! Well In my case it would allow me to be able to run unit tests from within an qemu emulator, of which it's serial output is connected via sockets.

ivankravets, 2022-05-06:
No need to use sockets here. Please use the custom testing command. See docs https://docs.platformio.org/en/latest/advanced/unit-testing/simulators/qemu.html

timkoers, 2022-05-21:
I can't get the test command to be executed.
It states no such file or directory whilst executing
`[Errno 2] No such file or directory: '/usr/local/bin/qemu-system-xtensa -nographic -M esp32 -m 4 -kernel /builds/timkoers/xx/Sources/ESP32/.pio/build/firmware.elf -s -serial stdio -global driver=timer.esp32.timg,property=wdt_disable,value=true'`

ivankravets, 2022-05-21:
Each argument in the new line?

timkoers, 2022-05-21:
That seems to have at least started the test. I'm getting a hang up error right now, how do I enable the test output?

ivankravets, 2022-05-21:
Does it work without PlatformIO if you run in a separate command shell?

ivankravets, 2022-05-23:
You can create a Python script that does multiple operations and prints output to the stdout. Later, use this script as a broker.

ivankravets, 2022-05-23:
I forgot. If these commands are part of upload process, you take a look at the advanced scripting (see docs), and leverage AddPost method for upload target. 

In other words, you have multiple options. PlatformIO Unit Testing just needs the raw output from the testing program.

commit 1a441905 (2022-05-03): Added support for "socket://" and "rfc2217://" protocols using "test_port" option // Resolve #4229
closed 2022-05-23 by ivankravets