platformio/platformio-core #4231: Native: PlatformIO reports test as "Failed" even when all passed

Di-Strix, 2022-04-16:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.04.4 LTS

**PlatformIO Version** (`platformio --version`): 5.2.5

### Description of problem
When running tests on native platform, PlatformIO marks some tests as "Failed" even when unity reports that all tests have been passed successfully and prints final message "OK". This appears only on native platform. The interesting thing about it is if test-run order is shuffled then everything is ok.
[image]
[image]

#### Steps to Reproduce

1. Clone repo: https://github.com/Di-Strix/PIOTestFailDemo
2. Run test natively using `pio test -e native`
3. Get "OK" from unity and "Fail" from PlatformIO
4. Swap test-run order so that 'test_injection_empty_file' runs first and then 'test_injection_url'
5. Run test again using `pio test -e native`
6. Get "OK" from unity and "Passed" from PlatformIO

ivankravets, 2022-04-16:
Could you try `pio upgrade --dev`? It should be fixed in the latest PlatformIO Core 6.0.

Di-Strix, 2022-04-16:
The same issue. Also after updating platformio I had to remove library dependencies because platformio tried to install them

ivankravets, 2022-04-16:
> I had to remove library dependencies because platformio tried

Could you provide more details? Do you mean you don't need them? What was your previous version of `platformio.ini`?

Di-Strix, 2022-04-16:
`platformio.ini` remained the same. In test project I have library located in lib folder and it have ArduinoJson and ESPAsyncWebServer as dependencies. In current release version of platformio these libraries were not installed if they aren't exist. I am not sure if they should be checked if `lib_compat_mode = off`

<details>
<summary>library.json: </summary>

```json
  "name": "Arduino Materialize Layout",
  "version": "2.0.0",
  "description": "With this library you can quickly and easily generate web pages powered by Materialize. Library allows you to listen for button click, input events and dynamically update texts on the page",
  "keywords": "materialize, layout, webpage, web page, generate, generate web page, material design",
  "repository":
  {
    "type": "git",
    "url": "https://github.com/Di-Strix/ArduinoMaterializeLayout"
  },
  "authors":
  [
    {
      "name": "Andrew Ushakov",
      "url": "https://github.com/Di-Strix"
    }
  ],
  "license": "MIT",
-  "dependencies": {
-    "bblanchon/ArduinoJson": "^6.17.2",
-    "me-no-dev/ESP Async WebServer": "^1.2.3"
-  },
  "frameworks": "Arduino",
  "platforms": "espressif8266"
```

</details>
<details>
<summary>platformio.ini:</summary>

```ini
[env]
build_type = debug

[env:test_d1_mini]
framework = arduino
platform = espressif8266
board = d1_mini
upload_speed = 921600
monitor_speed = 115200
build_flags =
  -I $PROJECT_DIR/lib/ArduinoMaterializeLayout/src/Mocks/ESPAsyncWebServer

[env:native]
platform = native
lib_compat_mode = off
build_flags = 
  -std=c++17
  -DARDUINOJSON_ENABLE_ARDUINO_STRING
  -DUNIT_TEST_NATIVE
  -I $PROJECT_DIR/lib/ArduinoMaterializeLayout/src/Mocks/Arduino
  -I $PROJECT_DIR/lib/ArduinoMaterializeLayout/src/Mocks/ESPAsyncWebServer
```
</details>

ivankravets, 2022-04-16:
That was the bug in PlatformIO Core 5.0, see https://github.com/platformio/platformio-core/issues/2910

PlatformIO Core MUST resolve all dependencies.

Di-Strix, 2022-04-16:
For the production these library are needed, but for testing they aren't since I have my own mocks for them. So that bug was a feature :)
Is there a way to disable this behavior? Since it is needed for class mocking.

ivankravets, 2022-04-16:
I see. So, there will be the cases where people need `lib_deps` using on-target testing and may need to disable the for "native" testing.

Could you try [lib_ignore](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore)? You can create a separate configuration "env" which extends "release" env, and skip unwanted libs.

Di-Strix, 2022-04-16:
It works just fine. Thanks

Di-Strix, 2022-04-18:
I have found that for some reason `p.returncode` (in `platformio/proc.py:117`) is set to -11. Whereas the `program` exits with code 0
[image]
Could be this issue caused by that fact that I'm running all the stuff in the virtual machine? I'm using VMware® Workstation 16 Player 16.2.3 build-19376536. I will try installing the Ubuntu on my pc and reproduce the same steps.

ivankravets, 2022-04-18:
Thanks for the update! We work on the new Unit Testing engine. It will be a part of PlatformIO Core 6.0. The first beta version will be available next week. This issue will be fixed.

Di-Strix, 2022-04-18:
Wow! That's sound promising. I will be glad to test it out asap!

ivankravets, 2022-04-21:
Could you reproduce this issue with the latest PO Core? Please upgrade to the dev version using `pio upgrade --dev`.

ivankravets, 2022-04-21:
Please reopen if you still need help

closed 2022-04-21 by ivankravets

Di-Strix, 2022-04-21:
Hello!
The problem is the same. But now I'm getting SIGSEGV message.
[image]
Interesting that test-swap still fixes this. As well as GDB still reports that program exits normally. 
It must be a problem with my code, but the behavior is really strange...

ivankravets, 2022-04-21:
Could you add `return 0` to the `Int main()`?

Di-Strix, 2022-04-21:
It is already there. 
<details>
<summary>src/Mocks/Arduino/Arduino.cpp</summary>

```cpp
#if UNIT_TEST && UNIT_TEST_NATIVE

#include "Arduino.h"

uint32_t millis()
{
    return std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::system_clock::now().time_since_epoch()).count();
}

void delay(unsigned long t) {
    std::this_thread::sleep_for(std::chrono::milliseconds(t));
}

int main()
{
    setup();

    return 0;
}

#endif
```
</details>

mcspr, 2022-04-21:
`SIGSEGV` looks like a user error. PIO does not show all of the debug prints, but it says what the problem is when running directly. You just need to fix memory management in the webserver code & `unregisterHandlers` func
```gdb
(gdb) run
Starting program: /home/runner/dev/pio4231/.pio/build/native/program
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
test/MaterializeLayout/TestMaterializeLayout.cpp:54:test_injection_url:PASS
test/MaterializeLayout/TestMaterializeLayout.cpp:55:test_injection_empty_file:PASS

-----------------------
2 Tests 0 Failures 0 Ignored
OK
free(): invalid pointer <<<<<<<<<<<<<<<<

Program received signal SIGABRT, Aborted.
0x00007ffff7b14aec in __pthread_kill_implementation () from /lib64/libc.so.6
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.35-5.fc36.x86_64 libgcc-12.0.1-0.16.fc36.x86_64 libstdc++-12.0.1-0.16.fc36.x86_64

(gdb) bt
#0  0x00007ffff7b14aec in __pthread_kill_implementation () from /lib64/libc.so.6
#1  0x00007ffff7ac49c6 in raise () from /lib64/libc.so.6
#2  0x00007ffff7aae82f in abort () from /lib64/libc.so.6
#3  0x00007ffff7b08d8e in __libc_message () from /lib64/libc.so.6
#4  0x00007ffff7b1e70c in malloc_printerr () from /lib64/libc.so.6
#5  0x00007ffff7b2058c in _int_free () from /lib64/libc.so.6
#6  0x00007ffff7b22ed3 in free () from /lib64/libc.so.6
#7  0x0000000000402aba in AsyncWebServer::removeHandler (this=<optimized out>, h=<optimized out>)
    at lib/ArduinoMaterializeLayout/src/Mocks/ESPAsyncWebServer/ESPAsyncWebServer.h:144
#8  0x00000000004043d4 in MaterializeLayout::unregisterHandlers (this=this@entry=0x4101a0 <MLPage>) at lib/ArduinoMaterializeLayout/src/MaterializeLayout.cpp:39
#9  0x0000000000404605 in MaterializeLayout::~MaterializeLayout (this=0x4101a0 <MLPage>, __in_chrg=<optimized out>)
    at lib/ArduinoMaterializeLayout/src/MaterializeLayout.cpp:19
#10 0x00007ffff7ac7065 in __run_exit_handlers () from /lib64/libc.so.6
#11 0x00007ffff7ac71e0 in exit () from /lib64/libc.so.6
#12 0x00007ffff7aaf597 in __libc_start_call_main () from /lib64/libc.so.6
#13 0x00007ffff7aaf649 in __libc_start_main_impl () from /lib64/libc.so.6
#14 0x00000000004022f5 in _start ()
```

Di-Strix, 2022-04-21:
@mcspr That's it! I will check these functions. Thanks

I'm not really familiar with the gdb. I've just tried to run the program through gdb and I had no error.

<details>
<summary>Log</summary>

```console
ubuntu@ubuntu:~/Documents/PIOTestFailDemo/.pio/build/native$ gdb ./program 
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./program...
(gdb) run
Starting program: /home/ubuntu/Documents/PIOTestFailDemo/.pio/build/native/program 
test/MaterializeLayout/TestMaterializeLayout.cpp:54:test_injection_url:PASS
test/MaterializeLayout/TestMaterializeLayout.cpp:55:test_injection_empty_file:PASS

-----------------------
2 Tests 0 Failures 0 Ignored 
OK
[Inferior 1 (process 3310) exited normally]
(gdb) 
```
</details>

Of course I re-compiled it previously. PIO shows the SIGSEGV but gdb doen't

mcspr, 2022-04-21:
I also don't see the issue with ubuntu 20.04 + stock gcc9, so only thing that comes to mind is compiler version.
`apt install g++10 gcc-10` and `update-alternatives` will show the malloc error message.

Di-Strix, 2022-04-22:
@mcspr Installed g++-10, gcc-10 and made them default. Finally got that message. I removed `delete h` from my ESPAsyncWebServer mock. So... now I'm still getting SIGSEGV error in platformio and none in gdb. Really interesting that that error was happening despite removeHander was overridden...

Could you please pull changes from repo and re-run the program?

mcspr, 2022-04-22:
Just consider when dtors happen, and note that `bt` points to the end of the program. Previous issue was 'delete'ing *stack* pointer. Right now, when tests end, you are still operating on a stack pointer to the server that is already invalid after the specific test function ends.

Di-Strix, 2022-04-22:
@mcspr ok. Thanks!

Di-Strix, 2022-04-25:
Finally I resolved this issue. That was completely my fault. Inside tests I create local server instances and provide them to `MaterializeLayout`. The problem is that `MaterializeLayout` is global and it is re-initialized before each test. The `MaterializeLayout` destructor tried to remove all handlers from the provided server but it was removed since test-function is completed. So destructor tried to access deleted local server instance. So there weren't any issues with PlatformIO core.

Thanks for help

ivankravets, 2022-04-25:
How was difficult to debug this issue? We heard a similar problem when people fail with a non-zero exit code on the host machine.

Di-Strix, 2022-04-25:
I had no abilities to debug this because in the best case I was getting `free(): invalid pointer` message and that's all... That was even harder because gdb reported zero-return code.