platformio/platformio-core #4232: Platformio.org domain is not accessible from Moldova

vladkozlov69, 2022-04-17:
Hi guys,

today platformio.org domain became inaccessible from Moldova (e.g. IP 89.149.84.67 of Orange ISP provider). Via VPN through Romania it works fine. Could you please check whether this ISP's IP range was accidentally banned?

maxgerhardt, 2022-04-17:
Could the reverse be true? Your ISP blocking PlatformIO because it e.g. thinks its russian? We had a similiar situation [here](https://community.platformio.org/t/platformio-is-a-ukrainian-project-please-help-us-stop-the-war/26330/47?u=maxgerhardt).

ivankravets, 2022-04-17:
We don't have any tracks to Russia. The load balancer is located in the German data center. @vladkozlov69 which IP do you see when 

```
ping platformio.org
```?

Could you try `1.1.1.1` DNS?

vladkozlov69, 2022-04-17:
Thanks a lot @ivankravets ! I just changed my DNS to 1.1.1.1 and it works now. Resolves 88.198.170.159 and VS Code plugin works fine. Then I'm going to close this issue and will ask my ISP support about wrong blocking.

closed 2022-04-17 by vladkozlov69

ivankravets, 2022-04-17:
We would be thankful if you provide us with the reason why they blocked PlatformIO :(

Happy coding with PlatformIO!