platformio/platformio-core #4233: Allow Serial object configurability for unit tests

maxgerhardt, 2022-04-17:
What kind of issue is this?

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64

**PlatformIO Version** (`platformio --version`):

### Description of problem

Per [forum post](https://community.platformio.org/t/how-to-find-upload-protocol-for-this-samd21g18a-board/27231/14?u=maxgerhardt). 

When using the default transprot protocol for boards with the Arduino framework, PlatformIO decides to use the `Serial` object

https://github.com/platformio/platformio-core/blob/f2fbdafe6461e8e4e2c081cedb8475f8a90ca1ae/platformio/commands/test/processor.py#L25-L35

On some boards users however which to use `SerialUSB` or some other serial instead, which means they need to use `test_transport = custom` and implement that themselves. This could be made even shorter by allowing a e.g. `test_transport_port` option that will be used as the serial object (e.g. `SerialUSB`).

#### Steps to Reproduce

1. Use a board where `SerialUSB` is the standard and not `Serial` (e.g. some ATSAMD21G18 based boards)
2. Try and use the default unit test transport protocol
3. Click 'test'

### Actual Results

Not working since output on wrong serial.

### Expected Results

Configurability of the test transport serial.

### If problems with PlatformIO Build System:

-/-
### Additional info

-/-

ivankravets, 2022-04-17:
Thanks for the feature request. We work absolutely on the new unit testing engine. It will be a part of PlatformIO Core 6.0. There will no be `test_transport` anymore, but we will introduce custom unit test runners. Developers will be able to provide their own implementation on how to gather results from the target device.

ivankravets, 2022-04-29:
Answered in https://community.platformio.org/t/how-to-find-upload-protocol-for-this-samd21g18a-board/27231/19?u=ivankravets

closed 2022-04-29 by ivankravets