platformio/platformio-core #4237: PVS Studio throws AttributeError: 'NoneType' object has no attribute 'lower'

TD-er, 2022-04-22:
### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

The error itself seems to be similar to this (closed) issue: https://github.com/platformio/platformio-core/issues/3989

I also have contacted the PVS Studio team about this and their reply was:

> Hello, Gijs.
> 
> We built and analyzed your project with settings as you described on the screenshot. It seems that the PlatformIO starts preprocessing incorrectly which causes fail of a preprocessor (non-zero return code) and an error log window to appear instead of the normal interface.
> 
> Unfortunately, this problem is specific for your project and doesn't reproduce with clear projects. Perhaps it will be a good idea to open an issue at the PlatformIO repository.

It is about my project ESPEasy: https://github.com/letscontrolit/ESPEasy

I do have several PIO envs defined for checking with PVS Studio. They all have names starting with: ``spec_debug_*``

A while ago, the PVS Studio inspect option was running, but I could not browse through the found issues anymore.
Still the error output contained all PVS Studio output, so I could browse through it by copying it to a notepad and process them by hand.

N.B. Due to a very busy time with me moving to my rebuilt house, I had not yet found the time to open this issue.
My mail contact about this with the PVS Studio team was in the first half of March.

valeros, 2022-04-26:
Hi @TD-er ! So how do I reproduce the issue? What environment to run? I'd appreciate a bit more details.

TD-er, 2022-04-26:
- Open [my project](https://github.com/letscontrolit/ESPEasy) in VS code.
- Make sure you have a license key in the directory one level up of where you did checkout the ESPEasy repository.
- Run "inspect" on any of the "spec_debug..." envs

[image]

And then observe the errors:
[image]

closed 2022-04-27 by valeros

TD-er, 2022-04-27:
Just curious, what was the bug?
Was it something strange I'm doing in my project?

valeros, 2022-04-27:
TBH, I wasn't able to reproduce it, although I figured out where it may fail. Anyway, please do retest with the latest development version of the PlatformIO. Please note I also updated the PVS package to the latest 7.18.

TD-er, 2022-05-17:
I just tested with PIO 6.0.1 and I'm sorry to report that it still isn't fixed :(

```
Error: Checking spec_debug_custom_ESP32_4M316k > pvs-studio (platform: https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-pvs-studio @ ~7.18.0
Downloading
Unpacking
Tool Manager: tool-pvs-studio@7.18.59865 has been installed!
C:\Users\gijsn\.platformio\packages\tool-pvs-studio\x64\pvs-studio --skip-cl-exe yes --language C++ --preprocessor gcc --cfg C:\Users\gijsn\AppData\Local\Temp\piocheckxxhozqsj\wxa0qeer.cfg --source-file c:\GitHub\TD-er\ESPEasy\src\ESPEasy-Globals.cpp --i-file C:\Users\gijsn\AppData\Local\Temp\piocheckxxhozqsj\e227jwtn.i --output-file C:\Users\gijsn\AppData\Local\Temp\piocheckxxhozqsj\5yo3eacu.pvs --platform=arm --analysis-mode=4 --errors-off=V532,V586,V795 --lic-file=../pvs_studio.lic

Error: Traceback (most recent call last):
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
cli() # pylint: disable=no-value-for-parameter
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
return self.main(*args, **kwargs)
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
rv = self.invoke(ctx)
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
return super().invoke(ctx)
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
return __callback(*args, **kwargs)
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\platformio\commands\check\command.py", line 136, in cli
rc = ct.check(
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\platformio\commands\check\tools\pvsstudio.py", line 247, in check
self._process_defects(self.parse_defects(self._tmp_output_file))
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\platformio\commands\check\tools\pvsstudio.py", line 140, in parse_defects
DefectItem(
File "C:\Users\gijsn\.platformio\penv\lib\site-packages\platformio\commands\check\defect.py", line 54, in __init__
if file.lower().startswith(get_project_dir().lower()):
AttributeError: 'NoneType' object has no attribute 'lower'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
`pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
https://docs.platformio.org/page/faq.html

* Report this problem to the developers
https://github.com/platformio/platformio-core/issues

============================================================
```

TD-er, 2022-05-17:
Not 100% sure what I may have changed, but now I do get some usable output from PVS Studio, but the output is not browseble in the PIO Home inspect window.

What I did (and have tried several times):
- Deleted everything related to PlatformIO and reinstalled
- Significantly reduced the requirements.txt files for my project + updated packages
- Recreated the license file (read: threw away too many files) for PVS Studio
- Probably something else

The output is like this:
```
Error: Checking spec_debug_custom_ESP8266_4M1M > pvs-studio (platform: espressif8266@2.6.3; board: esp8266_4M1M_board; framework: arduino)
--------------------------------------------------------------------------------
C:\Users\gijsn\.platformio\packages\tool-pvs-studio\x64\pvs-studio --skip-cl-exe yes --language C++ --preprocessor gcc --cfg C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\ur0uiu79.cfg --source-file c:\GitHub\TD-er\ESPEasy\src\ESPEasy-Globals.cpp --i-file C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\5wf2b0yt.i --output-file C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\irqmzngu.pvs --platform=arm --analysis-mode=4 --errors-off=V532,V586,V795 --lic-file=../pvs_studio.lic
C:\Users\gijsn\.platformio\packages\tool-pvs-studio\x64\pvs-studio --skip-cl-exe yes --language C++ --preprocessor gcc --cfg C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\ur0uiu79.cfg --source-file c:\GitHub\TD-er\ESPEasy\src\ESPEasy.ino --i-file C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\5wf2b0yt.i --output-file C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\irqmzngu.pvs --platform=arm --analysis-mode=4 --errors-off=V532,V586,V795 --lic-file=../pvs_studio.lic
C:\Users\gijsn\.platformio\packages\tool-pvs-studio\x64\pvs-studio --skip-cl-exe yes --language C++ --preprocessor gcc --cfg C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\ur0uiu79.cfg --source-file c:\GitHub\TD-er\ESPEasy\src\ESPEasy_common.cpp --i-file C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\5wf2b0yt.i --output-file C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\irqmzngu.pvs --platform=arm --analysis-mode=4 --errors-off=V532,V586,V795 --lic-file=../pvs_studio.lic
C:\Users\gijsn\.platformio\packages\tool-pvs-studio\x64\pvs-studio --skip-cl-exe yes --language C++ --preprocessor gcc --cfg C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\ur0uiu79.cfg --source-file c:\GitHub\TD-er\ESPEasy\src\_C001.cpp --i-file C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\5wf2b0yt.i --output-file C:\Users\gijsn\AppData\Local\Temp\piocheckx7kocp9x\irqmzngu.pvs --platform=arm --analysis-mode=4 --errors-off=V532,V586,V795 --lic-file=../pvs_studio.lic

<cut about 2200 lines>

src\src\Globals\ESPEasyWiFiEvent.h:26: [low:note] The 'IPAddress' class implements a copy constructor, but lacks the copy assignment operator. It is dangerous to use such a class. [V690]
src\src\WebServer_tmp\__tmpfile.cpp:4665: [medium:warning] Two or more case-branches perform the same actions. Check lines: 4665, 4738 [V1037]
src\src\WebServer_tmp\__tmpfile.cpp:9879: [high:error] Expression 'refreshCount != 0' is always false. [V547]
src\src\WebServer_tmp\__tmpfile.cpp:10981: [low:note] Consider checking for misprints. It's possible that the 'value_msec' should be checked here. [V1051]
src\src\WebServer_tmp\__tmpfile.cpp:11654: [medium:warning] A part of conditional expression is always true: !hasWiFiCredentials. [V560]
src\src\WebServer_tmp\__tmpfile.cpp:12791: [medium:warning] The 'varName' function argument possesses the same name as one of the class members, which can result in a confusion. [V688]
src\src\WebServer_tmp\__tmpfile.cpp:12064: [low:note] It is odd that the body of 'json_prop' function is fully equivalent to the body of 'json_number' function. [V524]
src\src\DataStructs_templ\SettingsStruct.cpp:14: [low:note] It is possible that not all members of a class are initialized inside the constructor. Consider inspecting: PID, Version, Build, IP, Gateway, Subnet, ... [V730]
Error: pvs-studio failed to perform check! Please examine tool output in verbose mode.

Component HIGH MEDIUM LOW
--------------------------------------------------------- ------ -------- -----
.pio\libdeps\spec_debug_custom_ESP8266_4M1M\ESPeasySerial 242 0 0
src 0 0 1
src\src\Commands 0 3 22
src\src\Commands_tmp 0 3 0
src\src\DataStructs 0 0 343
src\src\DataStructs_templ 0 0 303
src\src\DataStructs_tmp 0 0 1
src\src\DataTypes 0 0 1
src\src\DataTypes_tmp 0 0 1
src\src\ESPEasyCore 0 1 2
src\src\Globals 1 2 32
src\src\Globals_tmp 1 2 2
src\src\Helpers 0 7 600
src\src\Helpers_tmp 0 7 12
src\src\Static 0 1 0
src\src\WebServer 1 3 2
src\src\WebServer_tmp 1 3 2

Total 246 32 1324

========================= [FAILED] Took 525.62 seconds =========================
Environment Tool Status Duration
---------------------------------------- ---------- -------- ------------
custom_ESP8266_4M1M cppcheck IGNORED
custom_IR_ESP8266_4M1M cppcheck IGNORED
custom_alt_wifi_ESP8266_4M1M cppcheck IGNORED
custom_302_ESP8266_4M1M cppcheck IGNORED
custom_beta_ESP8266_4M1M cppcheck IGNORED
custom_beta_IR_ESP8266_4M1M cppcheck IGNORED
custom_beta_debug_ESP8266_4M1M cppcheck IGNORED
custom_ESP8266_4M2M cppcheck IGNORED
custom_ESP8266_4M2M_LittleFS cppcheck IGNORED
custom_ESP8266_1M cppcheck IGNORED
custom_alt_wifi_ESP8266_1M cppcheck IGNORED
custom_beta_ESP8266_1M cppcheck IGNORED
custom_ESP8266_2M256 cppcheck IGNORED
normal_ESP8266_1M cppcheck IGNORED
normal_alt_wifi_ESP8266_1M cppcheck IGNORED
normal_ESP8266_1M_VCC cppcheck IGNORED
[... cut ...]
test_D_ESP32s2_4M316k cppcheck IGNORED
energy_ESP32s2_4M316k cppcheck IGNORED
display_ESP32s2_4M316k cppcheck IGNORED
spec_debug_custom_ESP8266_4M1M pvs-studio FAILED 00:08:45.619
spec_debug_custom_IR_ESP8266_4M1M pvs-studio IGNORED
spec_debug_beta_custom_ESP8266_4M1M pvs-studio IGNORED
spec_debug_custom_ESP32_4M316k pvs-studio IGNORED
spec_debug_max_ESP32_16M8M pvs-studio IGNORED
spec_memanalyze_ESP8266 cppcheck IGNORED
==================== 1 failed, 0 succeeded in 00:08:45.619 ====================
```

TD-er, 2022-05-17:
And even more weird...
I now ran another of my special debug envs:

[image]

So I'm a bit lost to why it now is working.
All I can think of is that after a full cleaning session, you usually need 2 or 3 builds to make a normal build.
Cleaning is deleting these folders:
[image]