platformio/platformio-core #4240: Add support for doctest

jcw, 2022-04-22:
- [x] **Feature Request**.

The "doctest" unit test framework is a variant of "Catch2", but much simpler (and faster). I've been using this on native builds for some time now. For an example, tied into PIO for native, but still using Unity for embedded tests, see https://git.jeelabs.org/monty/tree/test?h=v1.6

See also https://github.com/doctest/doctest/issues/506 which tries to run Doctest on embedded.

The key benefit, as with Catch2, is the amount of info you can see when tests fail, i.e. a test such as  `CHECK(abc == 42);` then the _actual_ value of abc will also be reported. Better still, you can capture other expressions tp print out, only on failure. This is such a game-changer that I've been going out of my way to test as much as possible on native (always a good strategy, of course).

If nothing else, I would be nice to support mixed doctest/unity testing, but supporting doctest on embedded would be even more powerful.

(This feature request was previously mentioned at https://github.com/platformio/platformio-core/issues/2185#issuecomment-1106593399)

closed 2022-05-06 by ivankravets

Di3mex, 2022-05-09:
@ivankravets Is it possible to use the VS Code GUI to run the tests? For example via the TestMate extension?

ivankravets, 2022-05-09:
Which features from that extension are the most important for you? You can use that extension for "native executables" but not for the embedded.

Di3mex, 2022-05-09:
[image]
Right now I cannot use the Test Explorer from Vs Code. I would like to be able to use the GUI. It doesn't have to be via the TestMate extension. It just seemed like it would/should work through the Extension. 

I also really would like to be able to simply press the debug unit tests feature from withing the GUI.

From previous experience with integrated unit testing framework in IntelliJ / AndroidStudio I really miss the integrated GUI overview tables that would show you the list of the tests, allow you to run individual (failing) tests, debug a specific test. 

Overall the unit testing experience is quite slow and cumbersome on PIO in VsCode, so I am really excited to see some advancements in a more modern unit testing direction! Thanks for the support and work on that!

commit 82778473 (2022-05-06): New: "doctest" testing framework // Resolve #4240