platformio/platformio-core #4247: Generate project SBOM in SPDX format

ivankravets, 2022-04-28:
Generate [Software bill of materials](https://en.wikipedia.org/wiki/Software_bill_of_materials) in the [SPDX](https://spdx.github.io/spdx-spec/composition-of-an-SPDX-document/) format that is already used for package manifest especially for `license` field.

Proposed CLI:

```
pio project spdx
```

simogaspa84, 2023-02-27:
Hi @ivankravets   Is this already implemented in platformio ? 
Thanks

pr0dus, 2026-10-30:
Until this lands in PlatformIO itself: I published a small open-source tool that generates SPDX 2.3 and CycloneDX 1.6 SBOMs from a PlatformIO project (`platformio.ini` envs incl. `extends`/`${...}` references, `lib_deps` from registry, Git and local paths, installed `.pio/libdeps`, platform/framework packages). It runs offline and also has a GitHub Action: https://github.com/pr0dus/fw-sbom

Versions it can't determine are marked unknown rather than guessed, so running `pio pkg install` first gives the most complete result. Feedback welcome.