platformio/platformio-core #4251: Show list of failed tests in the summary

nbolton, 2022-04-30:
Within my project, I have many test functions. When one fails, finding out which one is tedious. Filtering is only useful if you know which test will fail (but when you have hundreds of tests, you need to search for the one that failed in the logs).

Idea: Show a list of failed tests in the summary at the end (so you don't have to hunt for failed tests in the log).

Moved here from: https://github.com/ThrowTheSwitch/Unity/issues/611#issuecomment-1113936305

Different to https://github.com/platformio/platformio-core/issues/4248 in that it is related to the summary.

Possible workaround:
> What I know is that Unity has a lot of helpers (Ruby scripts). I've not tried them personally. See https://github.com/ThrowTheSwitch/Unity/tree/master/auto
- @ivankravets

ivankravets, 2022-05-03:
Thanks for the feature request. Please re-test with `pio upgrade --dev`.

nbolton, 2022-05-04:
You're awesome.

[image]

commit b12d9f62 (2022-05-03): Show list of failed tests in the summary // Resolve #4251