platformio/platformio-core #4253: Library dependency finder ignores `src_build_flags`

mcspr, 2022-05-02:
- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: Fedora 36

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.0.0a2

### Description of problem

MCVE of the currently not working code is at https://github.com/mcspr/pio-scan-deps
Latest action run reproduces the error below https://github.com/mcspr/pio-scan-deps/runs/6246889310

The project is:
```
> tree
.
├── platformio.ini
├── src
│   ├── extra.cpp
│   ├── extra.h
│   └── main.cpp
└── vendor
    └── testlib
        ├── library.json
        └── src
            ├── testlib.cpp
            └── testlib.h
```

Where `extra.cpp` conditionally includes a local library
```cpp
#if THIS_IS_A_TEST
#include <testlib.h>
... the rest of the module's code ...
#endif
```

When `lib_ldf_mode = deep+` is set, PIO scanner will ignore the testlib.h
GCC still picks it up, though, resulting in a failed build
```
> env PLATFORMIO_BUILD_FLAGS="-DTHIS_IS_A_TEST=1" pio run -s
> env PLATFORMIO_SRC_BUILD_FLAGS="-DTHIS_IS_A_TEST=1" pio run -s
src/extra.cpp:4:10: fatal error: testlib.h: No such file or directory

*****************************************************************
* Looking for testlib.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:testlib.h"
* Web  > https://registry.platformio.org/search?q=header:testlib.h
*
*****************************************************************

    4 | #include <testlib.h>
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio/build/d1_mini/src/extra.cpp.o] Error 1
```

While I have noticed that examples in documentation only mention the `build_flags = ...` as a way to modify the dependency scanner behaviour...
I am not really sure whether this was intentional that project build flags are ignored?
Should the scanner do a 2nd pass with the project dependencies, while using the flags provided in the `src_build_flags`?

#### Steps to Reproduce

See the repo

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env]
platform = espressif8266
framework = arduino
build_flags = -g
lib_ldf_mode = deep+
lib_extra_dirs =
    vendor

[env:d1_mini]
board = d1_mini
```

**Source file to reproduce issue:**
```cpp
// src/main.cpp

#include <Arduino.h>
#include "extra.h"

void setup() {
#if THIS_IS_A_TEST
    test();
#endif
}

void loop() {
}
```
```cpp
// src/extra.cpp

#include <Arduino.h>

#if THIS_IS_A_TEST
#include <testlib.h>

void test() {
}
#endif
```
```cpp
// src/extra.h

#pragma once

void test();
```

Plus, a dummy library consisting of `lib/testlib/src/testlib.h` and `lib/testlib/src/testlib.cpp`. Both are empty.

ivankravets, 2022-05-02:
Where is `THIS_IS_A_TEST` defined? It seems this issue is a duplicate of https://github.com/platformio/platformio-core/issues/3113

The only solution to this issue is to switch to a native GCC Preprocessor. Currently, PlatformIO uses a virtual SCons/Pythonbased Preprocessor.

mcspr, 2022-05-02:
Flag is set through the command line `-DTHIS_IS_A_TEST=1` (or `0` to disable), and it does work correctly when global `build_flags = ...` is used.

As I understood the #3113 problem is with expressions and defines referencing other defines, [which I have also happened to notice](https://github.com/mcspr/pio-nested-define). SCons scanner still works out simple flags like this, though.

ivankravets, 2022-05-02:
> and it does work correctly when global build_flags = ... is used.

`src_build_flags` was renamed in Core 6.0 to the `build_src_flags`. It means flags will be applied ONLY to the files located in the main `src` folder of a project.

mcspr, 2022-05-02:
Not yet renamed though, https://github.com/platformio/platformio-core/issues/4245, and grep does not find that name
Yes, the flag is only seen by the project file, but it hides a library include. Shouldn't it create a dependency on the lib?

ivankravets, 2022-05-05:
Thanks, the option has been finally renamed.

> #if THIS_IS_A_TEST

You have a typo here, please use `#ifdef`.

mcspr, 2022-05-05:
...I have demonstrated that the option / flag works with build_flags = ... though! This is not a typo, it is intended to be used as a boolean.
The feature request / question was about the ability to influence LDF with project flags hiding library includes. Right now this just does not work.

mcspr, 2022-05-05:
See https://github.com/mcspr/pio-scan-deps, updated with what works and what does not as environments.

ivankravets, 2022-05-05:
Thanks, this is a bug. We will fix it in 6.0.

ivankravets, 2022-05-06:
Please re-test with `pio upgrade --dev`.

closed 2022-05-06 by ivankravets

mcspr, 2022-05-07:
Everything works! Thank you.
https://github.com/mcspr/pio-scan-deps/runs/6335188901?check_suite_focus=true

commit c4388a69 (2022-05-06): Fixed an issue when LDF ignores build_src_flags in the “deep+” mode // Resolve #4253