platformio/platformio-core #4257: Upload to board fails if specified wildcards in upload_port option

wackyfrog, 2022-05-06:
### Configuration

**Operating system**:
mac os 12.3.1

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.2.5

### Description of problem

Upload fails if specified wildcards in upload_port

#### Steps to Reproduce

`pio run -t upload`

### Actual Results

```
Looking for upload port...
**Error: Please specify `upload_port` for environment or use global `--upload-port` option.**
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1

```
**The content of `platformio.ini`:**
```

[env:uno]
platform = atmelavr
board = uno
framework = arduino
upload_port = /dev/cu.usbmodem*

```
### Additional info
`pio device list`

```
/dev/cu.wlan-debug
------------------
Hardware ID: n/a
Description: n/a

/dev/cu.Bluetooth-Incoming-Port
-------------------------------
Hardware ID: n/a
Description: n/a

/dev/cu.usbmodem11201
---------------------
Hardware ID: n/a
Description: n/a
```

wackyfrog, 2022-05-06:
Have found the solution:

```
brew reinstall platformio
brew link --overwrite platformio
```
The problem was in incorrect version of pySerial.

ivankravets, 2022-05-06:
How it was possible? PlatformIO uses the latest PySerial package.

How did you install PlatformIO?

ivankravets, 2022-05-06:
Please uninstall "platformio" via "brew" and use the [get-platformio.py](https://docs.platformio.org/en/latest/core/installation.html#installer-script) installer script. It will create isolated virtual environment with dependent packages.

closed 2022-05-06 by ivankravets

adgbu, 2023-10-12:
I have the same issue of ```upload_port``` with wildcard suffix not working.

These are my settings in ```platformio.ini```:
```
; This works! Specifying the exact device.
upload_port = /dev/tty.usbmodemDC5475C515F82

; This does not work! Specifying the device prefix and end with wildcard.
upload_port = /dev/tty.usbmodem*
```

This is the output  I get when trying to upload (libraries omitted):
```
CONFIGURATION: https://docs.platformio.org/page/boards/renesas-ra/uno_r4_wifi.html
PLATFORM: Renesas RA (1.2.0) > Arduino Uno R4 WiFi
HARDWARE: RA4M1 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink)
PACKAGES: 
 - framework-arduinorenesas-uno @ 1.0.4 
 - tool-bossac @ 1.10901.0 (1.9.1) 
 - tool-jlink @ 1.78811.0 (7.88.11) 
 - tool-openocd @ 3.1200.0 (12.0) 
 - toolchain-gccarmnoneeabi @ 1.100301.220327 (10.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
```
... omitting libraries
```
Building in release mode
Checking size .pio/build/uno_r4_wifi/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.4% (used 4720 bytes from 32768 bytes)
Flash: [==        ]  16.0% (used 41935 bytes from 262144 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
```

I am using PlatformIO (Core 6.1.11, Home 3.4.4) in VSCode 1.83.1 on macOS 13.5.2