platformio/platformio-core #4259: Renaming the "test_build_project_src" to test_build_src prevents tests from running on esp32

drorgl, 2022-05-11:
What kind of issue is this?

- [V ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.0.0rc1

### Description of problem
Can't run tests after update to latest platformio core on esp32

#### Steps to Reproduce

1. Install platformio latest version (not sure why it installed version 6 if its RC1 on my machine)
2. Build a simple project with a single test configured on platform = espressif32, board = esp32doit-devkit-v1, framework = espidf
3. pio test

### Actual Results
Warning! `test_build_project_src` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `test_build_src` instead
InvalidProjectConfError: Invalid '\platformio.ini' (project configuration file): 'No option 'test_build_project_src' in section: 'env:esp32'':
  File "\penv\lib\site-packages\platformio\builder\main.py", line 184:
    env.SConscript("$BUILD_SCRIPT")
  File "\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "\platforms\espressif32\builder\main.py", line 282:
    target_elf = env.BuildProgram()
  File "\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 63:
    env.ProcessProgramDeps()
  File "\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 126:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 346:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 660:
    return method(*args, **kw)
  File "\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "\platforms\espressif32\builder\frameworks\espidf.py", line 1286:
    if "__test" not in COMMAND_LINE_TARGETS or env.GetProjectOption(
  File "\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "\penv\lib\site-packages\platformio\builder\tools\pioproject.py", line 30:
    return env.GetProjectConfig().get("env:" + env["PIOENV"], option, default)
  File "\penv\lib\site-packages\platformio\project\config.py", line 317:
    raise exception.InvalidProjectConfError(self.path, str(e))
Building stage has failed, see errors above. Use `pio test --verbose` option to enable verbose output.

### Expected Results
Tests should run

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32]
platform = espressif32
board = esp32doit-devkit-v1
framework = espidf
test_speed = 115200
test_build_src = true
test_build_project_src = true
```
A quick workaround was to rename the field in platforms\espressif32\builder\frameworks\espidf.py 
```python
if "__test" not in COMMAND_LINE_TARGETS or env.GetProjectOption(
    "test_build_project_src"
):
```

ivankravets, 2022-05-11:
Thanks for the report!

1) We will check the issue with renamed option.
2) You need to use custom test runner and disable default PlatformIO's Unity https://docs.platformio.org/en/latest/advanced/unit-testing/frameworks/custom/examples/custom_unity_library.html

drorgl, 2022-05-12:
Sorry, I didn't test your suggestion, I've reverted back to PlatformIO 5.

ivankravets, 2022-05-12:
Could you re-run `pio upgrade --dev` and re-test with RC2? You will need to disable PlatformIO's latest Unity version and use built-in IDF. See docs https://docs.platformio.org/en/latest/advanced/unit-testing/frameworks/custom/examples/custom_unity_library.html

Does it work now?

ivankravets, 2022-05-12:
Fixed on our side. Please re-run `pio upgrade --dev`. Your ESP-IDF project should work with PIO Core 6.0 now without any changes.

commit 79b2bfde (2022-05-12): Fix an issue with multiple symbol definitions when framework uses own Unity // Resolve #4259