platformio/platformio-core #4260: piodebuggdb makes mistakes in parsing command parameters

asereq, 2022-05-13:
Hi, everybody,
really thanks for the great project!

I am having problems debugging with Eclipse.

Eclipse is calling this command:
`piodebuggdb --interpreter mi2 --nx -q --interpreter console -ex "new-ui mi /dev/pts/16" -ex "set pagination off" -ex "show version"`
but if I execute this on terminal it results on:
`Error: Unknown environment names 'x'. Valid names are 'zero'`
I thinks there is an error on parsing gdb parameters, the "-ex" is interpreted as "-e" of piodebuggdb argument (environment) plus "x" which is not a valid environment name. It is possible?

Thanks in advance,
Andrés.

------------------------------------------------------------------

### Configuration

**Operating system**: Linux (Arch)

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.0.0rc2

**Eclipse version**:
Eclipse IDE for Embedded C/C++ Developers (includes Incubating components)
Version: 2022-03 (4.23.0)
Build id: 20220310-1457
OS: Linux, v.5.17.4-arch1-1, x86_64 / gtk 3.24.33, WebKit 2.36.1
Java vendor: N/A
Java runtime version: 18.0.1+10
Java version: 18.0.1

ivankravets, 2022-05-13:
Thanks! Updated docs. You need to split PlatformIO's options with GDB's using a double hyphen. 

```
piodebuggdb --interpreter mi2 -- --nx -q --interpreter console -ex "new-ui mi /dev/pts/16" -ex "set pagination off" -ex "show version"
```

closed 2022-05-13 by ivankravets

asereq, 2022-05-13:
Thank you!
I'm not sure... The PlatformIO preparation for the Eclipse IDE will now takes into account those modifications?

ivankravets, 2022-05-13:
I don't remember the Eclipse tempalte. Where do you put `--nx -q --interpreter console -ex ... `?

asereq, 2022-05-13:
I don't know, for me this is so obscure. I think I am using default configurations. Also, I don't know where is the .pioinit file. I only follow the guide here: https://docs.platformio.org/en/latest/integration/ide/eclipse.html#debugging

ivankravets, 2022-05-14:
I see. You just need https://docs.platformio.org/en/latest/projectconf/section_env_debug.html#debug-init-cmds

Put your initial commands here.

murmurlab, 2026-01-01:
how to set `--interpreter=mi2` to `=mi`

commit 7c93167d (2022-05-13): Docs: Document double hyphen for "pio debug" // Resolve #4260