platformio/platformio-core #4261: Doctest integration doesn't decode or show exceptions

Di3mex, 2022-05-13:
------------------------------------------------------------------

### Configuration

**Operating system**:

Windows 10 64 bit, gcc installed via Cygwin 

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 6.0.0rc1

### Description of problem

#### Steps to Reproduce

### Actual Results

ERROR: test case THREW exception: unknown exception

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

[env:native]
platform = native
build_type = debug
debug_test = test_native
debug_build_flags = -O0 -g3 -ggdb3
test_framework = doctest
build_flags = 
  -O0
  -Wall 
  -D DOCTEST_CONFIG_NO_POSIX_SIGNALS
```

**Source file to reproduce issue:**
```
#define DOCTEST_CONFIG_IMPLEMENT // REQUIRED: Enable custom main()
#include <doctest.h>

struct my_little_exception {};

TEST_CASE("I'll just throw something") {
    SUBCASE("I hope someone is going to catch it") {
        throw my_little_exception();
    }
}

int main(int argc, char **argv) {
    doctest::Context context;

    // BEGIN:: PLATFORMIO REQUIRED OPTIONS
    context.setOption("success", true);     // Report successful tests
    context.setOption("no-exitcode", true); // Do not return non-zero code on failed test case
    // END:: PLATFORMIO REQUIRED OPTIONS

    // YOUR CUSTOM DOCTEST OPTIONS

    context.applyCommandLine(argc, argv);
    return context.run();
}
```

### Additional info

valeros, 2022-05-13:
Hi @Di3mex ! I'm no expert on Doctest, but I guess PlatformIO is not of any help here. It seems that Doctest itself is not able to recognize the exception and PlatformIO simply parses the Doctest output and passes it to you in a unified format. Is there any way to force Doctest to print any additional info, maybe a specific configuration flag or something?

Di3mex, 2022-05-15:
@valeros I did check the configuration options and tried the ones that had something to do with exception handling. They didn't change anything.

ivankravets, 2022-05-15:
Does it work for you if you use the Doctest WITHOUT PlatformIO? I don't think we can solve this issue on our part.

closed 2022-05-19 by ivankravets