platformio/platformio-core #4265: Compile breaks by unused library parts.

JoergSH, 2022-05-14:
I’ve been having trouble compile for a while. Code that used to always compile without any problems now breaks because files were not found. But these are files from unused parts of the library that shouldn’t actually be compiled.

You could use every sample code from adafruit to reproduce the problem.

```
[ [env:az-delivery-devkit-v4]
platform = espressif32
board = az-delivery-devkit-v4
framework = arduino
lib_deps =
adafruit/Adafruit BNO055@^1.5.3
adafruit/Adafruit Unified Sensor@^1.1.5
```

I have got help with a work around in the Community by adding 

#include <Adafruit_SPIDevice.h>

but that's  not a real solution.

```
Processing az-delivery-devkit-v4 (platform: espressif32; board: az-delivery-devkit-v4; framework: arduino)
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION:
PLATFORM: Espressif 32 (3.5.0) > AZ-Delivery ESP-32 Dev Kit C V4
HARDWARE: ESP32 240MHz, 520KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:

framework-arduinoespressif32 3.10006.210326 (1.0.6)
tool-esptoolpy 1.30100.210531 (3.1.0)
toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder →
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies…
Dependency Graph
|-- 1.5.3
| |-- 1.11.6
| | |-- 1.0.1
| |-- 1.0.1
| |-- 1.1.5
|-- 1.1.5
|-- 1.0.3
|-- 1.0.1
Building in release mode
Compiling .pio/build/az-delivery-devkit-v4/src/main.cpp.o
Generating partitions .pio/build/az-delivery-devkit-v4/partitions.bin
Compiling .pio/build/az-delivery-devkit-v4/libd2e/Wire/Wire.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_I2CDevice.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_SPIDevice.cpp.o
In file included from .pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_BusIO_Register.h:10:0,
from .pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_BusIO_Register.cpp:1:
.pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_SPIDevice.h:9:17: fatal error: SPI.h: No such file or directory
Looking for SPI.h dependency? Check our library registry!
CLI > platformio lib search “header:SPI.h”
Web > [PlatformIO Registry](https://registry.platformio.org/search?q=header:SPI.h)
compilation terminated.
Compiling .pio/build/az-delivery-devkit-v4/lib45f/Adafruit Unified Sensor/Adafruit_Sensor.cpp.o
In file included from .pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_SPIDevice.cpp:1:0:
.pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_SPIDevice.h:9:17: fatal error: SPI.h: No such file or directory

Looking for SPI.h dependency? Check our library registry!
CLI > platformio lib search “header:SPI.h”
Web >
compilation terminated.
Compiling .pio/build/az-delivery-devkit-v4/lib09c/Adafruit BNO055/Adafruit_BNO055.cpp.o
*** [.pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o] Error 1
*** [.pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_SPIDevice.cpp.o] Error 1
======================================================= [FAILED] Took 2.50 seconds =======================================================
Der Terminalprozess “platformio ‘run’” wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

Executing task in folder TestBNO: platformio run <

Processing az-delivery-devkit-v4 (platform: espressif32; board: az-delivery-devkit-v4; framework: arduino)
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION:
PLATFORM: Espressif 32 (3.5.0) > AZ-Delivery ESP-32 Dev Kit C V4
HARDWARE: ESP32 240MHz, 520KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:

framework-arduinoespressif32 3.10006.210326 (1.0.6)
tool-esptoolpy 1.30100.210531 (3.1.0)
toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder →
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies…
Dependency Graph
|-- 1.5.3
| |-- 1.11.6
| | |-- 1.0.1
| |-- 1.0.1
| |-- 1.1.5
|-- 1.1.5
|-- 1.0.3
|-- 1.0.1
Building in release mode
Compiling .pio/build/az-delivery-devkit-v4/src/main.cpp.o
Generating partitions .pio/build/az-delivery-devkit-v4/partitions.bin
Compiling .pio/build/az-delivery-devkit-v4/libd2e/Wire/Wire.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_I2CDevice.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_SPIDevice.cpp.o
In file included from .pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_BusIO_Register.h:10:0,
from .pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_BusIO_Register.cpp:1:
.pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_SPIDevice.h:9:17: fatal error: SPI.h: No such file or directory
Looking for SPI.h dependency? Check our library registry!
CLI > platformio lib search “header:SPI.h”
Web >
compilation terminated.
Compiling .pio/build/az-delivery-devkit-v4/lib45f/Adafruit Unified Sensor/Adafruit_Sensor.cpp.o
In file included from .pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_SPIDevice.cpp:1:0:
.pio/libdeps/az-delivery-devkit-v4/Adafruit BusIO/Adafruit_SPIDevice.h:9:17: fatal error: SPI.h: No such file or directory

Looking for SPI.h dependency? Check our library registry!
CLI > platformio lib search “header:SPI.h”
Web > [PlatformIO Registry](https://registry.platformio.org/search?q=header:SPI.h)
compilation terminated.
Compiling .pio/build/az-delivery-devkit-v4/lib09c/Adafruit BNO055/Adafruit_BNO055.cpp.o
*** [.pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o] Error 1
*** [.pio/build/az-delivery-devkit-v4/lib997/Adafruit BusIO/Adafruit_SPIDevice.cpp.o] Error 1
======================================================= [FAILED] Took 2.55 seconds =======================================================
Der Terminalprozess “platformio ‘run’” wurde mit folgendem Exitcode beendet: 1.
```

ivankravets, 2022-05-14:
> src/main.cpp

Add `#include <SPI.h>` to your `main.cpp`.

closed 2022-05-14 by ivankravets

JoergSH, 2022-05-14:
Why I should add spi when I did not use it? 
The sensor is connected by i2c.
And last Year that code compiles without problem.

ivankravets, 2022-05-14:
https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ldf-mode

If you add `lib_ldf_mode = chain+` to your configuration file - does it work now?

JoergSH, 2022-05-14:
No did not helps. Only to add SPI.h

ivankravets, 2022-05-14:
It looks like a known issue and depends on https://github.com/platformio/platformio-core/issues/3113

The only solution is to help the library dependency finder to resolve includes. Please note that SPI will not be built at all. Do you see it in the Dependency Graph?

JoergSH, 2022-05-14:
I think it is build

```
Building in release mode
Compiling .pio\build\az-delivery-devkit-v4\src\main.cpp.o
Archiving .pio\build\az-delivery-devkit-v4\lib8b5\libWire.a
Compiling .pio\build\az-delivery-devkit-v4\libb37\SPI\SPI.cpp.o
Compiling .pio\build\az-delivery-devkit-v4\lib4ab\Adafruit BusIO\Adafruit_BusIO_Register.cpp.o
Compiling .pio\build\az-delivery-devkit-v4\lib4ab\Adafruit BusIO\Adafruit_SPIDevice.cpp.o
Archiving .pio\build\az-delivery-devkit-v4\lib88a\libAdafruit BNO055@1.5.0.a
Archiving .pio\build\az-delivery-devkit-v4\lib061\libPCF8575.a
Archiving .pio\build\az-delivery-devkit-v4\lib404\libEEPROM.a
Archiving .pio\build\az-delivery-devkit-v4\lib4ab\libAdafruit BusIO.a
Archiving .pio\build\az-delivery-devkit-v4\libb
```

ivankravets, 2022-05-14:
Sorry, I didn't read well your code. The library which you use depends on SPI. See https://github.com/adafruit/Adafruit_BusIO/blob/master/Adafruit_SPIDevice.h

JoergSH, 2022-05-14:
No, SPI and i2c is supported. I use an BNO055 with i2c
So there should no dependency to spi.

ivankravets, 2022-05-14:
It is hard to answer why it does not work now. Please use strict dependency declaration. I see your `platformio.ini` contains `adafruit/Adafruit BNO055@^1.5.3`. The 1.5.3 version has been released a few days ago.

It seems that you didn't use semantic declaration and it will not be easy to find out which version of that library worked for you. As you can see, the latest BusIO library checks `SPI_INTERFACES_COUNT` macro. If it is not defined, it builds SPI.

You don't need to worry about this. If you don't use SPI functionality, the linker will drop SPI from your firmware.