platformio/platformio-core #4270: This is all I get today, nothing past this, tried several diffent builds on pc that worked, last time I compiled was on 8th, not sure if windows update or what... I updated everyting I could in VSS and PIO and got a warning about powershell so I also updated that and reinstalled from microsoft....

Forunnerod, 2022-05-16:


ivankravets, 2022-05-16:
Could someone provide a simple project to reproduce this issue?

The `src_filter` option has been renamed to the `build_src_filter`, but the previous version should work too.

SISLANGER, 2022-05-16:
@ivankravets Same on Mac when compiling Marlin Firmware

ivankravets, 2022-05-16:
@thinkyhead sorry that disturb you. Did we break something in PlatformIO Core 6.0? It has been released a few hours ago.

MakeNPrint, 2022-05-16:
@ivankravets  I can confirm renaming occurances src_filter with build_src_filter resolves the issue. So appears the backward compatability is not working.

MakeNPrint, 2022-05-16:
@ivankravets update it fails at the last hurdle.

It now compiles but has a load of yellow warning and at the end fails to produce the firmware.bin. So while it fixes the PlatformIO issue, I suspect it breaks Marlin.

ivankravets, 2022-05-16:
We found a bug on our side and will provide a quick fix soon. Sorry.

While this issue has not been fixed, please downgrade temporary to PlatformIO Cotr 5.0 via opening PlatformIO Core CLI in VSCode and typing:

```
pip install "platformio<6"
```

It also might require removing the following folders:

- `~/.platformio/platforms`
- `~/.platformio/packages`

Did it help temporarily?

JSo10, 2022-05-16:
the auto-build marlin is saying to update to the latest version. after downgradeing.
[image]

TD-er, 2022-05-16:
> We found a bug on our side and will provide a quick fix soon. Sorry.
>
> While this issue has not been fixed, please downgrade temporary to PlatformIO Cotr 5.0 via opening PlatformIO Core CLI in VSCode [...]

Nope, have been trying for over an hour but I don't seem to be able to reliably let PIO build.
Had it able to build once using PIO6, but then do a clean and build again and it failed again.
I now have reverted to PIO 5.2.5, but still see these linker errors cause by excluded folders that were built anyway.
Have deleted the mentioned folders, but still it fails to build.

Any ETA on when this might be fixed?

aufdenkampe, 2022-05-16:
I've had no end of troubles today trying to get projects to build with PIO6. 
The most challenging thing is that the errors aren't even reproducible; I seem to get a different set of errors each time I try to rebuild everything (by deleting the `.pio` folder and restarting VSCode).

I'll try downgrading with `pip install "platformio<6`

jbaiao, 2022-05-16:
> We found a bug on our side and will provide a quick fix soon. Sorry.
>
> While this issue has not been fixed, please downgrade temporary to PlatformIO Cotr 5.0 via opening PlatformIO Core CLI in VSCode [...]

Yes, this work!
Tks!

SikVac, 2022-05-17:
Platformio 6 is on beta and incompatible with Marlin ! Go back to 5.2.5 !
[image]
pip install platformio==5.2.5

ivankravets, 2022-05-17:
Guys, sorry for the issue. Could we ask you to re-test the latest development version?

1. Open PlatformIO Core CLI
2. Type `pio upgrade --dev`
3. Restart VSCode.

If it works for you, we will release PIO Core 6.0.1.

Thanks.

closed 2022-05-17 by ivankravets

MakeNPrint, 2022-05-17:
@ivankravets not for me (Mac) still giving 

collect2: error: ld returned 1 exit status
*** [.pio/build/BIGTREE_SKR_PRO/firmware.elf] Error 1

ivankravets, 2022-05-17:
[image]

Which version of Marlin do you use?

ivankravets, 2022-05-17:
[image]

Tested both envs with the latest stable Marlin 2.0.

MakeNPrint, 2022-05-17:
@ivankravets I also just tried deleting below again and no luck. 
.platformio/platforms
.platformio/packages

That one was Marlin 2.7.2 STM32 as I was testing yesterday 2.9.3 wasnt working so I tried earlier version.
Tried Marlin 2.9.3 STM32 as well no luck causing Marlin errors
Tried Marlin 2.9.3 STM32 - Different board and MCU. It works
Tried Marlin 2.9.2 LPC1769 and it works

So 2.7.2 not working.

First 2.9.3 test was a fresh download config example. So maybe a parameter in Configuartion files for Marlin I missed??

Marlin 2.9.2 - Working

Marlin 2.9.3 2nd Test known working config files - Working

MakeNPrint, 2022-05-17:
@ivankravets 
Marlin 2.9.3 - BTT SKR Pro V1.2 (default_envs = BIGTREE_SKR_PRO) (STM32F407ZGT6)
Fails compiling

collect2: error: ld returned 1 exit status
*** [.pio/build/BIGTREE_SKR_PRO/firmware.elf] Error 1

ivankravets, 2022-05-17:
I've created a PR that switches Marlin to the PIO Core 6.0. I don't see any failed builds https://github.com/MarlinFirmware/Marlin/actions/runs/2338187390

@MakeNPrint could you provide me a PlatformIO Project to reproduce your issues?

Sonicboom247, 2022-05-17:
just followed this and is working with Marlin 2.0.9.3 Bugfix from last night! if you don't restart VSCode you will still see the warnings but it will compile. Thank you!

Open PlatformIO Core CLI
Type pio upgrade --dev

ivankravets, 2022-05-17:
Yes, please ignore yellow warnings.

MakeNPrint, 2022-05-17:
@ivankravets Ok, to ensure it wasnt ghosting errors, I downloaded fresh Marlin and been manually copying over the settings and placed it into a new folder path.

It now compiles with out any red errors.

mrrstrat, 2022-05-17:
Hey gang - this threw me for a 24 hour loop. There is a fleet of production and lab machines that like to have this working :-)

So where does this leave things: is this a stable fix or will it require another release to fully correct?

Haiang, 2022-05-17:
Confirm pio upgrade --dev
It worked thanks
[image]
.

xrunner, 2022-05-17:
**pio upgrade --dev**

^^^ fixed it here for me too!

ivankravets, 2022-05-17:
The PlatformIO Core 6.0.1 is out! https://github.com/platformio/platformio-core/releases/tag/v6.0.1

Please upgrade via `pio uprade`.

Sonicboom247, 2022-05-17:
updated successfully but still getting warnings

[image]

[image]
compiling successfully so don't really care about the warnings, thank you!

MakeNPrint, 2022-05-17:
> The PlatformIO Core 6.0.1 is out! https://github.com/platformio/platformio-core/releases/tag/v6.0.1
> 
> Please upgrade via `pio uprade`.

Awesome even Marlin 2.7.2 is working, thank you for your hard work.

aufdenkampe, 2022-05-17:
Although 6.0.1 worked for me to build my project again, it appears that I can no longer have PlatformIO enabled for both Atom and VSCode with the "Use Builtin PIOCore" set to true for each. This worked for me previously, and was very helpful. However, now when I open one IDE after the other, it wants to reinstall PIO and this has led to lots of problems (and me uninstalling everything and starting from scratch).

Is there a solution to this?

mrrstrat, 2022-05-18:
Good job on this! Upgraded to PlatformIO Core 6.0.1  and all is working :-)

aufdenkampe, 2022-05-19:
@ivankravets, these fixes are still not working for me.

It seemed to compile my code, but only the first time after reinstalling it (which I had to do frequently with the weird issue with Atom).

The second time compiling after a clean reinstall gives me this error, every time.

Note that I'm running on Mac OS 12.3 and have Anaconda 2022-05 installed for Python.

The exact same code and .ini file worked for me last week.

```bash
Processing mayfly (board: mayfly; platform: atmelavr; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/mayfly.html
PLATFORM: Atmel AVR (3.4.0) > EnviroDIY Mayfly
HARDWARE: ATMEGA1284P 8MHz, 16KB RAM, 127KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
Converting RCB-4.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- EnviroDIY_ModularSensors @ 0.32.2
|   |-- EnviroDIY_DS3231 @ 1.3.4
|   |   |-- Wire @ 1.0
|   |-- EnableInterrupt @ 1.1.0
|   |-- SdFat @ 2.1.2
|   |   |-- SPI @ 1.0
|   |-- TinyGSM @ 0.11.5
|   |-- PubSubClient @ 2.8.0
|   |-- Adafruit BusIO @ 1.7.5
|   |   |-- Wire @ 1.0
|   |   |-- SPI @ 1.0
|   |-- Adafruit Unified Sensor @ 1.1.5
|   |-- Adafruit ADS1X15 @ 1.2.0+sha.7d67b45
|   |   |-- Wire @ 1.0
|   |-- Adafruit AM2315 @ 2.2.1
|   |   |-- Adafruit BusIO @ 1.7.5
|   |   |   |-- Wire @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- Wire @ 1.0
|   |-- Adafruit BME280 Library @ 2.1.4
|   |   |-- Adafruit Unified Sensor @ 1.1.5
|   |   |-- SPI @ 1.0
|   |   |-- Wire @ 1.0
|   |-- DHT sensor library @ 1.4.3
|   |   |-- Adafruit Unified Sensor @ 1.1.5
|   |-- Adafruit INA219 @ 1.1.1
|   |   |-- Wire @ 1.0
|   |   |-- Adafruit BusIO @ 1.7.5
|   |   |   |-- Wire @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- Adafruit MPL115A2 @ 1.1.3
|   |   |-- Wire @ 1.0
|   |-- OneWire @ 2.3.6
|   |-- DallasTemperature @ 3.9.1
|   |   |-- OneWire @ 2.3.6
|   |-- SDI-12 @ 2.1.4
|   |-- MS5803 @ 0.1.2
|   |   |-- Wire @ 1.0
|   |-- SensorModbusMaster @ 0.6.8
|   |-- KellerModbus @ 0.2.2
|   |   |-- SensorModbusMaster @ 0.6.8
|   |-- YosemitechModbus @ 0.2.5
[... 70 lines cut ...]
Fatal error: can't create .pio/build/mayfly/src/RCB-4.ino.cpp.o: No such file or directory
Assembler messages:
Fatal error: can't create .pio/build/mayfly/lib58c/SdFat/FatLib/FatFile.cpp.o: No such file or directory
*** [.pio/build/mayfly/src/RCB-4.ino.cpp.o] Error 1
*** [.pio/build/mayfly/lib58c/SdFat/FatLib/FatFile.cpp.o] Error 1
FileNotFoundError: [Errno 2] No such file or directory: '/Users/aaufdenkampe/Documents/Arduino/EnviroDIY_deployments/.pio/build/mayfly/.sconsign36.tmp':
  File "/Users/aaufdenkampe/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/Main.py", line 1401:
    _exec_main(parser, values)
  File "/Users/aaufdenkampe/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/Main.py", line 1365:
    _main(parser)
  File "/Users/aaufdenkampe/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/Main.py", line 1134:
    nodes = _build_targets(fs, options, targets, target_top)
  File "/Users/aaufdenkampe/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/Main.py", line 1340:
    jobs.run(postfunc = jobs_postfunc)
  File "/Users/aaufdenkampe/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Job.py", line 110:
    postfunc()
  File "/Users/aaufdenkampe/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/Main.py", line 1337:
    SCons.SConsign.write()
  File "/Users/aaufdenkampe/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/SConsign.py", line 133:
    syncmethod()
  File "/Users/aaufdenkampe/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/dblite.py", line 148:
    with self._open(self._tmp_name, "wb", self._mode) as f:
=============================================== [FAILED] Took 4.13 seconds ===============================================
The terminal process "platformio 'run'" terminated with exit code: 1.
```

TD-er, 2022-05-19:
To me it looks like a lib_deps is missing one of the libraries you need (or maybe there is a naming conflict with another library or fork) or it can't find a lib you have in your project.

adquarters, 2022-05-19:
After 3 days of trying I still can not compile.
I have a **Creality CR10 S5** and desperately need to install the firmware.
Can anyone compile the code and send me the install file to put on my SD card????

ivankravets, 2022-05-19:
@adquarters please share the full output from the building process.

adquarters, 2022-05-19:
> Executing task: platformio run <

Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Processing mega2560 (board: megaatmega2560; platform: atmelavr@~3.4; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- EEPROM @ 2.0
|-- SoftwareSerial @ 1.0
Building in release mode
Linking .pio/build/mega2560/firmware.elf
/var/folders/7q/17449rn12ylfh4npzdbcmq800000gn/T//cctM5dZd.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0xe0): undefined reference to `setup'
<artificial>:(.text.startup+0xe8): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/mega2560/firmware.elf] Error 1
================================================== [FAILED] Took 5.05 seconds ==================================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:05.049
============================================= 1 failed, 0 succeeded in 00:00:05.049 =============================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

ivankravets, 2022-05-19:
Have you read this thread from the beginning?

mrrstrat, 2022-05-19:
> To me it looks like a lib_deps is missing one of the libraries you need (or maybe there is a naming conflict with another library or fork) or it can't find a lib you have in your project.

Other languages can generate such an error if the interface references within libs has changed in some unaccomodatable way.

adquarters, 2022-05-19:
Yes but I am definitely not a coder

mrrstrat, 2022-05-19:
> Yes but I am definitely not a coder

We still like you anyway :-)

ivankravets, 2022-05-19:
You need to upgrade PlatformIO Core to the latest stable version. The instruction is added above. Simply, click on the PlatformIO icon in VSCode (left activity bar) => Misc... => Upgrade PlatfomIO Core. 

Restart VSCode.

Does it work now?

adquarters, 2022-05-19:
doing it now

adquarters, 2022-05-19:
SUCCESS!!!!!

adquarters, 2022-05-19:
Hey Ivan, Thank you very much! I have uploaded the firmware and it seems to be working. I'm sure my issue was simple but as I said I an no coder but I'm one hell of a 3D printer. 
THANKS!!!

Sal

ivankravets, 2022-05-19:
Glad that it works now. Yes, please ignore warnings, they don't have any impact on the final firmware. The Marlin team works on fixing these warnings, see https://github.com/MarlinFirmware/Marlin/pull/24205

manuugonz, 2023-01-08:
@ivankravets I have been following the thread from the beginning and am currently upgraded to the latest more stable version 6.1.5.

But it keeps giving me these problems:

> Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead

Does anyone know how to solve this problem?

atmassie476, 2023-01-16:
I feel your Pain Im having the same problem this is so over my head. I have a hard time trying  to edit and compile firmware for my printers but when you start talk about digging around in VS code to fix it my eyes glaze over and I fall into a mild coma lol.

LeonaK, 2023-01-31:
This is NOT repaired.  I have been trying to compile files that was compiling perfectly well for months.  Now, NOthing compiles.  I have reloaded 5.2.5,, no help. I updated to the newest version, no help. This is just awful.  I imagine 3d printers all over the country stalled because of YOUR problem.
I understand that mistakes can be made, but to just go on for months, pretending that the problem is solved is shameful.

 *  Executing task in folder G:: C:\Users\klajd\.platformio\penv\Scripts\platformio.exe run --target clean 

*******************************************************************************************************************************************************Obsolete PIO Core v6.1.6 is used (previous was 6.1.7a1)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
*******************************************************************************************************************************************************Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` 
instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Processing BIGTREE_SKR_2_F429 (board: marlin_STM32F429VGT6; platform: ststm32@~12.1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\klajd\\.platformio\\packages\\framework-arduinoststm32\\variants\\MARLIN_F4x7Vx\\hal_conf_extra.h':
  File "C:\Users\klajd\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript(item, exports="env")
  File "C:\Users\klajd\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\klajd\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "G:\buildroot\share\PlatformIO\scripts\generic_create_variant.py", line 48:
    shutil.rmtree(variant_dir)
  File "C:\Users\klajd\.platformio\python3\lib\shutil.py", line 749:
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\klajd\.platformio\python3\lib\shutil.py", line 627:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\klajd\.platformio\python3\lib\shutil.py", line 625:
    os.unlink(fullname)
============================================================= [FAILED] Took 3.62 seconds =============================================================

Environment         Status    Duration
------------------  --------  ------------
BIGTREE_SKR_2_F429  FAILED    00:00:03.618
======================================================== 1 failed, 0 succeeded in 00:00:03.618 ========================================================
 *  The terminal process "C:\Users\klajd\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'clean'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task in folder G:: C:\Users\klajd\.platformio\penv\Scripts\platformio.exe run --target clean 

[... cut ...]
============================================================= [FAILED] Took 1.73 seconds =============================================================

Environment         Status    Duration
------------------  --------  ------------
BIGTREE_SKR_2_F429  FAILED    00:00:01.732
======================================================== 1 failed, 0 succeeded in 00:00:01.732 ========================================================
 *  The terminal process "C:\Users\klajd\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'clean'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

LeonaK, 2023-03-09:
> I am also experiencing this issue today.
> I have tried to downgrade PIO Core to 5.2.5 but then when I try to build it says 5.2.5 is deprecated.
>
> Is there a new solution or fix that works?

I assume you are using VSCode?  The problem is not Marlin or VSCode.  It's PlatformIO.  erase it and reinstall a new version.  Be sure to get any references (Platformio, PIO...) in Users/Your User..
Then, you will have to reload and use the new Marlin2.0. 

Sorry that Marlin won't help.

ivankravets, 2023-03-09:
1. Be sure that you use the latest [Marlin Firmware](https://github.com/MarlinFirmware/Marlin)
2. Remove `%USERHOME%/.platformio` folder
3. Restart VSCode.

KyleCurrier, 2023-06-11:
Reinstalll Platform IO does not fix it. Still fails on Marlin build with most recent updates. Everything worked fine before. I have not touched it. I needed to change a calibration value which is a very simple thing. Build, and now it fails. SO I go back, surely its not because I changed a value, thats ridiclous... I revert to what I KNOW worked before. Failed again.... Yall broke something and none of the stuff you're suggesting is fixing it. I should have blocked VSC auto updating services, I know better.... 

Error as follows.... 

"Converting Marlin.ino
cc1plus.exe: fatal error: opening output file C:\Users\DevWolf\Desktop\DevWolfs Creality 4.2.7 board- firmware v213\Firmware\Marlin\Marlin.ino.cpp:
compilation terminated."

Full console log from build below > 

Executing task: C:\Users\DevWolf\.platformio\penv\Scripts\platformio.exe run 

Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_s
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filt
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `bu
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_s
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please us
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please us
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_sr
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `b
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_f
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_f
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_f
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_f
Processing STM32F103RCT6_creality (platform: ststm32@~12.1; board: genericSTM32F103RC; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'buildroot\share\PlatformIO\scripts\common-dependencies.py'
File "C:\Users\DevWolf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168, in <module>

warning: Ignoring missing SConscript 'buildroot\share\PlatformIO\scripts\common-cxxflags.py'
File "C:\Users\DevWolf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168, in <module>

warning: Ignoring missing SConscript 'buildroot\share\PlatformIO\scripts\preflight-checks.py'
File "C:\Users\DevWolf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168, in <module>

warning: Ignoring missing SConscript 'buildroot\share\PlatformIO\scripts\stm32_serialbuffer.py'
File "C:\Users\DevWolf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168, in <module>

warning: Ignoring missing SConscript 'buildroot\share\PlatformIO\scripts\generic_create_variant.py'
File "C:\Users\DevWolf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168, in <module>

warning: Ignoring missing SConscript 'buildroot\share\PlatformIO\scripts\random-bin.py'
File "C:\Users\DevWolf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168, in <module>
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.10900.200819 (1.9.0)
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
cc1plus.exe: fatal error: opening output file C:\Users\DevWolf\Desktop\DevWolfs Creality 4.2.7 board- firmware v213\Firmware\Marlin\Marlin.ino.cpp:
compilation terminated.
*** Error 1
AssertionError: :
  File "C:\Users\DevWolf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\DevWolf\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
[... cut ...]
===================================================================================== [FAILED] Took 0.69 seconds ==================================
Environment             Status    Duration
----------------------  --------  ------------
STM32F103RCT6_creality  FAILED    00:00:00.688
=============================================================================== 1 failed, 0 succeeded in 00:00:00.688 =============================

 *  The terminal process "C:\Users\DevWolf\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

franmuzi1, 2023-06-17:
Still havingf the same issue, i have to use an old version of marlin because of the bug of the new ones for my printer, is there any solution?

JPU571, 2023-07-20:
I updated the platformio.ini file and replaced all instances of ' src_filter ' with 'build_src_filter'  AND replaced 'src_build_flags' with 'build_src_flags' and resolved the warnings. Not able to compile yet, but no errors

thinkyhead, 2023-07-20:
This issue can be closed. It's purely a Marlin issue whether the `platformio.ini` and other `.ini` files are prepared for PlatformIO 6 and up. See the [Marlin Releases page](https://github.com/MarlinFirmware/Marlin/releases) for legacy versions of Marlin patched for PlatformIO 6 and up.

commit b764a222 (2022-05-17): Improved support for the renamed configuration options // Resolve #4270