platformio/platformio-core #4271: Broken after v6.0 upgrade

hmronline, 2022-05-17:
Started failing as soon as v6.0 was installed. Going back to v5 does work as expected.
Tested in both Linux and Mac, everything fails like this:

```
> Executing task: platformio run --target clean --environment mini_ota <

Can not remove temporary directory `/Users/h/Develop/b/project/firmware/.pio/build`. Please remove it manually to avoid build issues
Error: Traceback (most recent call last):
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super().invoke(ctx)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 142, in cli
    process_env(
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 184, in process_env
    print_processing_header(name, config, verbose)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 218, in print_processing_header
    for k, v in config.items(env=env):
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 224, in items
    return [(option, self.get(section, option)) for option in self.options(section)]
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 224, in <listcomp>
    return [(option, self.get(section, option)) for option in self.options(section)]
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 341, in get
    value = self.getraw(section, option, default)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 242, in getraw
    return self._getraw(section, option, default)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 303, in _getraw
    return self._expand_interpolations(section, value)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 312, in _expand_interpolations
    return self.VARTPL_RE.sub(
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 313, in <lambda>
    lambda match: self._re_interpolation_handler(parent_section, match), value
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 329, in _re_interpolation_handler
    value = self.get(section, option)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 341, in get
    value = self.getraw(section, option, default)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 242, in getraw
    return self._getraw(section, option, default)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 303, in _getraw
    return self._expand_interpolations(section, value)
  File "/Users/h/.platformio/penv/lib/python3.8/site-packages/platformio/project/config.py", line 312, in _expand_interpolations
    return self.VARTPL_RE.sub(
TypeError: sequence item 1: expected str instance, int found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process "platformio 'run', '--target', 'clean', '--environment', 'mini_ota'" terminated with exit code: 1.

```

ivankravets, 2022-05-17:
Could you share your `platformio.ini`?

hmronline, 2022-05-17:
The directory .pio/build does not exist and even if it exist the issue is the same.
As stated above, if I downgrade platformio to v5 it works as expected, so this is an issue introduced with v6, regardless of platformio.ini contents.

Having said that, here is my platformio.ini:
```
; PlatformIO Project Configuration File

[platformio]
description = Firmware
default_envs = mini_ota

[env]
;targets = upload, monitor
framework = arduino
board_build.filesystem = littlefs

[env:esp8266]
platform = espressif8266 @ ^3.2.0
monitor_speed = 74880
build_flags = 
	-D SERIAL_BAUD_RATE=${env:esp8266.monitor_speed}

[env:mini_ota]
extends = env:esp8266
board = esp01
lib_deps = 
build_flags = 
	${env:esp8266.build_flags}
	-D FEATURE_SERIAL_OUTPUT_ENABLED
	-D LOG_LEVEL_INFO
	-D LOG_LEVEL_NOTICE
	-D LOG_LEVEL_WARNING
	-D LOG_LEVEL_ERROR
	-D FEATURE_WIFI_ENABLED
	-D FEATURE_WIFI_MANDATORY
	-D FEATURE_WIFI_CLIENT_SERVICES_ENABLED

```

Thanks for your help!

oomek, 2022-05-17:
I'm experiencing the same issues.

```
> Executing task: C:\Users\Radek\.platformio\penv\Scripts\platformio.exe run <

Can not remove temporary directory `D:\Documents\GitKraken\HELLO\.pio\build`. Please remove it manually to avoid build issues
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: baremetal)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'D:\\Documents\\GitKraken\\HELLO\\.pio\\build':
  File "C:\Users\Radek\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    os.makedirs(env.subst("$BUILD_DIR"))
  File "C:\Users\Radek\.platformio\python3\lib\os.py", line 215:
    makedirs(head, exist_ok=exist_ok)
  File "C:\Users\Radek\.platformio\python3\lib\os.py", line 225:
    mkdir(name, mode)
========================================================== [FAILED] Took 0.54 seconds ==========================================================
The terminal process "C:\Users\Radek\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
There is no build project in there, I removed it.

Here is a clue:
I have all my personal folders redirected in Windows 11 to D drive
Documents pionts to D:\Documents etc
When I move the project folder to any other folder on C or D drive this problem during building doesn't occur.

Core 6.0.1a1  Home 3.4.1

ivankravets, 2022-05-17:
Thanks for the report. Please re-test with `pio upgrade --dev`.

closed 2022-05-17 by ivankravets

oomek, 2022-05-17:
Thanks for a quick response, but that didn't work for me unfortunately.

ivankravets, 2022-05-17:
@oomek your issue is not related to the PlatformIO Core. Try to remove `D:\Documents\GitKraken\HELLO\.pio` folder, PULL THE LATEST wizio dev-platform (it has bug that was a few hours ago fixed), and try building again.

oomek, 2022-05-17:
I was the one that reported this issue. The fix applied.
My issue is also relevant to `.pio/build` folder error

ivankravets, 2022-05-17:
Please report your issue to https://community.platformio.org/

commit abf63048 (2022-05-17): Fixed an issue when using "Interpolation of Values" and merging str+int options // Resolve #4271