platformio/platformio-core #4272: Out of memory (>60G) from scons.py compiling Marlin

larroy, 2022-05-17:
What kind of issue is this?

I tried to compile Marlin from my personal fork https://github.com/larroy/Marlin/tree/pedro_bugfix which was working fine in platformIO for lpc1769

After a few days, with no changes to the code (tried several revisions back which I had previously compiled) scons.py keeps ballooning and using all the memory available. This happened in Windows 10 and also Linux. I tried uninstalling Platform IO several times, cleaning up data and home directories, switching to PIO in Linux...

This is the process which is eating all the memory:

```

piotr@panther:0: ~/d/Marlin [(HEAD detached at d6f21475a1)]> ps aux | grep scons
piotr      15942 99.8 23.6 20527848 15540704 pts/3 R+ 22:54   1:16 /home/piotr/.platformio/penv/bin/python /home/piotr/.platformio/packages/tool-scons/scons.py -Q --warn=no-no-parallel-support --jobs 32 --sconstruct /home/piotr/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py PIOVERBOSE=0 ISATTY=1 PIOENV=TFBDMTc2OQ== PROJECT_CONFIG=L2hvbWUvcGlvdHIvZGV2ZWwvTWFybGluL3BsYXRmb3JtaW8uaW5p PROGRAM_ARGS=W10= PLATFORM_MANIFEST=L2hvbWUvcGlvdHIvLnBsYXRmb3JtaW8vcGxhdGZvcm1zL254cGxwYy1hcmR1aW5vLWxwYzE3NngvcGxhdGZvcm0uanNvbg== BUILD_SCRIPT=L2hvbWUvcGlvdHIvLnBsYXRmb3JtaW8vcGxhdGZvcm1zL254cGxwYy1hcmR1aW5vLWxwYzE3NngvYnVpbGRlci9tYWluLnB5
piotr      16356  0.0  0.0   8552  2440 pts/1    S+   22:55   0:00 grep --color=auto scons
```

In windows is also Python but I don't know how to see the commandline.

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:   Windows 10, Linux Ubuntu 21.10

**PlatformIO Version** (`platformio --version`):   v2.4.3

```

piotr@panther:0: ~> .platformio/penv/bin/platformio --version
PlatformIO Core, version 6.0.0

```
### Description of problem

See above

#### Steps to Reproduce

1. Open project
2. Go to PIO tab,
3. Build for LPC1769

### Actual Results

Out of memory.

### Expected Results

Successful build

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

See  https://github.com/larroy/Marlin/blob/pedro_bugfix/platformio.ini

**Source file to reproduce issue:**
See linked branch above.

### Additional info

lvlie, 2022-05-17:
Same issue here on Windows with fresh install. Python eats up all memory (~30Gb) a couple of times (causing system to have a high cpu and disk activity, for swap?) and eventually:
```
MemoryError: :
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 182:
    env.SConscript(item, exports="env")
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
```

ivankravets, 2022-05-17:
Could you re-run `pio upgrade --dev`. Does it work now?

See https://github.com/MarlinFirmware/Marlin/actions/runs/2338187390

AJO3d, 2022-05-17:
@ivankravets it was my problem this morning. Like I said in the post https://github.com/MarlinFirmware/Marlin/pull/24194 the problem was solved after update to the dev version

AJO3d, 2022-05-17:
Yes, it was exactly my problem: MemoryError. I searched in the net any problem by this, but I found nothing and then I searched by the warnings in log and I found the post where you recommends update to the dev version.

ivankravets, 2022-05-17:
Thank you so much for the clarification! 

Duplicate of https://github.com/platformio/platformio-core/issues/4270

lvlie, 2022-05-17:
It starts building now, but I have errors like:
`error: 'manual_feedrate_mm_s' was not declared in this scope`. Which is a totally different error of course. Reckon the original issue is fixed with the 6.0.1

ivankravets, 2022-05-17:
> error: 'manual_feedrate_mm_s' was not declared in this scope

Yes, it seems this is not related to the PlatormIO.

lvlie, 2022-05-17:
Yeah, apparently theres a bugfix and release version with the same version, and they are incompatible. Works now, thanks!