platformio/platformio-core #4277: Platform components are build with the project source build flags

nomis, 2022-05-18:
### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`): 6.0.1

### Description of problem

PlatformIO is building platform components with the project source build flags.

#### Steps to Reproduce

1. git clone https://github.com/nomis/scd30 -b test
2. platformio run -e s2_mini -v

There are defines included in every `build_flags` and `src_build_flags` used by this environment to illustrate the problem.

### Actual Results

Platform component LittleFS.cpp is compiled with `build_flags` and `src_build_flags` flags.

```
23:56:52.829  xtensa-esp32s2-elf-g++ -o .pio/build/s2_mini/lib8c1/LittleFS/LittleFS.cpp.o -c ... \
  -DPIO_COMMON_BUILD_FLAGS -DPIO_S2_MINI_BUILD_FLAGS \
  -DPIO_COMMON_SRC_BUILD_FLAGS -DPIO_S2_MINI_SRC_BUILD_FLAGS \
  ...
```

### Expected Results

Platform component LittleFS.cpp is compiled with only `build_flags` flags.

### Additional info

The project source build flags include `-Wall -Wextra -Werror` which causes the build to fail because the platform components have errors in them.

nomis, 2022-05-19:
@ivankravets You've left debug print statements in the commit.

ivankravets, 2022-05-19:
Thanks! Fixed. Does the latest build work for you? 

```
pio upgrade --dev
```

nomis, 2022-05-19:
> Thanks! Fixed. Does the latest build work for you?

Yes

ivankravets, 2022-05-26:
@richcj10 please forward your question to https://community.platformio.org/

commit 06ed9ba7 (2022-05-19): Fixed an issue when "build_src_flags" were applied outside project scope // Resolve #4277