platformio/platformio-core #4279: Native unit tests from 5.x no longer run on 6.0.0 or higher, excluded tests run instead

kensettleatwagz, 2022-05-20:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu:18.04

**PlatformIO Version** (`platformio --version`):
6.0.0, 6.0.1, 6.0.2a

### Description of problem
Native unit tests no longer work, excluded embedded tests are run, and they fail, breaking our CI process. Every build fails the unit tests. The native tests normally pass, but they aren't run.

#### Steps to Reproduce

1. Build code, get-platformio.py pulls the master release 6.0.0 or higher.
2. run native unit tests: pio test -e native -v
3. Unit tests fail as detailed in Description of problem, platformio.ini unit test environment worked with 5.x

### Actual Results
Unit tests fail. The embedded unit tests are not expected to be able to run natively.

### Expected Results
Unit tests work as before. Changes to platformio.ini should not required to maintain backwards-compatibility. That is, by definition NOT BACKWARDS COMPATIBLE!

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
default_envs = esp32dev

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
upload_port = COM18
monitor_port = COM18
monitor_speed = 115200
monitor_flags = 
    --filter 
    direct
board_build.partitions = partitions.csv
build_flags = 
    -DFW_VERSION="\"v1.0\""
    -DCORE_DEBUG_LEVEL=5

[env:native]
platform = native
lib_ignore =
           Kernel
           KernelInterface
           HttpsClient
           Timer
           BleService
test_ignore = test_embedded
lib_extra_dirs =
               ./src
               ${sysenv.IDF_PATH}/components/json/
               ./lib/WiFi
build_flags = 
    -lm
```

**Source file to reproduce issue:**
```cpp
// Any valid Unity unit test in ./test/test_desktop/* does not get run
```

### Additional info
Tried released versions and --dev version. Nothing improved. Tried many different ways to get back to the last stable release. Nothing worked (always ended up still at 6.0.x).

I'm sure that we're not the only ones who are seeing this issue. This is a good case for versioning in your ini file, and having an actual backwards-compatibility mode for existing deployments, or at least have the common decency to allow people to downgrade to the last stable version if the new version is breaking their builds!

ivankravets, 2022-05-20:
Could you provide an output of `tree ./test` ?

kensettleatwagz, 2022-05-20:
```tree ./test
./test
├── README
├── test_desktop
│   ├── README
│   ├── TestDebug.c
│   ├── TestDebug.h
│   ├── TestTimeConverter.c
│   ├── TestTimeConverter.h
│   ├── TestWiFiNetworkStorage.c
│   └── TestWiFiNetworkStorage.h
├── test_device_settings.json
└── test_embedded
    ├── test_audio
    │   └── test_audio.c
    └── test_hw_info
        └── test_hw_info.c
```

The platformio.ini file has "test_ignore = test_embedded" yet those are the only tests that run. The intended tests in test_desktop (which were formerly run in [env:native]) are no longer run. This has been working correctly for years, and the regression was introduced starting in 6.0.0.

It may be that others are being impacted by the fact that some of their CI unit tests silently no longer being run. Unless they look at their logs, they might not realize that since PlatformIO Core version 6.0.0 was released, *some* of their unit tests are no longer being run. This version was thrust upon us without our consent, and there seems to be no way to get back, unless you want to document a proper way to go back, or deploy a fix for the root cause of the problem.

ivankravets, 2022-05-20:
The group of tests should not start with `test_` prefix. That was a bug in PIO Core 5.0. Please rename `test/test_embedded` to the `test/embedded`

ivankravets, 2022-05-20:
See also docs with an example https://docs.platformio.org/en/latest/advanced/unit-testing/structure.html#test-hierarchy

kensettleatwagz, 2022-05-20:
> See also docs with an example https://docs.platformio.org/en/latest/advanced/unit-testing/structure.html#test-hierarchy

Those docs apply to new behavior for the new version. **Can you explain why the tests in test_embedded ARE being run, when "test_ignore = test_embedded"?** That seems like a bug, even after reading those docs, since your latest documents say that we should not use test_ prefixes any more, and they shouldn't work.

Also, other people *will* be affected by this, even though you consider the fact that test_ prefixes work OK to be a bug in 5.0, it still worked up until now. If 6.0 is backwards-compatible, it should be able to tolerate the "bug" in 5.0, rather than changing behavior. That is a breaking change. Anybody who followed the older examples and set it up with test_ prefixes will suddenly have their unit tests stop running, unless they ask for them to be ignored, in which case, they **will** run, when they shouldn't.

Apparently the only option given to me at this time is to Close with Comment, but I do NOT believe that this issue should be closed, because the problem IS NOT FIXED, and there are no workarounds given, short of changing our filenames, directory names and platformio.ini file, which we cannot do retroactively for our entire repo.

Please keep in mind that the disruption that is caused by this problem will affect people who try to rebuild older versions of their code from source, since they will need to update their file structures and platformio.ini files in order to get their unit tests to work correctly.

Not having any option to use an older build of PlatformIO Core that worked correctly and not fixing an obvious bug is not acceptable. As I have said many times PIO Core 6.0 IS NOT BACKWARDS COMPATIBLE, so please stop pretending that it is (or fix it so that it's actually backwards compatible). I hope that someone else who is affected by this issue picks up the torch and continues this discussion (at this point, you may need to file new tickets), because this is not an ideal situation for anyone. People are using PlatformIO as an important part of their production flow. Please take it more seriously! Thanks!

ivankravets, 2022-05-20:
Thanks, we will update the migration guide and a note about this case. Sorry.

kensettleatwagz, 2022-05-20:
A better fix for this would be to make get-platformio.py take an argument to specify that you want 6.0 or higher, and if that argument isn't passed, then revert to the last legacy 5.x release. Then people can upgrade when they want the improvements of 6.0, and their older repos won't be inexplicably broken if they try to rebuild an old version. 

Also, hopefully you will change your docs to indicate that it PIO Core 6.0 is not backwards-compatible, because it isn't.

ivankravets, 2022-05-20:
See related commit https://github.com/platformio/platformio-docs/commit/5c3659ac2fb3bd27e54ca0fe3d151c7a200158d8

So, we never recommend using grouped test suites without the `test_` prefix. A lot of tutorials and examples that were created 6 years ago contained the `test_` prefix. We saw on the internet that people use Unit Testing with bad practice. We could not solve this issue in PIO Core 5.0 due to the limited architecture of the Unit Testing solution that was introduced in 2016 as a concept.

The new PIO Core 6.0 has a new refactored a scratch Unit Testing engine. Hope the note in the migration guide will help people to save time on debugging this isue.

commit 626640cc (2022-05-20): Accurate information about broken test suites in PIO Core 5.0 // Resolve #4279
closed 2022-05-20 by ivankravets