platformio/platformio-core #4282: Error: Unknown development platform

torntrousers, 2022-05-21:
I have some projects on the nano 33 iot board that no longer work and now fail with "Error: Unknown development platform "

I've tried with a completly new project, make a new directory, cd into it and do ```pio init`, edit platformio.ini to add:
```
[env:nano_33_iot]
platform = atmelsam
board = nano_33_iot
framework = arduino
```
then ```pio run``` gives:
```
pio run
Processing nano_33_iot (platform: atmelsam; board: nano_33_iot; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Error: Unknown development platform 'PackageItem <path=C:\Users\tornt\.platformio\platforms\atmelsam metadata=PackageMetaData <type=platform name=atmelsam version=7.1.0 spec={'owner': 'platformio', 'id': 8007, 'name': 'atmelsam', 'requirements': None, 'uri': None}'
```

Any ideas? Is this something related to the v6 release?

ikto-art, 2022-05-21:
I had a similar problem with the latest version of PlatformIO on VSCode (impossible to use board wt32-eth01 because PIO thought it had the latest packages) and I removed the folder C:\Users\ ... \\.platformio\platforms\espressif32.
After which I ran in the project folder `pio pkg update` but it said that everything was up to date.
Then I tried `pio pkg uninstall`, said the same thing.
Then I did `pio pkg install` and finally it reinstalled and updated everything (framework-arduinoespressif32 and platform espressif32). Now everything works.

Maybe you can try the same by deleting the atmelsam folder in the platforms folder and then either `pio pkg install` or do the 3 commands like I did.

ivankravets, 2022-05-21:
Yes, try to delete HOMEDIR/.platformio/platforms directory and build project again. 

Does it work now?

torntrousers, 2022-05-22:
yes deleting stuff did fix it.

closed 2022-05-22 by torntrousers