platformio/platformio-core #4283: Add board :- ESP-12F

ben700, 2022-05-21:
My board is actually a ESP-12F but I select ESP-12E as ESP-12F isn't listed as a known board in platformio 

Many Thanks

Ben

maxgerhardt, 2022-05-22:
Issues that refer to this: https://github.com/platformio/platform-espressif8266/issues/195, https://github.com/platformio/platform-espressif8266/issues/274

ivankravets, 2022-05-22:
Thanks, @maxgerhardt, for the solution.

closed 2022-05-22 by ivankravets

ben700, 2022-05-23:
How is this request the same as those issues?

ben700, 2022-05-23:
> Thanks, @maxgerhardt, for the solution.

Is that because the codebase isn't being maintained or because ESP-12F is so similar to esp12e?  

Ben

needfulthing, 2023-05-16:
The only difference between 12E and 12F seems to be an improved antenna design. See https://components101.com/article/major-differences-between-esp12e-and-esp12f