platformio/platformio-core #4286: Project build failures when migrating to v6.0.1

julianneswinoga, 2022-05-22:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**: Tested on manjaro, debian

**PlatformIO Version**: `5.2.5` vs `6.0.1`

### Description of problem
Hi! I'm a developer on the [Open Astro Tracker Firmware](https://github.com/OpenAstroTech/OpenAstroTracker-Firmware) and we've recently found an uptick in build failures for our users when using PlatformIO. After some digging I've found that our firmware compiles fine with PIO `5.2.5` but not on `6.0.1`. The reason I'm raising this is that it states on the [5.x to 6 migration guide](https://docs.platformio.org/en/latest/core/migration.html):
> PlatformIO Core 6.0 is FULLY BACKWARD COMPATIBLE* with PlatformIO 5.0 projects

Which doesn't seem to be the case for our firmware :sweat_smile: 

The issue lies somewhere in applying build flags selectively to source files but not libraries. Basically we turn on a lot of compiler warnings for our source code but we don't want those warnings enabled for libraries (obviously).

#### Steps to Reproduce
```bash
git clone https://github.com/OpenAstroTech/OpenAstroTracker-Firmware
cd OpenAstroTracker-Firmware
python3 -m venv .pio_env
.pio_env/bin/pip install platformio==6.0.1

# basic configuration setup just to compile things
cat <<EOF >> Configuration_local.hpp
#define RA_STEPPER STEPPER_TYPE_ENABLED
#define RA_DRIVER TMC2209_UART
#define RA_MOTOR_CURRENT_RATING 1
#define RA_OPERATING_CURRENT_SETTING 1
#define DEC_STEPPER STEPPER_TYPE_ENABLED
#define DEC_DRIVER TMC2209_UART
#define DEC_MOTOR_CURRENT_RATING 1
#define DEC_OPERATING_CURRENT_SETTING 1
EOF
.pio_env/bin/platformio run -e mksgenlv21
```

### Actual Results
Build failures from `-Werror` that should only be enabled on repo source files, but it's coming from libraries.

### Expected Results
Builds passing, because 6.x is supposed to be backwards compatible :smiling_face_with_tear:

PamudithaSomarathne, 2022-05-23:
I have also experienced this issue. VSCode had updated PlatformIO while I was away for a week and library imports stopped working with "Invalid simple block" error. Currently working after downgrading to PlatformIO core 5.2.5

TD-er, 2022-05-23:
In my project (ESPEasy), a number of users also report build issues.
See for example: https://github.com/letscontrolit/ESPEasy/issues/4068#issuecomment-1134222604

I myself noticed on my laptop that I was also not able to build anymore. Each build varied from scons39.dblite not found, missing setup() and loop() and another one, all alternating on each build attempt.
As suggested by one of my users, when building the default env. first, it seems to 'stabilize'

ivankravets, 2022-05-23:
Thanks for the report! Could you try the latest Development version via `pio upgrade —dev`?

This is bug and we fixed it. If the latest dev version works for you, we will release Core 6.0.2.

julianneswinoga, 2022-05-23:
@ivankravets Yep confirmed fixed! Sorry for not trying the dev channel :sweat_smile:

closed 2022-05-23 by julianneswinoga

ivankravets, 2022-05-23:
Thanks, @julianneswinoga!

Yes, sorry that our "FULLY BACKWARD COMPATIBLE" didn't work with Core 6.0.0. Hope finally Core 6.0.2 will fix all compatibility issues. 

It was the biggest release ever with a rich list of additions and changes. We would be thankful if you continue using DEV version of PlatformIO Core and report any issues. Typically, the DEV version is very stable, it is covered with a ton of integration and unit tests. However, planned refactoring might have some influence.

thomastech, 2022-05-24:
Too bad this issue is closed.

The latest PlatformIO Dev release (6.0.2a2) is better, but builds are still randomly successful. 

The common build error I see is scons39.dblite not found. I've noticed that when the problem occurs the \.pio\build\device folder is not being populated. This folder is cleared at each new build.

Every time I launch VS Code the project's build fails. But eventually success if I perform Clean operations and keep trying. And after a build success the issue does not re-appear until the next time I launch VS Code.

- Thomas

TD-er, 2022-05-24:
Yep, me too.
I simply can't seem to get my laptop to complete a single build.
I only had one build succeed, but that was when connected to power instead of running on battery. So maybe it is kinda timing related?
A few of those build errors are related to  the generated .cpp (Arduino code) being missing.
So perhaps VS code is also inspecting other PIO envs which in turn also generate and delete the .cpp file?

On my desktop PC (roughly 10x as fast as the laptop when considering compiling) I didn't have as many build issues, but there I already moved back to PIO 5.2.5 as I simply need that one to work.

ivankravets, 2022-05-24:
Does it work without VSCode using PlatformIO Core CLI?

thomastech, 2022-05-24:
@ivankravets thanks for monitoring this closed issue.
I'm sorry but I don't have any experience with building projects using the PlatformIO Core CLI. Maybe @TD-er can try this?

- Thomas

TD-er, 2022-05-24:
> Does it work without VSCode using PlatformIO Core CLI?

Can't remember if I tried this on the laptop. Will try within an hour as I go downstairs for another beer :)
I do suspect you mean using the CLI in VS code, so we're sure it is exactly the same Python env?

TD-er, 2022-05-24:
Cleaned everything on my laptop regarding PlatformIO, previously fetched lib_deps, etc.
Then VScode reinstalled PlatformIO 6.0.1 and after that (restart inbetween) I updated to PIO 6.0.2a

After starting a build via CLI in VScode, I got this error:
[image]
This does suggest it is time based to retry to build an index.
So that may explain why my laptop fails to build it over-and-over again as a build may easily take a few minutes (compared to 30 - 50 sec on my desktop PC)

Anyway, I now have these errors, suggesting some very strange lib_deps scan output, which I don't have on my desktop:
```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Install the latest PowerShell for new features and improvements! https://aka.ms/PSWindows

PS C:\GitHub\TD-er\ESPEasy> pio run -e max_ESP32_16M8M_LittleFS
Processing max_ESP32_16M8M_LittleFS (board: esp32_16M8M; framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip)
-------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-scons @ ~4.40300.0
Unpacking  [####################################]  100%
[WinError 145] The directory is not empty: 'C:\\Users\\gijsn\\.platformio\\packages\\tool-scons\\scons-local-4.3.0\\SCons' 
Please manually remove the file `C:\Users\gijsn\.platformio\packages\tool-scons\scons-local-4.3.0\SCons`
[WinError 145] The directory is not empty: 'C:\\Users\\gijsn\\.platformio\\packages\\tool-scons\\scons-local-4.3.0' 
Please manually remove the file `C:\Users\gijsn\.platformio\packages\tool-scons\scons-local-4.3.0`
[WinError 145] The directory is not empty: 'C:\\Users\\gijsn\\.platformio\\packages\\tool-scons'
Please manually remove the file `C:\Users\gijsn\.platformio\packages\tool-scons`
Tool Manager: Warning! Package Mirror: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\gijsn\\.platformio\\packages\\tool-scons'
Tool Manager: Looking for another mirror...
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40300.1 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
 Concat ./src/src/Commands/*.cpp to ./src/src/Commands_tmp\__tmpfile.cpp 
 Add:   ./src/src/Commands\Blynk.cpp
 Add:   ./src/src/Commands\Common.cpp
 Add:   ./src/src/Commands\Controller.cpp
 Add:   ./src/src/Commands\Diagnostic.cpp
 Add:   ./src/src/Commands\GPIO.cpp
 Add:   ./src/src/Commands\HTTP.cpp
 Add:   ./src/src/Commands\i2c.cpp
 Add:   ./src/src/Commands\InternalCommands.cpp
 Add:   ./src/src/Commands\MQTT.cpp
 Add:   ./src/src/Commands\Networks.cpp
 Add:   ./src/src/Commands\Notifications.cpp
 Add:   ./src/src/Commands\Provisioning.cpp
 Add:   ./src/src/Commands\RTC.cpp
 Add:   ./src/src/Commands\Rules.cpp
 Add:   ./src/src/Commands\SDCARD.cpp
 Add:   ./src/src/Commands\Servo.cpp
 Add:   ./src/src/Commands\Settings.cpp
 Add:   ./src/src/Commands\System.cpp
 Add:   ./src/src/Commands\Tasks.cpp
 Add:   ./src/src/Commands\Time.cpp
 Add:   ./src/src/Commands\Timer.cpp
 Add:   ./src/src/Commands\UDP.cpp
 Add:   ./src/src/Commands\wd.cpp
 Add:   ./src/src/Commands\WiFi.cpp
 Files not changed, will not touch __tmpfile.cpp
 -------------------------------
 Concat ./src/src/ControllerQueue/*.cpp to ./src/src/ControllerQueue_tmp\__tmpfile.cpp
 Add:   ./src/src/ControllerQueue\C011_queue_element.cpp
 Add:   ./src/src/ControllerQueue\C015_queue_element.cpp
 Add:   ./src/src/ControllerQueue\C016_queue_element.cpp
 Add:   ./src/src/ControllerQueue\C018_queue_element.cpp
 Add:   ./src/src/ControllerQueue\DelayQueueElements.cpp
 Add:   ./src/src/ControllerQueue\MQTT_queue_element.cpp
 Add:   ./src/src/ControllerQueue\queue_element_formatted_uservar.cpp
 Add:   ./src/src/ControllerQueue\queue_element_single_value_base.cpp
 Add:   ./src/src/ControllerQueue\SimpleQueueElement_string_only.cpp
 Files not changed, will not touch __tmpfile.cpp
 -------------------------------
[... 574 lines cut ...]
                 from .pio/libdeps/max_ESP32_16M8M_LittleFS/Adafruit GFX Library/Adafruit_SPITFT.h:25,
                 from .pio/libdeps/max_ESP32_16M8M_LittleFS/Adafruit GFX Library/Adafruit_SPITFT.cpp:36:
C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:209:10: fatal error: pins_arduino.h: No such file or directory        

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\max_ESP32_16M8M_LittleFS\lib345\Adafruit GFX Library\Adafruit_GFX.cpp.o] Error 1
*** [.pio\build\max_ESP32_16M8M_LittleFS\lib345\Adafruit GFX Library\Adafruit_SPITFT.cpp.o] Error 1
================================================================ [FAILED] Took 95.82 seconds ================================================================

Environment               Status    Duration
------------------------  --------  ------------
max_ESP32_16M8M_LittleFS  FAILED    00:01:35.825
=========================================================== 1 failed, 0 succeeded in 00:01:35.825 ===========================================================
PS C:\GitHub\TD-er\ESPEasy> 
```

thomastech, 2022-06-01:
@TD-er, you mentioned the problem affects the laptop more than your desktop. Maybe this bug involves the folder order in the PC's path environment?

If you don't mind, please compare the two machines' path environments. Are the VSCode, PlatformIO, and Python related paths in the same order?

For reference, my path has these related folders appended to it:

`;D:\Lang\nodejs\;C:\Program Files (x86)\dotnet\;C:\Users\Tom\AppData\Local\Microsoft\WindowsApps;C:\Users\Tom\.platformio\penv\Scripts;D:\Lang\VSCode\bin;C:\Users\Tom\AppData\Roaming\npm`

I have no idea if that is the optimum folder order for PlatformIO. 

- Thomas

TD-er, 2022-06-01:
> @TD-er could you reproduce this issue WITHOUT VSCode? Bare PIO Core.

You mean not even having VS code open at all?
I did run it in the VS code CLI, but of course this does have VScode open.

If so, then I can collect my laptop from downstairs to test it.

ivankravets, 2022-06-01:
Yes, please close VSCode completely. Open Windows `cmd.exe`, and type
```
cd C:/Users/{NAME}
.platformio\penv\Scripts\activate
pio --version
pio run -d /path/to/project
```

TD-er, 2022-06-01:
OK, will collect my laptop now (and some beer) and start testing. (was still on a call, so late response from my side)

TD-er, 2022-06-01:
Just to make sure there were no half built things in the PIO folder, I did a clean of the PlatformIO stuff.
Also moved back to the PIO 6.0.2 and built via the command line as you mentioned.
This build was successful in the first run.
Will now try to build a few other envs in my config to see if that will succeed too, but first I will try this same env to build from within VS code.

TD-er, 2022-06-01:
And the same env building in VScode failed (not even did a clean, just built it again, but now from within VScode)

```
...
Compiling .pio\build\max_ESP32_16M8M_LittleFS\libe1b\MD_Parola\MD_Parola_Random.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\libe1b\MD_Parola\MD_Parola_Scan.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\libe1b\MD_Parola\MD_Parola_Slice.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\libe1b\MD_Parola\MD_Parola_Sprite.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\libe1b\MD_Parola\MD_Parola_VScroll.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\libe1b\MD_Parola\MD_Parola_Wipe.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\lib681\MFRC522\MFRC522.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\lib412\SparkFun_APDS-9960_Sensor_Arduino_Library\SparkFun_APDS9960.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\lib637\SparkFun_AS7265x\SparkFun_AS7265X.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\lib30a\XPT2046_Touchscreen\XPT2046_Touchscreen.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\src\ESPEasy-Globals.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\src\ESPEasy.ino.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\src\ESPEasy_common.cpp.o
xtensa-esp32-elf-g++: error: src/ESPEasy.ino.cpp: No such file or directory
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pio\build\max_ESP32_16M8M_LittleFS\src\_C001.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\src\_C002.cpp.o
Compiling .pio\build\max_ESP32_16M8M_LittleFS\src\_C003.cpp.o
*** [.pio\build\max_ESP32_16M8M_LittleFS\src\ESPEasy.ino.cpp.o] Error 1
=============================================================== [FAILED] Took 336.20 seconds ===============================================================

Environment               Status    Duration
------------------------  --------  ------------
max_ESP32_16M8M_LittleFS  FAILED    00:05:36.195
=========================================================== 1 failed, 0 succeeded in 00:05:36.195 ===========================================================
The terminal process "C:\Users\gijsn\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'max_ESP32_16M8M_LittleFS'" terminated with exit code: 1.
```

TD-er, 2022-06-01:
I have tried to build a number of other envs in VS code and also via the command line without VS code active.
All seem to work now, so not sure what you did to 6.0.2 compared to the previous dev version, but it seems to have helped here.
Apart from the occasional missing ESPEasy.cpp file as shown above, it all seems to build.
But this missing ESPEasy.cpp (and sometimes missing `setup()` and `loop()`) was already an issue on previous PIO versions.

thomastech, 2022-06-01:
+1 Using the IDE I'm able to build again. "Clean" builds are OK too. 
pio --version reports:
`PlatformIO Core, version 6.0.2rc2`

> All seem to work now, so not sure what you did to 6.0.2 compared to the previous dev version, but it seems to have helped here.

Same here. The recent 6.0.x dev releases did not help me, but today's Core release is so far a winner. Thank you!

Just curious, what was causing all this hair pulling?

- Thomas

TD-er, 2022-06-02:
>  Just curious, what was causing all this hair pulling?

Well I only managed to pull out the grey ones :)

ivankravets, 2022-06-02:
Glad to hear that it works from CLI. The 6.0.2 should improve support for VSCode. Please upgrade to 6.0.2 and restart VSCode. 

But! The main problem is using INO files. PlatformIO converts them to the dynamic CPP, and Windows OS very often does not like it. To avoid potential issues, we DO NOT recommend using of INO files. See guide https://docs.platformio.org/faq/ino-to-cpp.html

If you need INO files, it may be better to use classic Arduino IDE for these projects.

TD-er, 2022-06-02:
Wel "need" is a matter of "not yet converted" for ESPEasy at least.
One thing is that it makes it a bit harder for others to incorporate "plugins" which are not officially part of the repo, but posted elsewhere (e.g. on a blog) to copy/paste into the project and build it.

But I will consider changing the last (about 100+) files into .cpp/.h if that takes away most of these issues.

ivankravets, 2022-06-02:
We supply PlatformIO Core as Python scripts, not a single binary. Very often different espionage “antivirus tools” treat PlatformIO Core as malware (because it generates files on the fly: INO -> CPP). 

If you don’t use INO files, we don’t apply any transformations.

TD-er, 2022-06-02:
Just one more question...
If I change the main source file (ESPEasy.ino) into .cpp/.h, can other files then still be .ino files?

ivankravets, 2022-06-02:
No. INO files are fake C++ files and can’t passed directly to the compiler. I don’t know who invented this shit.

TD-er, 2022-06-02:
Well if it isn't known who did invent it, then I guess he or she already feels too ashamed to admit it ;)

thomastech, 2022-06-02:
FWIW, the two different projects I'm building do not contain ino files. No issues with 5.2.5, but wacky problems with the 6.0.x releases. Mostly reported that .sconsign39.dblite file or its folder was missing with random other errors tossed in on occasion. 

My observation was that the .pio device folder would be cleared at the start of a build and would not repopulate. BTW, sometimes the projects would build after I deleted the .pio folder.  But mostly joyless episodes of hair pulling.

But 6.0.2rc2 is a winner. No problems after this upgrade was auto installed yesterday. So whatever you did to fix the ino related issue had the extra benefit of fixing my builds. The perfect BOGO deal.

Thanks again!
Thomas

ivankravets, 2022-06-02:
Yes, the 6.0.2 fixes project rebuilding. Please upgrade to the stable PIO Core 6.0.2

ivankravets, 2022-06-02:
It is stable, just includes extra fixes. See https://github.com/platformio/platformio-core/blob/develop/HISTORY.rst