platformio/platformio-core #4288: The unit test core does not flowing symbolic links during searching for tests

EugenWiens, 2022-05-24:
### Configuration

**Operating system**:
Ubuntu 20.04

**PlatformIO Version** (`platformio --version`): 
PlatformIO Core, version 6.0.1

### Description of problem
The unit test core does not flowing symbolic links during searching for tests.

#### Steps to Reproduce
Project with following striucture
Project
```
  `- src
  `- lib
      `- module1
            `-test
                 `- test_one
  `- test
        `- module1 -> ./lib/module1/test
```

### Actual Results
the unit test core does't find any tests

### Expected Results
the unit test core should find the folder `test_one` 

### If problems with PlatformIO Build System:

ivankravets, 2022-05-26:
Thanks for the report. Please re-test with `pio upgrade --dev`. 

Does it work now?

EugenWiens, 2022-05-30:
yes with 6.0.2b1 the symbolic link works again. When we can see it in the production version?

ivankravets, 2022-05-30:
We plan Core 6.0.2 for today/tomorrow.

lemonkey, 2022-06-04:
@4303  Thanks for bringing this up! 💯