platformio/platformio-core #4292: Device Monitor crashes when device is disabled

zfields, 2022-05-25:
I have a target device that "goes to sleep" by being externally disabled via the `EN` pin. When my device is disabled, the serial monitor crashes.

This is unfortunate, because I cannot connect to a port that is not open (i.e. a disabled device). I have no way to connect to the device unless it is running, which means I will miss all the startup logs.

I need the monitor to gracefully handle the loss of connection, await for the device to reappear, and reestablish the connection.

ivankravets, 2022-05-25:
I updated this issue with a temporary solution. I'm not sure that we can fix this on our side. A related issue https://github.com/platformio/platformio-core/issues/3939

ivankravets, 2022-06-10:
Duplicate of https://github.com/platformio/platformio-core/issues/3939

closed 2022-06-10 by ivankravets