platformio/platformio-core #4294: ESP32S2 - Loosing USB conection and Serial Monitor output on chip reset

Algonsi, 2022-05-29:
Kind of issue:
- [ X] **Feature Request**.

------------------------------------------------------------------

### Configuration
platform = espressif32
board = lolin_s2_mini
framework = arduino

**Operating system**:
Windows 11

**PlatformIO Version** (`platformio --version`):
PLatformIO Core, version 6.0.1

### Description of problem
Since the ESP32S2 chip implements the USB connection itself (it has the USB peripheral), when it restarts, it temporarily loses its USB connection (so the monitor output crashes) and then reconnects (but monitor does not actualizes). This is normal behavior. The same goes for other microcontrollers implementing USB themselves, such as Atmel ATSAMD21 chips.

Other microcontroller boards don’t do this if they have a dedicated USB-serial chip (e.g., CH340, CP2102, …) on them that is not affected by the restart of the main microcontroller, and thus don’t have this problem.

I’m not sure whether anything can be done against that at all – The USB disconnect is pretty much bound to happen when you ESP.restart(). I guess PlatformIO could have an option where it tries to reconnect to the lost device for a while.

#### Steps to Reproduce
ESP.restart();

### Actual Results
Loss of USB connection and serial monitor output

### Expected Results
Serial monitor automatically continues showing output after USB reconnection

ivankravets, 2022-05-31:
@Wick3dJack do you use the latest version of ESP32 dev platform? Please forward your issue to https://community.platformio.org/

egnor, 2022-06-07:
This seems related to https://github.com/platformio/platformio-core/issues/3939, perhaps even the same issue?

ivankravets, 2022-06-07:
Duplicate of https://github.com/platformio/platformio-core/issues/3939

Please check temporary https://docs.platformio.org/en/latest/faq/advanced-serial-monitor.html

closed 2022-06-07 by ivankravets