platformio/platformio-core #4296: Add "project_dir" to platformio.ini to customize project directory

guestisp, 2022-05-30:
What kind of issue is this?

- [X ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

Actually, it's impossible to set a custom project directory , project dir is always the same where platformio.ini file is located.
If, for example, you would put platformio.ini file in the root directory of the project  but keeping the "firmware" files in a subdirectory called `firmware` you have to set 5-6 directory manually in platformio.ini like this:

```
[platformio]
src_dir = subproject/src
lib_dir = subproject/lib
include_dir = subproject/include
test_dir = subproject/test
data_dir = subproject/data
```

probably, adding a setting called `project_dir` would be useful, if set, `src_dir`, `lib_dir` and so on, should be relative to that `project_dir`

One flag, to change all.

ivankravets, 2022-05-30:
Thanks for the feature request. I don't think we will provide an ALTERNATIVE layout that can lead to misunderstanding. 

PlatformIO Core allows you to pass your own configuration file, even if it is not located in the project directory. See https://docs.platformio.org/en/latest/core/userguide/cmd_run.html#cmdoption-pio-run-c

guestisp, 2022-05-30:
but when opening the repository directory with vscode, if you don't have platformio config file in the root, vscode doesn't start pio at all... i have to move the conf file one level up but after that the project directory is moved one level up and can't build anything

as you just have a lot of settings allowing to set custom folders, why not adding one setting more to configure the project folder? custom layout is already possible, just change one dir setting and you have a custom layout

ivankravets, 2022-05-30:
Other software depends on the `platformio.ini`, including PlatformIO IDE. The `platformio.ini` file is an indicator of the PlatformIO Project.

guestisp, 2022-05-30:
i'm not asking to remove the file, just adding an option inside...

ivankravets, 2022-05-30:
Sorry, it was my mistake, I didn't read well your request. Yes, it was marked as a feature request. Thanks!

guestisp, 2022-05-30:
From this:
```
[platformio]
   src_dir = subproject/src
   lib_dir = subproject/lib
   include_dir = subproject/include
   test_dir = subproject/test
   data_dir = subproject/data
```

to this:
```
[platformio]
   project_dir = subproject
```

the result is the same, because `src_dir` and others are relative to the projectdir, but actually the projectdir is where the platformio file is stored. With my request, we can store the platformio one level up and just change 1 parameter to set the sub-directories.

It's already supported, by changing 5-6-7 parameters, i'm just asking to add 1 parameter.

ivankravets, 2022-06-02:
I've just made some quick research. We can't implement this feature. The `project_dir` in PlatformIO is a CWD. So, having multiple values for `project_dir` will lead to unpredictable behavior. You have 2 solutions to this issue:

1. Override values as you did above
2. Use a custom project configuration file with `-c` option to run/test/debug commands.