platformio/platformio-core #4302: inversion of RTS/CTS  parameter in  esp32-hal-uart.c :: uartSetPins.

moriamoria, 2022-06-01:
### Configuration

*ESP32*:

PlatformIO Version latest

### Description of problem

**esp32-hal-uart::uartSetPins invert RTX/CTX parameter**

uartSetPins (ctsPin,rtsPin) invoque uart_set_pin(_ctsPin_, **rtsPin**) but  the signature of the method uses 

esp_err_t uart_set_pin(... **rts_io_num**,  _cts_io_num_);

void uartSetPins(uart_t* uart, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)
{
...
    uart_set_pin(uart->num, txPin, rxPin, ctsPin, rtsPin); 
...
}

The correct code should be 👍 

    uart_set_pin(uart->num, txPin, rxPin, rtsPin, ctsPin); 

Regards

ivankravets, 2022-06-01:
Please forward to https://github.com/espressif/arduino-esp32/issues

moriamoria, 2022-06-01:
Hello,

The problem is allready fix in the master of espressif, could you update your package ?

https://github.com/espressif/arduino-esp32/pull/6816

[me-no-dev](https://github.com/me-no-dev) merged commit [5484129](https://github.com/espressif/arduino-esp32/commit/548412952ba58e299ad83d932cea417a2490fd78) into espressif:master [5 days ago](https://github.com/espressif/arduino-esp32/pull/6816#event-6695685084)

Regards

ivankravets, 2022-06-01:
https://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version