platformio/platformio-core #4303: Allow global source filter

maxgerhardt, 2022-06-03:
What kind of issue is this?

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64

**PlatformIO Version** (`platformio --version`): `6.0.2`

### Description of problem

Source filters can be applied to folders within `src` by using [`build_src_filter`](https://docs.platformio.org/en/latest/projectconf/section_env_build.html?highlight=build_src_filter#build-src-filter) and in self-controlled libraries through [`srcFilter`](https://docs.platformio.org/en/latest//manifests/library-json/index.html#srcfilter) in the `library.json`

However, sometimes, especially in STM32Cube projects though, this is not sufficient and a global source filter that applies to every source file in the build system would be helpful. 

Take for example [this official STM32Cube H7 ADC example](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/STM32H747I-DISCO/Examples/ADC/ADC_DualModeInterleaved/SW4STM32/STM32H747I_DISCO/.project#L134-L138). It uses the Drivers\BSP\STM32H747I-DISCO library, but only a single source file from it: `stm32h747i_discovery.c`. When telling PlatformIO to use that BSP variant with

```
[env:disco_h747xi]
platform = ststm32
board = disco_h747xi
framework = stm32cube
build_flags = -DCORE_CM7 -Iinclude
board_build.stm32cube.variant = STM32H747I-DISCO
```
(and the `stm32h747i_discovery_conf.h` is copied from the [official example](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/STM32H747I-DISCO/Examples/ADC/ADC_DualModeInterleaved/Common/Inc/stm32h747i_discovery_conf.h)), still a ton of include errors occur from source files like `stm32h747i_discovery_audio.c` which were never supposed to be compiled in the first place.

Since the library [is added by the builder script](https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/stm32cube.py#L344-L354) and not in the user's control (if he wants to use what's in `framework-stm32cubeh7`), the user cannot easily exclude these source files. Things like

```ini
build_src_filter = +<*> -<**\Drivers\BSP\STM32H747I_DISCO\*> +<**\Drivers\BSP\STM32H747I_DISCO\stm32h747i_discovery.c>
```

or an extra script with
```py
Import("env")
def skip_build(node):
    if "node.name" == "stm32h747i_discovery.c":
        return node
    return None
env.AddBuildMiddleware(skip_build, "**/STM32H747I-DISCO/*.c")
```

have *no effect*. Only remedies are a manual copy of the needed files. 

This is not nice and discourages people from going using PlatformIO with STM32Cube projects, all which do similiar things with their selective source code selection.

Thus, a e.g. `build_global_src_filter` option or similiar would be nice.

#### Steps to Reproduce

1. Use the provided platformio.ini and code
2. Copy [stm32h747i_discovery_conf.h](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/STM32H747I-DISCO/Examples/ADC/ADC_DualModeInterleaved/Common/Inc/stm32h747i_discovery_conf.h) into `include/` folder
3. Observe errors

### Actual Results

6 fatal inlude errors.

### Expected Results

Successfull compilation given the usage of a mechanism that excludes the non-needed files.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:disco_h747xi]
platform = ststm32
board = disco_h747xi
framework = stm32cube
build_flags = -DCORE_CM7 -Iinclude
board_build.stm32cube.variant = STM32H747I-DISCO
```

**Source file to reproduce issue:**
```cpp
#include <stm32h7xx_hal.h>
#include <stm32h747i_discovery.h>

void main() {
}
```

### Additional info

ivankravets, 2022-06-03:
> or an extra script with

Did you use `pre:` extra script?

maxgerhardt, 2022-06-03:
Okay with `pre:` the `env.AddBuildMiddleware()` script works, the `env["SRC_FILTER"]` one not. Still, someone trying to just port a STM32Cube project to PlatformIO as hi first task will have difficulties coming up with that one.

ivankravets, 2022-06-03:
Thanks! I'll think about how to implement this.