platformio/platformio-core #4305: Allow post scripts in library.json

maxgerhardt, 2022-06-06:
What kind of issue is this?

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64

**PlatformIO Version** (`platformio --version`): 6.0.2

### Description of problem

When writing a library, it sometimes is necessary to affect build settings of the `projenv` environment which is used to build the `src/` folder. Such as in [this example](https://community.platformio.org/t/force-rtti-from-library/28176), where the inclusion of the library should cause all `-fno-rtti` flags to vanish.

This is only partially possible in the `extraScript` of a library. Using e.g. as script
```py
Import("env")
global_env = DefaultEnvironment()
for e in (env, global_env):
    try:
        # key may or may not exist
        e['CXXFLAGS'].remove("-fno-rtti")
    except:
        pass
    e['CXXFLAGS'].append("-frtti")
```

leads to all `-fno-rtti` flags being removed in the library and framework compilation, but not for the files in `src/`. Of course, `
build_unflags = -fno-rtti` can be set in the `platformio.ini` to counter this, but then the user has to do an extra step when including the library.

Another use case is e.g. [this library](https://github.com/likeablob/ulptool-pio) where users are being told to add `extra_script` to their `platformio.ini` that points into the library, because the `library.json` has no mechanism for post and pre scripts.
```ini
lib_deps =
  https://github.com/likeablob/ulptool-pio

extra_scripts =
  pre:/$PROJECT_LIBDEPS_DIR/$PIOENV/ulptool-pio/pre_extra_script_ulptool.py
  post:/$PROJECT_LIBDEPS_DIR/$PIOENV/ulptool-pio/post_extra_script_ulptool.py
```

It would be nice to specify `pre:` and `post:` in the `extraScripts` field / array of the `library.json` to allow this better without user interaction.

#### Steps to Reproduce

1. Download and open https://github.com/maxgerhardt/rtti-repro
2. Inspect `lib/MyLib/with_rtti.py`
2. Build project

### Actual Results

Compilation failure, `-fno-rtti` not fully removed.

```
src\main.cpp: In function 'void setup()':
src\main.cpp:12:40: error: cannot use 'typeid' with '-fno-rtti'
   12 |     const std::type_info& ti1 = typeid(d);
      |                                        ^
```

### Expected Results

Possibility to affect `src/` compilation in `extraScript`. 

### If problems with PlatformIO Build System:

See linked project.

### Additional info
-/-

cziter15, 2022-06-16:
Looks like projenv feature is working as expected :)

cziter15, 2022-06-16:
After digging some more in projects it doesn't look as good. Don't really know what's wrong, but using projenv and ProcessUnFlags gives unexpected linker errors in some projects. Using unflags in platformio.ini does not bring these issues. More detais at https://community.platformio.org/t/force-rtti-from-library/28176

cziter15, 2022-06-21:
Jost FYI - all problems have been resolved. Thanks!

mathemaphysics, 2024-07-06:
This doesn't work. It ignores me and builds e.g. libFS.a with no RTTI. I have no idea what is going on.

I have no control over this and typeinfo vanishes for things I need it for randomly.