platformio/platformio-core #4307: PlatformIO test building code twice run

robsonos, 2022-06-07:
What kind of issue is this?

- [x] **PlatformIO Core**.
      
------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.0.2

### Description of problem

PlatformIO test seems to be compiling twice

#### Steps to Reproduce

1. Create a project
2. Create a test
3. Run `pio test -vvv`

### Actual Results
The two images below are the results from the same `pio test -vvv` run. I split them in two to increase the resolution 
[image]
[image]

### Expected Results
 PlatformIO test to build once per run

### Additional info
 Commenting lines 139, 141 and 142 from the file below seems to do the trick. I am not sure what impact this would have on other platforms.

https://github.com/platformio/platformio-core/blob/11a43b26930c87c4b853f4dde756f7217d838c51/platformio/test/runners/base.py#L139-L142

ivankravets, 2022-06-07:
There are 2 different stages: firmware building and firmware uploading. The firmware uploading stage depends on the building stage. There is no "rebuilding", the "uploading" stage reuses all objects that were compiled at the first stage.

If your project has custom requirements, you can override default configuration with a custom testing framework https://docs.platformio.org/en/latest/advanced/unit-testing/frameworks/custom/index.html

robsonos, 2022-06-08:
Hi @ivankravets,

I am aware of those stages, what I am pointing out is that Platformio is effectively building the test firmware twice, which can be very time consuming for large unit tests.

I am running this test in a simple uno project:

platfromio.ini:
```
[env:uno]
platform = atmelavr
board = uno
framework = arduino
```

src/main.cpp (not really needed, but here it is):
```
#include <Arduino.h>

void setup() {}

void loop() {}
```

test/tets_test/main.cpp:
```
#include <Arduino.h>

void setup(){
    Serial.begin(9600);
}

void loop(){
    Serial.println("test");
    delay(1000);
}
```

As previously mentioned, my understanding is that `stage_uploading()` verifies for `if self.options.without_building` , which is never true for unit testing (unless of course if `--without-building` is passed in the cli). Thus `"__test")` is added to the target list, which causes a second build just prior to uploading.

ivankravets, 2022-06-08:
Thanks, I think we can merge these stages into the one if there are no custom stages are passed.

ivankravets, 2022-06-12:
Thanks for the report! Please re-test with `pio upgrade --dev`.

Does it work as expected now?

closed 2022-06-12 by ivankravets

robsonos, 2022-06-14:
No worries. It works like a charm now. 
Cheers.

commit 4b5bc91a (2022-06-12): Merged the Unit Testing “building” stage with “uploading” for the embedded target // Resolve #4307