platformio/platformio-core #4308: `pio device monitor` sometimes picks the wrong port

egnor, 2022-06-07:
What kind of issue is this?

- [✔️] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 22.04 (Linux 5.15.0) -- but I suspect it's a universal issue

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.0.2

### Description of problem
If `pio device monitor` can't find a port matching the current board config, [it will pick any old port](https://github.com/platformio/platformio-core/blob/575f0ae3008c3ae34795ced1311d72e244f4b5ba/platformio/device/finder.py#L61) (preferring USB ports).

This creates difficulty because a very normal thing is to do something like
`pio run --target upload && pio device monitor`

However, boards that self-host USB very typically drop off USB for a moment after flashing and before restarting the app. That means that right at the instant `pio device monitor` is working, the port doesn't appear. Instead, it picks some other random serial port on the system (necessarily incorrect) and you are now talking to your webcam or something. Yuck.

IMHO: I would suggest that `find_serial_port` should be pickier. If it's given a board config, it should insist on finding a serial port matching that board config. (And if it's given a blackmagic upload protocol, it should insist on finding a serial port using whatever that weird algorithm is.) If one doesn't exist, it should retry for a few seconds. If that doesn't work, it should fail, and suggest that the user should explicitly supply a port.

(Without the criteria of board config etc, it should probably be willing to grab a single serial port if there's only one, otherwise failing if there are zero or many.)

ivankravets, 2022-06-07:
What is your board?

egnor, 2022-06-07:
The board I'm using right now is a [Seeeduino XIAO](https://www.seeedstudio.com/Seeeduino-XIAO-Arduino-Microcontroller-SAMD21-Cortex-M0+-p-4426.html). I've seen the same problem with an [Adafruit Metro M4](https://www.adafruit.com/product/3382). Like many modern dev boards, they both implement USB on-chip, and disconnect during the reset-to-user-app transition.

(A different failure case is finding the old port that was set up by the bootloader, connecting to it, and having it disappear milliseconds later as the bootloader reboots into the app. That one's a bit harder to resolve, except through general disconnect/retry handling, or perhaps just making sure the bootload process leaves enough of a delay or has an explicit handshake to be sure the old port is gone before declaring victory.)

ivankravets, 2022-06-14:
How do you use `pio device monitor` that you hit with this "reboot" delay?

egnor, 2022-06-14:
My (extremely) typical use case is:

`pio run --target upload && pio device monitor`

Build, upload, run, watch the output... isn't this what everyone does? I ended up replacing it with

`pio run --target upload && sleep 1 && pio device monitor`

But that's gross. I do tend to have various other devices connected which show up as USB serial ports, so, that exacerbates the problem.

ivankravets, 2022-06-15:
Thanks for the issue. Could you re-test the latest development branch? 

```
pio upgrade --dev
```

Does it work now?

closed 2022-06-15 by ivankravets

egnor, 2022-06-16:
Yes it does seem to work! Thank you!!

commit 743a3e2c (2022-06-15): Improved a serial port finder for a board with predefined HWIDs // Resolve #4308