platformio/platformio-core #4309: `build_unflags` removes flags specified in `build_flags` (can't remove and replace -D with new value)

egnor, 2022-06-08:
What kind of issue is this?

- [✔️] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 22.04 LTS (Linux 5.15.0)

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.0.2

### Description of problem

I wanted to remove and replace a platform build flag with my own, so I put the following in `platformio.ini`:

```
build_unflags = "-DUSB_PRODUCT=\"Seeeduino XIAO\""
build_flags = "-DUSB_PRODUCT=\"My Project\""
```

However, while this did successfully remove the platform `-DUSB_PRODUCT`, it also removed my replacement, leaving me with no such flag (as visible in `pio run-v`), and the platform code used a default value instead. As far as I can tell this is because, first, unflags are run through a parser and considered to match if the flag-part (`-DUSB_PRODUCT`) but not necessarily the value-part (`\"Seeeduino XIAO\"`) matches; second, `build_unflags` will apparently remove flags that were added with `build_flags`.

Changing either one of these two things would allow this to work. (Removing and replacing `-D` type flags has to be a common desire?) Note, I was able to work around this specific case with the following:

```
build_flags = '-UUSB_PRODUCT "-DUSB_PRODUCT=\"PG Color Sensor\""'
```

However, not every flag one might wish to remove and replace has a `-U` type antiflag.

#### Steps to Reproduce

1. Add `-DFOO=bar` to `build_unflags`
2. Add `-DFOO=bat` to `build_flags`
3. Observe the actual flags used with `pio run -v`

### Actual Results

No `-DFOO=...` is included

### Expected Results

`-DFOO=bat` is included

ivankravets, 2022-06-17:
Thanks for the report! Please re-test with `pio upgrade --dev`.

closed 2022-06-17 by ivankravets

egnor, 2022-06-20:
Seems to work!! Thank you as always!!!

commit db6b8a6d (2022-06-17): Fixed an issue when `build_unflags  operation ignores a flag value // Resolve #4309